MarlinFirmware/Marlin #11214: Assistance porting to Einstart S mega2560 based board

eltoozero, 2018-07-07:
### Description
I've got functional printing, but it was quite a hack to get everything functional on a board that uses pins not enabled in the Arduino IDE.

Initially I created  a 'nude' atmega2560 variant (attached), which untangles the pins, exposing pins not on an "arduino mega" and allows direct mega2560 centric pin selection in the Configuration.h

Things that were working when run as pure sensor test sketches (motors, thermistors, endstops), would not function in Marlin due to fastio HAL, which bypasses the pin order via register manipulation.

So I copied my port specs into the fastio library wrapped in an #if MB(EINSTART_S), and that worked.

So now I'm printing, but it seems like a nasty hack to have to add a cpu variant to the IDE, so I'm wondering if there's a more elegant way of providing support for this machine.

#### no menu nav

Separately, I had to create a custom LCD definition thanks to this thing using the SH1106 driver instead of SSD1306; but that was easy enough, however I'm not getting menus with my cardinal buttons. The definitions appear in M43 and change state when pressed, but are default high, and I don't believe there's an option to invert buttons in the config.

I would assume either the lack of inversion on menu buttons or the custom LCD definition to be part of the problem, but my kill pin works, and that's just configured for the right arrow, indeed M43 confirms BTN_RT and KILL_PIN are one and the same, and it doesn't seem to care about inverting state there...ponderous.

#### add chassis LED support

The final "feature" I'd like to implement is adding support for the I2C PCA9533 PWM LED driver, which this machine uses for RGB illumination of the chassis, this should be fairly simple since a somewhat similar I2C PCA9632 is already natively supported.

### Additional Information

Go easy on me, I'm capable at this but but I'm pretty sure nobody would hire me as a developer.

All my modifications were done on a downloaded copy of the 2.0.0-bugfix release, I've got no method in-hand to identify all the modifications I've made, but I figured once I got it working it could be sorted out.

[Marlin-bugfix-2.0.x + Einstart Changes.zip](https://github.com/MarlinFirmware/Marlin/files/2172368/Marlin-bugfix-2.0.x.Einstart.Changes.zip)

I've also included my "naked" atmega2560 variant which uses native chip pin numbers.
[nude avr variant.zip](https://github.com/MarlinFirmware/Marlin/files/2172370/nude.avr.variant.zip)

[Video of machine printing](https://youtu.be/P81mUy7h6ak)
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2172365/configs.zip)

Board Images:
[image]
[image]
[image]

thinkyhead, 2018-07-07:
The best way to manage a modified Marlin is to create a fork under your Github account. Use the Github interface plus the GitHub Desktop application to clone your fork to your desktop. Then create a new work branch where you will be applying your changes. Make the necessary changes to your work branch and commit those changes using the GitHub Desktop application. You can then push them to your fork on GitHub for public review. Github and the Github Desktop application makes it easy to track your changes. See GitHub's clear and thorough documentation for guidance.

eltoozero, 2018-07-07:
👍🏻 Thanks Scott!

First time for everything, hope I did this correctly: https://github.com/eltoozero/Marlin/tree/changes-for-einstart-s

Indeed mucking about in the HAL is not required, and the changes are simply an updated draft boards.h and pins_EINSTART-S.h plus changes to ultralcd.h and impl files for the new declaration.

eltoozero, 2018-07-17:
My branch is ready for merge pending any disagreement, is @thinkyhead on vacation? ;)

I've since added support for the PCA9533 to incorporate LED support, since it's probably only used in this one machine should I add it to this branch or make a separate one for those changes? (PWM only on Red/Blue LEDs so no orange but at least heating violet -> red looks right)

I got no commentary on my notes re: u8glib not utilizing fastio, which is a bummer since it means **when LCD is connected to unexposed pins an IDE level AVR variant incorporating unexposed pins is required for functional LCD support**; I proposed putting my variant in the config examples folder with explanation in the readme, but got no feedback.

Finally, is there an open issue to discus optimization of cardinal button handling? It's awkward as it stands but I'm prepared to assist with fixes.

https://github.com/eltoozero/Marlin/tree/changes-for-einstart-s

closed 2018-11-05 by thinkyhead