MarlinFirmware/Marlin #11217: MKS SBASE 1.3 - Error: Printer halted. kill() called! after G28. All led switch on

maar1201gh, 2018-07-07:
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:1_20180707_MR EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:EEPROM:0
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:0
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:1
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: ok
Send: M21
Recv: echo:Unknown command: "M21"
Recv: ok
Send: M155 S2
Recv: ok
[...]
Send: M111 S247
Recv: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
Recv: ok
[...]
Send: G28
Recv: echo:G28
Recv: >>> G28
Recv: Machine Type: Core
Recv: Probe: NONE
Recv:   current_position=(0.00, 0.00, 0.00) : bracket_probe_move
Recv: > endstops.enable(true)
Recv: Raise Z (before homing) to 5.00
Recv: >>> do_blocking_move_to(0.00, 0.00, 5.00)
Recv: <<< do_blocking_move_to
[...]
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -412.50, [20.00])
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv: echo:busy: processing
Printer seems to support the busy protocol, adjusting timeouts and setting busy interval accordingly
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 275.00
Recv: > home_offset[X] = 0.00
Recv:   current_position=(0.00, 0.00, 5.00) :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
Recv:   current_position=(0.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, 300.00, [20.00])
Recv:   current_position=(0.00, 0.00, 5.00) : sync_plan_position
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 200.00
Recv: > home_offset[Y] = 0.00
Recv:   current_position=(0.00, 200.00, 5.00) :
Recv: <<< set_axis_is_at_home(Y)
Recv:   current_position=(0.00, 200.00, 5.00) : sync_plan_position
Recv:   current_position=(0.00, 200.00, 5.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(Y)
Recv: Z_SAFE_HOMING >>>
Recv:   current_position=(0.00, 200.00, 5.00) : sync_plan_position
Recv:   destination=(20.00, 20.00, 5.00) : Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(20.00, 20.00, 5.00)
Recv: <<< do_blocking_move_to
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: >>> homeaxis(Z)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -420.00, [6.00])
Recv:   current_position=(20.00, 20.00, 0.00) : sync_plan_position
Recv: echo:busy: processing
[...]
Recv: <<< do_homing_move(Z)
Recv: Move Away:
Recv: >>> do_homing_move(Z, 2.00, [6.00])
Recv:   current_position=(20.00, 20.00, 0.00) : sync_plan_position
Recv: <<< do_homing_move(Z)
Recv: Home 2 Slow:
Recv: >>> do_homing_move(Z, -4.00, 1.50)
Recv:   current_position=(20.00, 20.00, 0.00) : sync_plan_position
Recv: echo:busy: processing
[...]
Recv: <<< do_homing_move(Z)
Recv: >>> do_homing_move(Z, 0.00, [6.00])
Recv:   current_position=(20.00, 20.00, 0.00) : sync_plan_position
Recv: Error:Printer halted. kill() called!

GMagician, 2018-07-10:
Try to read #11224 maybe you have the same issue

maar1201gh, 2018-07-10:
@GMagician, Thank you for your help. #11224 is my issue
I changed   #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0.025 and after 2 weeks i finally have my printer live again.

closed 2018-07-10 by maar1201gh

thinkyhead, 2018-08-22:
Duplicate of #11224