MarlinFirmware/Marlin #11221: AD595 Thermocouple and RE-ARM incorect reading

moth4017, 2018-07-08:
Issue using Thermocouple  & AD595 and RE-Arm , it used to work with the Arduino , , i have set #define TEMP_SENSOR_0 -1 , In pins.h #define TEMP_0_PIN 3 , but the temperature  reading displayed  is now 500 Deg.C (should be 18 Deg.C) will this be a different pin on the RE-ARM than the Arduino?

Bob-the-Kuhn, 2018-07-08:
See https://reprap.org/wiki/Re-ARM for details.

The analog inputs are different on Re-ARM with RAMPS than 
2560 with RAMPs.

Only the T0, T1 & T2 pins are useable as analog inputs.

The default Re-ARM pin definitions are:
```
//
// Temperature Sensors
//  3.3V max when defined as an analog input
//
#define TEMP_0_PIN         0   // A0 (T0) - (67) - TEMP_0_PIN
#define TEMP_BED_PIN       1   // A1 (T1) - (68) - TEMP_BED_PIN
#define TEMP_1_PIN         2   // A2 (T2) - (69) - TEMP_1_PIN
```

I suggest moving the thermocouple wire to the T0 pin on the Re-ARM main board.

moth4017, 2018-07-08:
The problem with using the Thermister inputs on the ramps board , the thermister inputs have a 4.7K resister to 0V  which effects the output from the AD595,  i could remove the resistor  but nice to be able to  use the AD595 without modifying the ramps board, after trying the AD565 connected to the T0 input at 18Deg C ambient the T0 input was displaying a temperature of 178 Deg C  .

moth4017, 2018-07-08:
I think i need help ;) 
so D63 on the Rearm is still an analogue i/p which seem to be connected to Ramps v1.4 Pin Aux1 pin 3 , which would have been   pins.h #define TEMP_0_PIN 3, but when i look at other documentation (rearm Wiki) it has D63 on Aux 2 pin 4.

I can get neither to work assuming that  Aux 2 pin 4 in the Pins_ramps.h  is pins.h #define TEMP_0_PIN 9.

moth4017, 2018-07-08:
i also tried using the Analogue inputs direct on the RE-Arm PCB by re soldering the pads to activate the inputs , but then released it also had the 4K7 resistor pull up so the displayed temperature was 178 deg C .

So i do need a Analogue ip , but where.. ?

p3p, 2018-07-09:
From the pins [file](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/pins_RAMPS_RE_ARM.h#L126) it looks like ADC channel 3 should be okay for use with a Thermocouple, I don't have one to check though.

Bob-the-Kuhn, 2018-07-09:
The Re-ARM board uses the RAMPS shield but some of the pins are assigned different logical pin numbers.

You need to be working with the **pins_RAMPS_RE_ARM.h** file.

Here's the default defines.

```
// Temperature Sensors
//  3.3V max when defined as an analog input
//
#define TEMP_0_PIN         0   // A0 (T0) - (67) - TEMP_0_PIN
#define TEMP_BED_PIN       1   // A1 (T1) - (68) - TEMP_BED_PIN
#define TEMP_1_PIN         2   // A2 (T2) - (69) - TEMP_1_PIN
#define TEMP_2_PIN         3   // A3 - (63) - J5-3 & AUX-2
#define TEMP_3_PIN         4   // A4 - (37) - BUZZER_PIN
//#define TEMP_4_PIN         5  // A5 - (49) - SD_DETECT_PIN
//#define ??               6   // A6 - ( 0)  - RXD0 - J4-4 & AUX-1
#define FILWIDTH_PIN       7   // A7 - ( 1)  - TXD0 - J4-5 & AUX-1
```

moth4017, 2018-07-09:
@Bob-the-Kuhn  that would explain why... my bad

moth4017, 2018-07-09:
Ok just checked pin A3 > Aux 1 pin 5 on Ramps has 323mv  but the display is still showing 55 deg C ( does not change when voltage changes on input to A3)  
i have #define TEMP_0_PIN        3 in pins_ramps_re-Arm.h
in Con*.h  #define TEMP_SENSOR_0 -1

p3p, 2018-07-09:
@moth4017 The Re-ARM is not an Arduino compatible board so you cant make assumptions, what would be A3 is connected to P0_27 which is not an ADC pin, ADC channel 3 is P0_26 connecting to what would be A9 on an Arduino passing through to Aux2-4 on the Ramps

moth4017, 2018-07-09:
BINGO :)  now reading correct temperature , thanks very much .. now to do the filament run out sensor

Has any one done an re-assignment chart yet , as the REarm board is sold as a Arduino update but as i discovered it was not just a case of swapping them over..

moth4017, 2018-07-09:
Anyway to calibrate the Thermo couple Ad565 as it is reading a bit high was within 1DegC on Arduino  but now reading 8Deg C higher than it should

moth4017, 2018-07-09:
when setting Hot end to 100DegC    thermocouple reads 101  but hot end is only at 57 Deg c

p3p, 2018-07-09:
I presume you have to power the ad565 off 5V, the resulting output being a faction of that 5V depending on temperature? if so the reading will be skewed as the Re-ARMs ADC reference voltage is 3.3v. If you go above that you risk damaging it so be careful.

moth4017, 2018-07-09:
just a quick check at a setpoint of  200Deg  C  hot end reached 101 Deg C

p3p, 2018-07-09:
> Yes it is powered from 5V but the module is +5 v to 15V with a 10mv/C output so the voltage should not make any difference

ADCs work by calculating the ratio between the signal and a reference voltage, say you have a temperature of 100C, thats 1000mV so on Arduino (1V / 5V) 0.2, on ARM (1V / 3.3V) 0.3, making the reading higher than expected. You will probably need to add a voltage divider to adjust the signal appropriately. 

In theory we could do it software but having a signal that could go above 3.3V and blow the board in not a good idea anyway.

moth4017, 2018-07-09:
As all these modules are going to be the same for the AD595 would it not be better to change the calculation in Marlin

p3p, 2018-07-09:
We can't support 5V devices on 3.3V ADC hardware, I'l say again that you will damage the Re-ARM .. it need a voltage divider.

moth4017, 2018-07-09:
when the output gets to 3.3V the hotend will be 330DegC  not many people print that hot, also the rearm states that it is able to accept 5V Max on its inputs.

Bob-the-Kuhn, 2018-07-09:
The 5V max is for almost all the digital I/O.  When configured as an analog input the max is 3.3V

moth4017, 2018-07-09:
thats easy then ,if and AD595 TC1 is selected  get marlin to set Max temp to 330DegC   and do the recalculation for the TC1 board to display the correct reading ,

I

thinkyhead, 2018-08-22:
@moth4017 — Yeah, we all started out not understanding pull requests. Keep reading GitHub's helpful documentation, watch some YouTube videos on the subject, etc., and you'll start to get it.

boelle, 2019-02-20:
@moth4017 still having an issue with latest bugfix 2.0?

closed 2019-02-20 by thinkyhead