MarlinFirmware/Marlin #11223: UBL Mesh edit loop hangs on all axis known

silentninja1, 2018-07-08:
### Description

UBL Mesh edit screen hangs

### Steps to Reproduce

Select any method of getting to the UBL mesh edit screens

**Expected behavior:Turn dial and hotend moves

**Actual behavior:System not responsive to commands

#### Additional Information
Commenting out the following line (2416) in ultralcd.cpp resolves the issue
if (!all_axes_known()) return lcd_goto_screen(_lcd_ubl_map_homing);

Im not sure this line is necessary as the homing command prior blocks this menu until its complete. Ive commented it out for my build, however I wanted to confirm if the function is still desired as a fail safe or if its safe to remove before issuing a PR

GMagician, 2018-07-08:
Yes it's important. Once you have done homing and don't move anything, motors will be turned off.
As every stepper, when turned off it may move to a "rest" position that is a full step.
This command will ensure to inhibit ubl when positions are not known exactly.

If this will solve maybe you have such timeout too short, or you wait too much time before doing ubl edit, or maybe a real bug is present.

Check your lcd if something is blinking or ?? appears instead of position.

Be sure to test bugfix branches beacuse 1.1.8 is "old" and some bug have been fixed in between.

Roxy-3D, 2018-07-08:
@silentninja1    Thank you for the fix...   I've zapped both bugfix branches with your correction.

> Commenting out the following line (2416) in ultralcd.cpp resolves the issue
```
if (!all_axes_known()) return lcd_goto_screen(_lcd_ubl_map_homing);
```

GMagician, 2018-07-08:
> Thank you for the fix...

what it means to not know Marlin in deep. I thought that was a protection and it was a bug.....lol

InsanityAutomation, 2018-07-08:
Gmagician I hear ya there! Following the process though, that validation was done before that line and the edit should already have been active so it jumped out as suspect. When I put braces around it and added serial pair outputs to look at the conditions through there, it showed it was stuck since that value was cleared when the homing finished and that screen was brought up. Still much simpler of a fix than what Scott's working on with the power recovery!

silentninja1, 2018-07-11:
With this - https://github.com/MarlinFirmware/Marlin/commit/fae2929b6b513c5ff6b68a7a8a88274d381f76e7  Im going to close the issue.

closed 2018-07-11 by silentninja1