MarlinFirmware/Marlin #11224: [bugfix-1.1x] X axis homing "ghost" endstop behavior

Murloc992, 2018-07-08:
### Description

I am trying to setup marlin bugfix-1.1.x on a Trigorilla 1.3 (not 32bit) board.

My X-Axis doesn't want to home no matter what I change. Endstop triggers are very fine. It feels like X axis somehow gets limited on the "MIN" side.

Other axis seems to be fine.

Before I disabled endstop validation printer would just halt itself on X homing.

### Steps to Reproduce

1. Manually move X axis to the maximum possible
2. Send a G28 or just G28 X0
3. Observe as X axis does a "ghost" endstop movement in a random spot of the axis, endstop is OPEN
4. Send multiple G28 (or X0) commands till it homes properly
5. Endstop is now TRIGGERED

**Expected behavior:** It should home on X axis without ghost endstops.

**Actual behavior:** What I mentioned in the steps part.

#### Additional Information
This includes configuration.h configuration_adv.h and pins_trigorilla_13.h:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2173962/Marlin.zip)
Video:
https://www.youtube.com/watch?v=jAUTN1cb_n8

GMagician, 2018-07-08:
Maybe you have unfiltered input and some electrical noise will send fake results.
You may try to solve enabling ENDSTOP_NOISE_FILTER in configuration.h.
If it solve you are in such situation and your hardware is weak to electrical noise. Solution is proposed directly in cnf file just above option. If you can't solve in hardware you must live with this option and all negative effects it has.

Don't know which endstop you have, if simple switch, you can try to connect com to "signal", NC to +5V and NO to GND, enable ENDSTOPPULLUP_XM??, but this depends on your knowledges. All these HW operations are intended to be done only if you know what you are doing, you may **destroy** your hardware

Murloc992, 2018-07-08:
My endstops are basic switches. I will try this option. The thing is, other endstops are the same for Y and dual Z endstops.

This is how they look like:
[image]

Pins are soldered, from the point of reference they are C NO NC.

There's also a LED connected to COM with a resistor.

Murloc992, 2018-07-08:
The noise filter has helped I think. Yet I don't understand what has changed in bugfix-1.1.x to have caused my X endstop to not work anymore?

When I flash older firmware I have (https://github.com/derhopp/Marlin-with-Anycubic-i3-Mega-TFT/tree/bugfix-1.1.x) it works without a budge with identical configuration file.

When I flash newer firmware back it does "ghost" endstop only on X...

GMagician, 2018-07-08:
For sure capacitor will filter noise. I think that your board doesn't have any (trigorilla I think is a ramps derived board).

Missing capacitor is a hardware design fault. If option has solved issue that capacitor will do the same (in a better way)

GMagician, 2018-07-08:
Note that you have inverting endstops to true so I think your board will close to 5V when micro is pressed. My solution was to close to GND

Murloc992, 2018-07-08:
My endstop wiring is like this now, looking at the PCB:
S goes to COM
G goes to NO
V goes through the LED and COM

GMagician, 2018-07-08:
Don't know how led is connected but maybe it's not good to change connection

Murloc992, 2018-07-08:
LED is connected between V and S. Would a 10uF capacitor do for a test or is it too much?

GMagician, 2018-07-08:
ok, so it's really as I told you.. when micro is pressed it close to GND and led will turn on.

I'm not smart in electronic but I think the bigger the capacitor is and stronger the filter will be. Too strong is not good, too few wil not filter enough.
There is a formula to find out filtered frequency.

Murloc992, 2018-07-08:
I looked into the wiring. X is the only one connected differently from the others.. It doesn't have the LED hooked up to anything. I bet it's one of the culprits. I might just try to guide a "hack" wire and adjust the connector to make it work like the others.

Murloc992, 2018-07-08:
I just added a 10nF capacitor for a test and it works. So if somebody could close this it would be nice. Huge thanks @GMagician !

GMagician, 2018-07-08:
You may close it by pressing butto below

Murloc992, 2018-07-08:
Woops :)

Murloc992, 2018-07-10:
No, adding 100nF capacitors does not fix the issue. It still halts.

I know the soldering job looks a bit shoddy, but multimeter says everything is as it should be.

[image]

Murloc992, 2018-07-10:
In fact it's only dual Z endstops that fail.  They do hit once, then slowly twice and printer halts itself.

GMagician, 2018-07-10:
Ok, just to be sure, now what happens? home should be X, Y and then Z. Before was X failing now is Z that fails?

GMagician, 2018-07-10:
What do you means with

> hits endstops again slowly. This is normal behavior with old firmware but this time it just fails when it triggers second time.

does/doesn't it stops?
or just it doesn't rise again?

Do you still have filter option enabled?

Murloc992, 2018-07-10:
Filter is disabled. It homes Z normally, it hits one time normal speed, one time slow speed and stops. After it stops it  halts.

GMagician, 2018-07-10:
Ok let's try with debug option and M111

Murloc992, 2018-07-10:
>  M111 S38
SENDING:M111 S38
echo:DEBUG:INFO,ERRORS
echo:busy: processing <<< homing Z happens here
Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

This is all I get.

Murloc992, 2018-07-10:
It printed me probably whole memory map of the board and then homed Z and halted. No extra info.

All I know is that it fails on endstops.cpp line 233 Endstops::validate_homing_move method.

yagona, 2018-07-10:
I also have a modified ANYCUBIC i3 Mega. (I replaced, e.g., all stepper drivers with TMC2208.) As long as I used derhopp's fork of Marlin 1.1.8 or bugfix-1.1.x everything was fine. Recently I merged derhopps's bugfix-1.1.x with the latest Marlin bugfix-1.1.x causing exactly the same problem with z homing as described by Murloc992. I also get the kill message as stated above. However, somehow I could get M111 S38 working with the following result:

LESEN: TFT Serial Command: A21 Z
LESEN: echo:enqueueing "G28 Z"
enqueueing "G28 Z"
LESEN: >>> G28
LESEN: Machine Type: Cartesian
LESEN: Probe: NONE
LESEN:   current_position=(-5.00, 0.00, 5.00) : setup_for_endstop_or_probe_move
LESEN: > endstops.enable(true)
LESEN: >>> homeaxis(Z)
LESEN: Home 1 Fast:
LESEN: >>> do_homing_move(Z, -307.50, [4.00])
LESEN:   current_position=(-5.00, 0.00, 0.00) : sync_plan_position
...
LESEN: <<< do_homing_move(Z)
LESEN: Move Away:
LESEN: >>> do_homing_move(Z, 2.00, [4.00])
LESEN:   current_position=(-5.00, 0.00, 0.00) : sync_plan_position
LESEN: echo:busy: processing
busy: processing
LESEN: <<< do_homing_move(Z)
LESEN: Home 2 Slow:
LESEN: >>> do_homing_move(Z, -4.00, 1.00)
LESEN:   current_position=(-5.00, 0.00, 0.00) : sync_plan_position
...
LESEN: <<< do_homing_move(Z)
LESEN: >>> do_homing_move(Z, 0.00, [4.00])
LESEN:   current_position=(-5.00, 0.00, 0.00) : sync_plan_position
LESEN: Error:Printer halted. kill() called!
LESEN: TFT Serial Debug: Kill command... J11

GMagician, 2018-07-10:
@yagona in your configuration do you have Z_DUAL_ENDSTOPS_ADJUSTMENT equal to 0?
@yagona and @Murloc992  try to set a different value for such parameter.

@thinkyhead and @ejtagle  I think that in some situation like this, after home is done, hit_on_purpose will clear `hit_state` variable and on dual stepper systems, with no offset between 2 steppers, `do_homing_move` will wrongly test at end if micro are hits and they are not (flag cleared before and not enough time elapsed to set them again in temperature isr). This will kill the system.

GMagician, 2018-07-10:
@yagona and @Murloc992 you can try also adding `endstops.update` call just before `endstops.validate_homing_move` call in `do_homing_move` function. This to temporary patch issue

Murloc992, 2018-07-10:
@GMagician Z_DUAL_ENDSTOPS_ADJUSTMENT is 0. I have tried to add endstops.update() call before line 3020 in marlin_main but nothing really changed.

What should I try with the adjustment? Positive or negative values?

yagona, 2018-07-10:
@GMagician I set Z_DUAL_ENDSTOPS_ADJUSTMENT to 0.025 by "M666 Z0.025". This seems to solve the issue. Output is now:

LESEN: TFT Serial Command: A21 Z
LESEN: echo:enqueueing "G28 Z"
enqueueing "G28 Z"
LESEN: >>> G28
LESEN: Machine Type: Cartesian
LESEN: Probe: NONE
LESEN:   current_position=(0.00, 0.00, 30.00) : setup_for_endstop_or_probe_move
LESEN: > endstops.enable(true)
LESEN: >>> homeaxis(Z)
LESEN: Home 1 Fast:
LESEN: >>> do_homing_move(Z, -307.50, [4.00])
LESEN:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
...
LESEN: <<< do_homing_move(Z)
LESEN: Move Away:
LESEN: >>> do_homing_move(Z, 2.00, [4.00])
LESEN:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
...
LESEN: <<< do_homing_move(Z)
LESEN: Home 2 Slow:
LESEN: >>> do_homing_move(Z, -4.00, 1.00)
LESEN:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
...
LESEN: <<< do_homing_move(Z)
LESEN: >>> do_homing_move(Z, 0.03, [4.00])
LESEN:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
LESEN: <<< do_homing_move(Z)
LESEN: >>> set_axis_is_at_home(Z)
LESEN: For Z axis:
LESEN:  home_offset = 0.00
LESEN:  position_shift = 0.00
LESEN:  soft_endstop_min = 0.00
LESEN:  soft_endstop_max = 205.00
LESEN: > home_offset[Z] = 0.00
LESEN:   current_position=(0.00, 0.00, 0.00) : 
LESEN: <<< set_axis_is_at_home(Z)
LESEN:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
LESEN:   current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
LESEN: <<< homeaxis(Z)
LESEN:   current_position=(0.00, 0.00, 0.00) : sync_plan_position
LESEN:   current_position=(0.00, 0.00, 0.00) : clean_up_after_endstop_or_probe_move
LESEN: X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
LESEN: <<< G28

Murloc992, 2018-07-10:
@GMagician @yagona interestingly only M666 worked. This solves the issue I guess? Kind of.

GMagician, 2018-07-10:
@Murloc992 value should be offset to align two steppers.
But I'm wondering if micro is pressed when such command is executed

GMagician, 2018-07-10:
@Murloc992 this is only a temporary patch. it's rare but M666 Z0 could be a possible value

Murloc992, 2018-07-10:
@GMagician both switches get pressed. I will look at this and see if any solution is proposed. :)

GMagician, 2018-07-10:
theoretically changing Z_DUAL_ENDSTOPS_ADJUSTMENT will move an axis and this will let `endstops.update` to be called. calling it before `endstops.validate_homing_move` should temporary fix as well, but it seems not. Maybe something is escaping from my check

yagona, 2018-07-10:
@GMagician I did some further testing according to your suggestions. The results are:
- "M666 Z0.025" suppresses the issue
- both microswitches get pressed (tested by misaligned Z1 and Z2 before homing)
- "Z_DUAL_ENDSTOPS_ADJUSTMENT at 0.025" suppresses the issue as well as M666
- adding "endstops.update" call does not seem to have any effect

GMagician, 2018-07-10:
ok, thanks, for the moment you have a solution (I hope you really need z adjustement to align Z1 with Z2). Then I hope someone with more knowledge will try to solve this

yagona, 2018-07-10:
@GMagician Thanks a lot for your help! Your suggestions were very useful to identify the probable cause of this issue.

I'd like to add that "M666 Z0.001" also suppresses the issue. This should be well below the resolution of the microswitches.

Murloc992, 2018-07-11:
M666 Z0.001 saves itself as Z0.00. It works fine. Maybe it's related to this "floating maths" fix?

GMagician, 2018-07-11:
Don't know but for sure also 0 should work. There is something somewhere that doesn't detect endstop status correctly and this will raise error. Changing value will put a movement in planner and there endstop are updated. Don't know more, maybe I'll try to investigate deeper but this goes beyond my limits

GMagician, 2018-07-11:
@Murloc992 found something. When stepper arte not moving `Endstops.update` doesn't update anything so basically doesn't workl

ejtagle, 2018-07-11:
As stated by @GMagician , M666 schedules a new movement to the queue. I don´t have a Delta, so i can´t test this, but, does M666 Z0 change the homing sequence at all ?

Endstops are updated 1000 times a second in that configuration (because there is no interrupt on endstop feature enabled) - So adding an extra movement should not make a difference

Your description of the issue is a little bit vague: Do you mean the printer halts and needs a reset to resume operation ? (a kill operation ?)

GMagician, 2018-07-11:
@ejtagle kill means it completely stop working with message "printer halted".
When movement is queued an endstops.update si called in planner and this wil set hit_state. After do_move hit_state is tested and if 0 kill

GMagician, 2018-07-11:
@ejtagle this is sequence that leat to issues:
 -last slow move to find micro
 -micro found, steppers stop and `endstops.validate_homing_move;` is called. This will test `hit_state`, that is ok, and reset it.
- do_home_moving is called to align steppers but since space=0 I think it will not insert anything into queue, but `endstops.validate_homing_move` is called and this time `hit_state` is 0 and printer is killed.

As told before by changing Z value with M666 if will put something into planner and since planner call `endstops.update` here 'hit_state' is updated and set != 0. Once `endstops.validate_homing_move` is called it doesn't fail.
I think that this issues is also present for all ?_DUAL when offset is 0

ejtagle, 2018-07-11:
The only way this could happen is if between the call to endstops.hit_on_purpose() and the call of endstops.validate_homing_move() less than 1 millisecond (and thus, not temperature ISR that updates the endstops) happen.
Adding a safedelay() call of about 5 milliseconds insde validate_homing_move() should fix this issue then. But not when using ENDSTOPS_INTERRUPTS ... As the endstop state is only updated on changes, and no changes are expected if no movements are done.
I maybe the logic is somehow flawed, as originally validate_homing_move() should only be called if there was a move!

GMagician, 2018-07-11:
> The only way this could happen is if between the call to endstops.hit_on_purpose() and the call of endstops.validate_homing_move()

I think this may be highly probable since no movement are involved, code may take less than 1ms to read failing call

> I maybe the logic is somehow flawed, as originally validate_homing_move() should only be called if there was a move!

`const bool is_home_dir = (axis_home_dir > 0) == (distance > 0)`

when distance is 0 may be also true and this will call failing function... it should be enough to not set this variable and everything will be ok... In such situation maybe also bl-bouch could be lowerd and this is not good

GMagician, 2018-07-11:
> Also, the call to stepper.quick_stop() can cause some problems... as that call just sets a flag to abort on the next step ISR. So, a small delay (1mS) could be needed.

or not to call `endstops.validate_homing_move` at all.
Note that as stated above, calling `do_homing_move` with distance = 0 will also deploy probe and this is not good. In my opinion to solve this should be enough:

`const bool is_home_dir = (axis_home_dir > 0) == (distance > 0) **&& distance**`

this will solve interrupt case, distance = 0 case and all other

ejtagle, 2018-07-11:
@GMagician : I remember when @thinkyhead added the `validate_homing_move()` call. It was to prevent the issue that when an endstop was not triggered, the homing procedure continued and that was a dangerous behaviour for deltas. If it was for me, i´d add an extra parameter fo the function, but, also changing the condition is fine.
Please, remember that `distance` is a float, so comparing with 0 will usually lead to errors. 
I am not sure on the is_home_dir condition. It is right the `axis_home_dir >0` , but the `distance == 0` ? 
Does it make sense to call do_homing_move with a distance of 0 ?
Maybe we should change this to

```
 const bool is_home_dir = (axis_home_dir > 0) == (distance > 0);
```

to
```
 const bool is_home_dir = (axis_home_dir > 0);
 ```

And at the start of the function:
```
if (UNEAR_ZERO(distance)) return;
```

That kind of strange comparison that is now there `(axis_home_dir > 0) == (distance > 0)` seems very strange

GMagician, 2018-07-11:
> I remember when @thinkyhead added the validate_homing_move() call. It was to prevent the issue that when an endstop was not triggered,

thats ok, but this code raise a "printer kill" and is not also good.

> distance is a float, so comparing with 0 will usually lead to errors.

That's true when value is derived from computation, 0 is a well defined value for FP IEEE so if  it comes from a #define then match will always succeed

Also test you made may lead to error since it should test something abs "below" MIN_STEPS_PER_SEGMENT converted to axis unit

ejtagle, 2018-07-11:
@GMagician : Completely agree with your comments 
We should somehow check if movement was queued or not. _populate_block already returns that information, and that information is returned by planner.buffer_line(), so, if it returns false, no movement was queued and no check should be done 👍

ejtagle, 2018-07-11:
But, if movement is too short, that function will also return true... :(

ruggb, 2018-07-13:
the problem with that endstop is that the signal line is connected thru an LED which has about 1.8V drop. You do not have endstop pullups enabled therefore the inactive voltage on the input pin as about 2.5 - 3V. That isn't a reliable off state with any amount of noise.
I recommend enabling the endstop pullups. If the endstops have been made more sensitive, that just means they have taken the anti-bounce logic out which is the cause of your problems. Instead of waiting for a solid low, it triggers on the first transition. The capacitors you added helped, but enabling the pullups should solve it completely.

My endstops have pullups and capacitors so I have the pullups disabled.

GMagician, 2018-07-13:
@ruggb as I know he has already remove filter and added pullup. Now problem is kill due to a bug in firmware

GMagician, 2018-07-13:
@Murloc992, @maar1201gh can you test posted code? it should fix issue but if someone can test it....remember to set offset to 0 and M502/M500 to reset saved value (or m666 just before home)

yagona, 2018-07-15:
#11261 fixed the homing issue on z for me. Thanks to everyone who contributed to this!

maar1201gh, 2018-07-15:
@GMagician , sorry for delay. I tested with #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0 and everything Ok.

Thanks

Murloc992, 2018-07-16:
@GMagician I was away for a bit, I will test this today and see what is up. :)

Murloc992, 2018-07-17:
Yes, it works perfectly!

Thanks @GMagician :)

closed 2018-07-17 by Murloc992

Seetth, 2018-08-25:
Sorry for re-open this again, I have the same issue but mine its a CR-10S, my ends stops are filtered with a cap since long time ago, I have no issues at all with marlin 1.8 but with 1.9 the printer almost always halt when  is doing Y homing, 1st do like a if was stuck advance a bit then go back a bit then halt the printer, only on Y axis aka the bed, so far I have found no solution for it thats why I ended here.

Thanks in advance.

ruggb, 2018-08-25:
mine did something like that when I forgot to change the polarity of the endstop input.

Seetth, 2018-08-26:
This is what the printer do when that happens, https://youtu.be/vjE2ugflItE 

@thinkyhead Far as I understand the ENDSTOP_NOISE_FILTER can affect the printer accuracy (im a noob so not everything its clear for me on many points) thats why im still refusing to use ENDSTOP_NOISE_FILTER option.

I use activate the linear advance option on 1.8 and I really like it now I can print really fast now w/o lose quality, I read that linear advance its better on 1.9 that its why im trying to make it run.

ruggb, 2018-08-26:
@Seetth I see no movement of the Z axis. X & Y appear to home properly. What am I missing?
Set your Z axis (vertical movement) mid point then run G28. What happens?

ruggb, 2018-08-26:
@Seetth I am having some difficulty understanding you. Sorry.
Your homing process, Z-lift 5mm then, Home X, Home Y, Home Z is normal.
Your video shows Y homing properly, so I am confused about which axis you are referring to.
I am assuming your X axis refers to the carriage position on the gantry; your Y axis refers to your bed's horizontal position front to rear; your Z axis refers to your bed's vertical position, or the vertical position of your gantry.

Using BLTouch as an endstop seems hard to do with consistent results but if you have it working, that is good. I am using a piezo triggered by my nozzle for ABL and endstop, which is awesome.

"(the switch for minimum position aka home im not sure if endstop its the right name)" is the same thing.

 Since the problem occurred when you went from 1.8 to 1.9, I am assuming the problem is in the transfer of your settings in the config files. However, bed level code has also changed and a few new variables added. So it might be that all your settings have been implemented but something new has been overlooked.

If you upload both your 1.8 and 1.9 configuration.h files I will compare them and see if I can determine if it is a config issue.

Seetth, 2018-08-27:
Sorry for the confusing video, I can confirm that the printer halt on Y axis randomly aka the bed and only in the bed, about the BLtouch as endstop at least for me become a really reliable endstop I never had troubles with it (im using the pseudo smart version) depends on what I doing I change the printer surface from tempered glass to borosilicate ( i think that is the name) to PEI to mirror thats why I attempt to use bltouch like that and im really happy with the results.

As soon I get home I upload both files.

Thanks for answer me.

[32 more replies]

PR #11260 (2018-07-14): [2.0.x] fix kill printer
This will fix kill message on dual stepper printers with 0 alignment offset

Fix #11224 and #11172