MarlinFirmware/Marlin #11227: TCI ultimaker Thermocouple AD595 PCB calculation incorrcet

moth4017, 2018-07-09:
See #11221

using the TCI AD595 PCB for a thermocouple in Marlin V2 with Re-arm  calculation incorrect.

thinkyhead, 2018-07-09:
The formulae for AVR readings of these thermocouples are…
```cpp
#define TEMP_AD595(RAW)  ((RAW) * 5.0 * 100.0 / 1024.0 / (OVERSAMPLENR) * (TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET)
#define TEMP_AD8495(RAW) ((RAW) * 6.6 * 100.0 / 1024.0 / (OVERSAMPLENR) * (TEMP_SENSOR_AD8495_GAIN) + TEMP_SENSOR_AD8495_OFFSET)
```
I presume that for the ReARM (and other boards) it's the 5.0, 6.6, `TEMP_SENSOR_AD*95_GAIN` and `TEMP_SENSOR_AD*95_OFFSET` values we need to modify. If you can determine the proper values, then it should be simple enough to modify them as appropriate for each MCU.

thinkyhead, 2018-11-05:
Anything actionable here? Do we need to get some thermocouples for testing? AD595, AD597, AD8495, PT100…?

p3p, 2018-11-05:
Any device that has a set amount of V per degree (rather than resistive) will not work on any of the 3.3V boards as the ADC reference is wrong so we will need to add the ability to adjust the calculations based of the ADCVref.

Although there is the problem that most of these devices are 5V in nature, and given enough temperature the signal will be more than 3.3V more than likely destroying the ADC.

moth4017, 2018-11-06:
Hi i still have my thermocouple set up  with the error i can do some testing .

craftoscar, 2018-11-07:
> Although there is the problem that most of these devices are 5V in nature, and given enough temperature the signal will be more than 3.3V more than likely destroying the ADC.

mostly, but AD8495 supply voltage can be as lowed as 2v 
I'm currently using two AD8495 with RAMPS-FD with Due

p3p, 2018-11-08:
@craftoscar The devices I meant were the boards that users have before they upgrade to 32bit that they will just plug into there new system, I know that the chips themselves don't necessarily have this issue. 

Requiring the platforms to define an ADCVRef to be used in the equations wouldn't be a problem I don't think, or just assuming ARM boards are 3.3v.

boelle, 2019-02-20:
@moth4017 tried with latest bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead