MarlinFirmware/Marlin #11229: BL-Touch v 2.0

Roxy-3D, 2018-07-09:
I have a few BL-Touch v 2.0 Z-Probes that I would like to get into Serious Marlin developer's hands.
I expect they will work identically...  except better than the original Classic BL-Touch Z-Probe.

If you are willing to use the new version of the probe and tweak the firmware if you see an issue...  Please post to that effect.

@Bob-the-Kuhn @thinkyhead  @p3p @fiveangle  @GMagician    I've already set one aside for each of you if you can use it.

RudyBenner, 2018-07-09:
I see nothing on the website https://www.antclabs.com/bltouch about this product.
My 3dtouch is working fine, although I catch a lot of flak for using a clone. It's the dark
coloured product, apparently from Geeektech. I assume the pinout is the same.

RudyBenner, 2018-07-09:
I am not a "Serious Marlin Developer", but would be happy to unplug my (ugh) 3dtouch and give it a go. MKS GEN L. Bugfix 1.1.x

dough29, 2018-07-09:
Do we have specs for this BLTouch v2 ?

@RudyBenner genuine BLTouch is, by far, more accurate than 3DTouch :+1:

RudyBenner, 2018-07-09:
Yes, I am interested now. 'far more accurate"? Seems information is yet scarce.

dough29, 2018-07-09:
I talk about BLTouch "smart" (latest known version) VS 3DTouch, I tested both and BLTouch is 4 times more accurate.

RudyBenner, 2018-07-09:
This one? https://spool3d.ca/bltouch-auto-bed-leveling-sensor/

dough29, 2018-07-09:
The one from Antclabs -> https://www.antclabs.com/store

Roxy-3D, 2018-07-09:
> Do we have specs for this BLTouch v2 ?

They sent one paper copy of the specifications.  It appears to be identical to this document:

https://docs.wixstatic.com/ugd/f5a1c8_77c6538efc934dbeab2f6e06e175ec35.pdf

Sineos, 2018-07-09:
From what I read the Smart v2 has a plastic touch pin (vs. metal on v1) with a rated break point to avoid damaging the solenoid in a crash and some improvements in the mounting of the magnetic coil.

Roxy-3D, 2018-07-09:
Yes...  The external portion of the pin is plastic.   But you can tell the internal portion of it is metal because you can feel the magnet pulling on it.

p3p, 2018-07-09:
@Roxy-3D If you have already set one aside for me who am I to refuse :wink:, thankyou, well only if the shipping isn't ridiculous, can send my address on slack?

GMagician, 2018-07-09:
@Roxy-3D well I'm still in doubt to remove my clone, not happy with it. As I can see my plate (glass) is almost plane and I'm currently printing without UBL.

Thanks for courtesy and to count me as a "Serious Marlin developer"

Roxy-3D, 2018-07-09:
> well only if the shipping isn't ridiculous, can send my address on slack?

Can you send a private message on Slack?   What I've been doing is asking people to send Roxy a private message on www.3dprintboard.com      The only down side is the person has to create an account to do that.

p3p, 2018-07-09:
> Can you send a private message on Slack?

Indeed you can, most of my conversations on there about Marlin development are private, but I can send via 3dprintboard if you prefer.

RudyBenner, 2018-07-09:
Opened up the 3dtouch. The metal pin goes in to a piece of what I presume is shrink tubing, along with a tiny magnet. No glue, loose fit. Dab of glue, put it back together. M48 I get SD of about 0.005

Bob-the-Kuhn, 2018-07-10:
I need help registering for 3dprintboard.  The **Human Verification Question:** is asking **Please enter the first letter and last letter of the word printing in the field above**.  I can't figure out what field it's talking about.  It's not the contents of the verification image.  

Sounds like there should be a single word scrolling/appearing nearby.

I've tried Microsoft Edge and Chrome.

Probably something obvious which is why I'm missing it.

Bob-the-Kuhn, 2018-07-10:
I finally figured it out.  The instructions should read  "In the field above, please enter the first letter and last letter of the word **printing**"

Roxy-3D, 2018-07-10:
I got the message with your address.   But there was no need for you to jump through those hoops.  I already had your address from the Re-ARM board.    

You are 'Good to Go!'

teemuatlut, 2018-07-10:
@Roxy-3D I've sent you a private message on Slack.

Roxy-3D, 2018-07-10:
@teemuatlut    I see the Slack message...    The only thing I'm asking is if you have a machine that it is easy to put it on and use.   If so...   You are welcome to it.      Antclabs wants to make sure their probes work well with Marlin.   So having active developers use their latest probe helps insure any quirks get ironed out and fixed.

teemuatlut, 2018-07-10:
"Easy" is always a bit relative but it shouldn't take much to design a new carriage/mount for any one of my printers. So yes I do think I can find a place for it.

Roxy-3D, 2018-07-10:
@teemuatlut   Please send me your mailing address....

teemuatlut, 2018-07-10:
Done on Slack.

RudyBenner, 2018-07-10:
Got any left over? I already have a 3dtouch, switching over should be trivial.

mylife4aiurr, 2018-07-10:
I have two of newest genuine bltouch versions with plastic tip. Have them running of my CR10s4 and a Delta printer. I'm not a pro coder but I can help test changes, or run test code. Plz merge #9310 is it directly effects leveling g33 and ubl

silentninja1, 2018-07-11:
Ive seen a few of these on formbot machines as well. I have 7 machines with BL Touch probes on them now, but havent gotten a plastic one on one of my machines yet. Mounting is the same, just the pin that changes.

mylife4aiurr, 2018-07-11:
Yes I've never had the older version. Just recently bought my 2 genuine bltouch's.
All the current code in marlin bugfix works. I've had no issues getting them to work.

On my CR10s4 I was able to significantly speed up G29 bed probing after watching this video:
https://youtu.be/BV11-VOQjMc
Much much Faster and still accurate.

On Delta the defaults in Marlin were not so hot. Standard probe speeds for Delta's are way too fast for bltouch. At standard speed the probe has zero chance to stop nozzle from slamming into bed. The effector is simply traveling to fast to stop in time. Also the standard slow probe speed is /2. That's also still too fast.
#define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z
// Speed for the "accurate" probe of each point
#define Z_PROBE_SPEED_SLOW (Z_PROBE_SPEED_FAST / 2)

For cartesian printers:
#define HOMING_FEEDRATE_XY (50*60)
#define HOMING_FEEDRATE_Z  (4*60)

For Delta's
Default is 200*60
That is ludricious speed as far as bltouch is concerned.

I ended up making my homing feedrate something like 100*60 or 80*60.
Slow probe something like /5.

I think for bltouch on Delta's the 1st probe speed should be way slower then 200*60
define Z_PROBE_SPEED_FAST HOMING_FEEDRATE_Z /3 or /4

tcm0116, 2018-07-11:
[image]

Sorry, couldn't resist 😁

mylife4aiurr, 2018-07-11:
@tcm0116..... That reference was intentional.... Glad u caught it.... I actually liked "Solo" star wars movie. It was decent

Roxy-3D, 2018-07-12:
The USA packages have gone out...    Now the struggle to figure out what it takes to ship to over sea locations.    I'm not looking forward to this part.

mylife4aiurr, 2018-07-16:
I had zero issues with my Cartesian printer setting up probe offsets for bltouch. For my Delta with the bltouch it hasnt been easy. #10129 had a similar issue. I think the main issue for me and possibly the user in #10129, was how probe offsets are depicted to user in instructions:

       +-- BACK ---+
       |           |
     L |    (+) P  | R <-- probe (20,20)
     E |           | I
     F | (-) N (+) | G <-- nozzle (10,10)
     T |           | H
       |    (-)    | T
       |           |
       O-- FRONT --+
     (0,0)
 
Fine for Cartesian.

Delta is different and a different diagram should be shown to user (idk how to make these diagrams, feel free to change for better a representation)
```
   +NW     -NE
    \       /
     \     /
      \   /
       \ /
      (0,0)
       N \
      /   \
     /     \
   P/       \
  +SW       -SE

example above Probe is 
#define X_PROBE_OFFSET_FROM_EXTRUDER   10.0  // X offset: -Southeast  +Northwest [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER   38.0  // Y offset: -Northeast  +Southwest [the nozzle]

```

Delta Bed Coordinate
```
Tower X+                                                       Tower Y-
       *                                                         *
            *                                              *
                 *                                   *
                       *                       *
                             *           *   
                                   *    (0,0)
                             *            *
                       *                       *
                 *                                    *
             *                                              *
        *                                                         *
Tower Y+                                                        Tower X-
```
Setting up the bltouch with G33 and G29 probe always traveled outside of printable area. I tried so many different combinations of probe offset X and Y. Different MIN_PROBE_EDGE values.....nothing worked. Because I was interpreting the diagram as shown. I was always gonna fail.
Using standard diagram my offsets should be
#define X_PROBE_OFFSET_FROM_EXTRUDER -38.00  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10.00   // Y offset: -front +behind [the nozzle]

Using alternative facts diagram:
#define X_PROBE_OFFSET_FROM_EXTRUDER   10.0  // X offset: -Southeast  +Northwest  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER   38.0  // Y offset: -Northeast +Southwest [the nozzle]

Using those values G29 ubl worked as expected, probe never tried to probe outside bed. min_probe_edge works. 
G33 still probes off defined print bed size, regardless of:
#define DELTA_PRINTABLE_RADIUS xx.x // mm
or
#define DELTA_CALIBRATION_RADIUS xx.xx // mm
Delta auto config probes is a circular pattern. The first instance follows closely printable radius and delta calibration radius. 2nd iteration is moves alittle farther out. But each newer iteration the probe moves a little farther out. Until it finds a deviation it likes. But as it searches for best deviation it doesn't obey the boundaries like UBL bed probing. I like to setup: #define Z_PROBE_LOW_POINT -1
for obvious reasons, its like a safety net. I cant use low point currently, because my pei sheet is 1mm higher then pritn bed. If I use low point on some iterations of g33 the bltouch will go off pei sheet and probe print bed. So because low point is set 1mm, G33 will fails, game over. I must reset printer. I could set a really low print size and delta raduis maybe? but I want as much printable space as possible not less.

There doesnt seems to be a hard limit that g33 cant go past. The limit exist for G29. 
Tested on bugfix 256f76d

Roxy-3D, 2018-07-17:
No...   You are over thinking it.    To get a Delta nozzle to the correct location involves a lot of math (and trig).   But the X_PROBE_OFFSET_FROM_EXTUDER & Y_PROBE_OFFSET_FROM_EXTRUDER numbers are very simple.   Pull out a ruler and measure how for the BL-Touch pin is from the extruder nozzle on each axis.   Do the measurement from center of the extruder nozzle hole to the center of the BL-Touch pin.

Those are the numbers you use.

mylife4aiurr, 2018-07-17:
Thanks for you response. Before I installed bltouch, I used nozzle as probe. Was electric contact nozzle to bed. No offsets, yay. I guess I just didnt understand the zoffset diagram as it relates to delta's, for the bltouch.

Roxy-3D, 2018-07-17:
It doesn't matter if it is a Delta, Scara or Cartesian.   The X_PROBE_OFFSET_FROM_EXTUDER & Y_PROBE_OFFSET_FROM_EXTRUDER numbers are the same.   They are the distance from the probe to the nozzle.

RudyBenner, 2018-07-17:
BLtouch just arrived, my 3dtouch (which I broke, my bad) was already taken out, so install was a breeze. Yes, it is better. I was getting, at best, and SD of about 0.005
Happy **Camper.**

>>> m48 p20 v4
SENDING:M48 P20 V4
M48 Z-Probe Repeatability Test
Positioning the probe...
Bed X: 110.000 Y: 110.000 Z: 0.006

1 of 20: z: 0.010 mean: 0.0100 sigma: 0.000000 min: 0.010 max: 0.010 range: 0.000
2 of 20: z: 0.016 mean: 0.0131 sigma: 0.003125 min: 0.010 max: 0.016 range: 0.006
3 of 20: z: 0.013 mean: 0.0129 sigma: 0.002569 min: 0.010 max: 0.016 range: 0.006
4 of 20: z: 0.010 mean: 0.0122 sigma: 0.002558 min: 0.010 max: 0.016 range: 0.006
5 of 20: z: 0.016 mean: 0.0130 sigma: 0.002806 min: 0.010 max: 0.016 range: 0.006
6 of 20: z: 0.016 mean: 0.0135 sigma: 0.002834 min: 0.010 max: 0.016 range: 0.006
7 of 20: z: 0.019 mean: 0.0143 sigma: 0.003194 min: 0.010 max: 0.019 range: 0.009
8 of 20: z: 0.016 mean: 0.0145 sigma: 0.003058 min: 0.010 max: 0.019 range: 0.009
9 of 20: z: 0.014 mean: 0.0144 sigma: 0.002893 min: 0.010 max: 0.019 range: 0.009
10 of 20: z: 0.016 mean: 0.0146 sigma: 0.002798 min: 0.010 max: 0.019 range: 0.009
11 of 20: z: 0.014 mean: 0.0145 sigma: 0.002680 min: 0.010 max: 0.019 range: 0.009
12 of 20: z: 0.016 mean: 0.0147 sigma: 0.002608 min: 0.010 max: 0.019 range: 0.009
13 of 20: z: 0.016 mean: 0.0148 sigma: 0.002540 min: 0.010 max: 0.019 range: 0.009
14 of 20: z: 0.016 mean: 0.0149 sigma: 0.002476 min: 0.010 max: 0.019 range: 0.009
15 of 20: z: 0.010 mean: 0.0146 sigma: 0.002687 min: 0.010 max: 0.019 range: 0.009
16 of 20: z: 0.014 mean: 0.0145 sigma: 0.002610 min: 0.010 max: 0.019 range: 0.009
17 of 20: z: 0.010 mean: 0.0143 sigma: 0.002747 min: 0.010 max: 0.019 range: 0.009
18 of 20: z: 0.008 mean: 0.0139 sigma: 0.003087 min: 0.008 max: 0.019 range: 0.011
19 of 20: z: 0.008 mean: 0.0136 sigma: 0.003326 min: 0.008 max: 0.019 range: 0.011
20 of 20: z: 0.010 mean: 0.0134 sigma: 0.003333 min: 0.008 max: 0.019 range: 0.011
Finished!
Mean: 0.013375 Min: 0.008 Max: 0.019 Range: 0.011
Standard Deviation: 0.003333

teemuatlut, 2018-07-31:
@Roxy-3D Any idea what's the current status on this?

tcm0116, 2018-07-31:
I got my Cartesian machine setup last night with the new BLTouch and the latest bugfix-1.1.x. I didn't have a lot of time to check it out, but I did build a mesh with UBL and checked a few points on the bed by just sliding a piece of paper between the nozzle and the bed. I found that the nozzle was too high at the back of the bed and to look at the front of the bed. I was having dominant issues with my previous setup (using a first generation BLTouch), so there must be something else going on. I'll keep looking at it and report my results.

Roxy-3D, 2018-07-31:
If you can...  try doing a G26 mesh validation pattern.   You should see that symptom across the entire bed.   (Front vs. Back)     Usually...   this problem happens when the gantry twists a little bit.   

But even so...   With the G26 pattern on the bed, you should be able to edit the mesh points up or down so you can get a good first layer.

tcm0116, 2018-08-01:
I can understand how the gantry twisting could cause issues on a non-leveled printer, but shouldn't the leveling account for that?

ManuelMcLure, 2018-08-01:
> I can understand how the gantry twisting could cause issues on a non-leveled printer, but shouldn't the leveling account for that?

If you use the nozzle as the leveling sensor (piezo sensor, for example) then yes. But for the most common case (twisted X axis) any Y offset between the nozzle and the sensor means that the twist will affect the position of the sensor either more or less (depending on if the sensor is further from or closer to the axis of the twist than the nozzle) than it affects the nozzle so the amount of correction calculated by the leveling system will be wrong.

Roxy-3D, 2018-08-01:
And...   It may not even appear as a 'twist'.   For example,   rubber wheels riding on a 20-20 or 40-40 beam do shift in height based on how the cable is pulling them.   The machine may be perfectly square.   But some how, the nozzle does change angle ever so slightly.    And the probe being offset from that...  changes more.

teemuatlut, 2018-08-13:
@Roxy-3D Any idea what's the current status on ~~this~~ the international shipping?

closed 2019-05-16 by thinkyhead