MarlinFirmware/Marlin #11234: Auto leveling does not work properly in 1.1.8

wuhao19920619, 2018-07-09:
I have tried all three auto leveling methods but none of them works correctly.
#define AUTO_BED_LEVELING_3POINT
#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR

No matter the right side of bed is lower or higher than the left side. After auto leveling the right side will always about 1mm lower than the left side. I have tested a lot and I think there should be no problem with my configuration. I guess there should be some problem about compensation rate or something?

//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
#define DEBUG_LEVELING_FEATURE

#if ENABLED(MESH_BED_LEVELING) || ENABLED(AUTO_BED_LEVELING_BILINEAR) || ENABLED(AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  //#define ENABLE_LEVELING_FADE_HEIGHT

  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVES
  #define LEVELED_SEGMENT_LENGTH 1.0 // (mm) Length of all segments (except the last one)

  /**
   * Enable the G26 Mesh Validation Pattern tool.
   */
  #define G26_MESH_VALIDATION   // Enable G26 mesh validation
  #if ENABLED(G26_MESH_VALIDATION)
    #define MESH_TEST_NOZZLE_SIZE     0.2   // (mm) Diameter of primary nozzle.
    #define MESH_TEST_LAYER_HEIGHT    0.1   // (mm) Default layer height for the G26 Mesh Validation Tool.
    #define MESH_TEST_HOTEND_TEMP   200.0   // (°C) Default nozzle temperature for the G26 Mesh Validation Tool.
    #define MESH_TEST_BED_TEMP       65.0   // (°C) Default bed temperature for the G26 Mesh Validation Tool.
  #endif

#endif

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 2
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 10
  #define RIGHT_PROBE_BED_POSITION 170
  #define FRONT_PROBE_BED_POSITION 10
  #define BACK_PROBE_BED_POSITION 170

  // The Z probe minimum outer margin (to validate G29 parameters).
  #define MIN_PROBE_EDGE 10

  // Probe along the Y axis, advancing X after each column
  //#define PROBE_Y_FIRST

  #if ENABLED(AUTO_BED_LEVELING_BILINEAR)

    // Beyond the probed grid, continue the implied tilt?
    // Default is to maintain the height of the nearest edge.
    #define EXTRAPOLATE_BEYOND_GRID

    //
[... 219 lines cut ...]
//#define Z_HOMING_HEIGHT 10  // (in mm) Minimal z height before homing (G28) for Z clearance above the bed, clamps, ...
                             // Be sure you have this distance over your Z_MAX_POS in case.

// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

// @section machine

// The size of the print bed
#define X_BED_SIZE 190

#define Y_BED_SIZE 190

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -62
#define Y_MIN_POS -66
#define Z_MIN_POS 0
//#define X_MAX_POS (X_BED_SIZE)
//#define Y_MAX_POS (Y_BED_SIZE)
#define X_MAX_POS (300 + X_MIN_POS)
#define Y_MAX_POS (300 + Y_MIN_POS)
#define Z_MAX_POS 320

RudyBenner, 2018-07-09:
Add M420 S1 to your start script, or M420 S1 Z4. This is corrected in bugfix 1.1.x The M420 must come AFTER your last G28.

wuhao19920619, 2018-07-09:
I tried M420 S1 but still the same.
The method I test is bilinear.
 
This is the record of Repetier
`21:53:07.174 : N23 G28*34
21:53:24.678 : X:122.00 Y:32.00 Z:8.14 E:0.00 Count A:30800 B:18000 Z:6512
21:53:30.350 : N24 M420 S1*81
21:53:30.350 : echo:Bed Leveling On
21:53:30.350 : X:122.00 Y:32.00 Z:8.40 E:0.00 Count A:30800 B:18000 Z:6512
21:53:36.416 : N25 G29*37
21:54:00.101 : Bilinear Leveling Grid:
21:54:00.101 : 0      1
21:54:00.101 : 0 +0.113 -0.467
21:54:00.101 : 1 +0.369 -0.278
21:54:00.117 : X:197.00 Y:107.00 Z:8.55 E:0.00 Count A:60800 B:18000 Z:6512
`

The right side still 1mm higher then the left.

BTW which bugfix version you mean? The current testing one 1.1.8 and I don't find any newer version.

wuhao19920619, 2018-07-09:
I just test to run g29 first. The right is 1mm higher.  Then I manually level it, and run G29 again. and now the bed is prefect level. But after I power off and power on again, run g29. the right is 1mm higher again.

gloomyandy, 2018-07-09:
bugfix-1.1.x has the latest fixes.

You really need to fill in the issue template when reporting problems and also attach your configuration files as requested in the template. It will make it much easier for people to help you.

insanity67, 2018-07-10:
Please search the issues here, there are so many with the same problem. Some say it´s a mechanical issue, others dont believe that.

nightgryphon, 2018-07-10:
Did you save final mesh with M500? May be it use the older mesh after load settings.
Also you can try to check actual bed level before run G29 to check the mechanical issues. Home with G28 than G29 S5 to disable mesh than move head to control points with G1 commands to see the actual bed level by distance between bed and nozzle at same Z height value.

AnHardt, 2018-07-10:
All kinds of bed levelings, using a probe, don't try to keep the nozzle at a constant distance to the bed. They keep the probes trigger point at a constant distance to the bed. This works well only, if the distance of the nozzle tip and the probes trigger point are the same all over the bed. If `Z_PROBE_OFFSET_FROM_EXTRUDER` isn't constant there is no chance to make bed levelling with auto probing work - you have a mechanical problem.
A workaround is to use UBL and to dial in the mesh points manually, maybe with help of G26. An other one, for all kinds of leveling, is to use the nozzle as the probe (PROBE_MANUALLY, ...).

wuhao19920619, 2018-07-11:
I have updated to the newest bugfix-1.1.x and then I have tried to level the bed manually first, then run the G28 and then G29. after that I test the leveling result by G1 and the right side is 1mm higher. After that I tried to lower the right side by 1mm and run g28,g29 and test with g1 again. The right side is still 1mm higher. I have done a couple of test with different bed leveling condition before G28. The result is no matter if bed is level or not before ABL. No matter the right side is higher or lower before ABL. after the ABL the right side will always 1mm higher than left side. Also the probe is fixed on the extruder. Therefore I am pretty sure Z_PROBE_OFFSET_FROM_EXTRUDER is constant. Due to this constant difference so I think this might not be a mechanical issue. The ABL methods I used are 3=point, bilinear and linear, They all give the same result.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2182530/config.zip)

RudyBenner, 2018-07-11:
Is this a core xy machine?
#define COREXY
are you using bltouch?

Roxy-3D, 2018-07-11:
> I have tried all three auto leveling methods but none of them works correctly.
```
#define AUTO_BED_LEVELING_3POINT
#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
```
There are currently 5 different bed leveling systems in Marlin.   

> A workaround is to use UBL and to dial in the mesh points manually, maybe with help of G26. An other one, for all kinds of leveling, is to use the nozzle as the probe (PROBE_MANUALLY, ...).

Yes!   Once again Anhardt is out in front.   If you are having problems...   Go to the basics.    What ever the mesh looks like when you probed it...    Now do a G26.   (Even if you are using Bi-Linear or the original Mesh bed leveling)     Print the G26 Mesh Validation Pattern.

Does it look right?   If not...   Start editing the mesh points up or down....    And re-print the G26 Mesh Validation Pattern.   Probably the pattern will look better but not be perfect.    Keep doing this 2 or 3 times.   Pretty soon...   Your mesh will accurately describe the bed (at least as far as the printer sees it).

If you are curious...    Re-probe the bed and compare the 'working' mesh with the probed mesh.  Probably, you will see a pattern.   And that might give you some hints on where to look for the root cause of the problem.   But even without doing that,   if you have a working mesh, you can just use it and not go after the root cause of the problem.  (Me???   I would go after the root cause and fix it!!!)

AnHardt, 2018-07-11:
> Therefore I am pretty sure Z_PROBE_OFFSET_FROM_EXTRUDER is constant. 

Don't believe in that. Measure it. With 1mm difference from right to left. it should be pretty easy to see the error by naked eye. Just take a look on the nozzle-bed distance on both sides of the bed when the probe triggers and compare.
Finding the reason could be a bit more difficult, because the huge (Y_PROBE_OFFSET_FROM_EXTRUDER 63) lever is amplifying the reason.

> Due to this constant difference so I think this might not be a mechanical issue.

All this proves is, that it's a systematic error, independent from the kind of bed level compensation. So a error in measurement is as likely as before.

megavolt0, 2019-02-18:
>  I have figured out the problem, @Roxy-3D and @AnHardt is right, My x-axis use two axis and I find they are not prefect parallel by a counter weight with a string. Thanks again for all your guys help.

Hi! I don't undersend... Have you solved the problem? I have the same problem on H-BOT with BLtouch... And I don`t know what to do....((

> Bilinear Leveling Grid:
> 0 1 2
> 0 +0.265 -0.510 -1.128
> 1 +0.237 -0.502 -1.195
> 2 +0.331 -0.425 -1.152

AnHardt, 2019-02-18:
Our "bed leveling"- algorithms try to keep the distance between **bed** and **nozzle** constant.
For that we try to find the Z position of the **bed** at several places with a **probe** and  
assume the distance between the **probe** and the **nozzle** is constant.
Now we try to keep the distance **probe** to **bed** constant.
If the distance **probe** **nozzle** changes, depending on where (XY) we are on the bed, we can keep the **probes** distance to the **bed** as exact as we want. The distance **nozzle** **bed** will vary.
There are some mechanical problems a printer can have what can cause this error. 

We can't find this problem for you because it's not in the software. It's up to you to find the mechanical problem..
One of the most common is a "twisted X-axis".