MarlinFirmware/Marlin #11235: just an explanation update

adolfie01, 2018-07-09:
hi

thanks for some really cool software, 

if you could just update the following description to the Config.h file. 

#define Z_PROBE_LOW_POINT          -5 // Farthest distance below the trigger-point to go before stopping

// say something like, "this is the point where the Z position was found during homing.  if you have an uneven bed and there are probing points that are below the homed Z point, set this find the lower point. otherwise the probe just sits there and does nothing because it is restricted by the Z home 0. 

took some time for me to figure it out.  :) 

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

closed 2018-11-05 by thinkyhead

stevenascott, 2019-06-06:
Wow. I finally stumbled on this thread and fixed my problem. Based on the original variable description, I thought this was more of a "how much further down will z-axis move after triggering on EACH probe." but yeah, it's more like, "maximum allowed deviation across entire bed."

Major problem in my eyes. Thank you so much for posting this here.

adolfie01, 2019-06-06:
Hi only with pleasure that is what communitys are about!

Virtualight, 2019-08-30:
Z_PROBE_LOW_POINT currently = Z_PROBE_LOW_POINT - Z_PROBE_OFFSET_FROM EXTRUDER. This requires one to know the Z_PROBE_OFFSET_FROM EXTRUDER value before setting Z_PROBE_LOW_POINT. This means if the value of Z_PROBE_OFFSET_FROM EXTRUDER is changed then you have to alter the Z_PROBE_LOW_POINT by the same amount which is not intuitive.

So the current "farthest" definition correctly defines Z_PROBE_LOW_POINT = Z_PROBE_LOW_POINT - Z_PROBE_OFFSET_FROM_EXTRUDER. I suggest that the better way to apply this offset in the negative direction is to change the word "Farthest" to "Extra"' distance below the trigger-point to go before stopping. The ofset would then ba applied in the background logic  Z_PROBE_LOW_POINT = Z_PROBE_LOW_POINT  **+** Z_PROBE_OFFSET_FROM EXTRUDER.

thinkyhead, 2019-09-03:
The distance is not extra. It is the limit.

Virtualight, 2019-09-04:
> The distance is not extra. It is the limit.
Yes I know - I am suggesting it should be 'extra' not the limit or farthest.

Roxy-3D, 2019-09-04:
Just to be clear...Z_PROBE_OFFSET_FROM_EXTRUDER was one of the very first constants in Configuration.h with the very first 3 point bed leveling system.    Everything after that has respected that definition and used it to do way more.   

We are not going to change that definition on a whim.

Virtualight, 2019-09-04:
> Just to be clear...Z_PROBE_OFFSET_FROM_EXTRUDER was one of the very first constants in Configuration.h with the very first 3 point bed leveling system. Everything after that has respected that definition and used it to do way more.
> 
> We are not going to change that definition on a whim. I re-read everything you wrote and I realize you never suggested that!

@Roxy-3D - I never suggested we change the Z_PROBE_OFFSET_FROM_EXTRUDER. If you carfully read my  post again I am suggesting that the definition of Z_PROBE_LOW_POINT be changed.

Virtualight, 2019-09-04:
@thinkyhead and @Roxy-3D - the left diagram on is the current definition of Z_PROBE_LOW_POINT and the right diagram is the suggested definition.

[image]

thinkyhead, 2019-09-04:
Your diagram is quite confusing. How is the red line the bed height, when the micro-switch is below that red line? Shouldn't the "microswitch trigger" line be labeled as the "bed height"? When the micro-switch is at the "bed height" the nozzle will be "above the bed" and thus always at a positive and non-zero Z position.

---

The `Z_PROBE_LOW_POINT` value indicates the lowest Z point in the native space that the probe will try to move to during probing. The `Z_PROBE_LOW_POINT` value is conveniently relative to the bed height. So if `Z_PROBE_LOW_POINT` is set to "0" probing will stop with the probe tip (or sensing field) exactly at the bed height (sometimes failing to trigger).

The `Z_PROBE_LOW_POINT` value is not "the Z point that was found during homing," (a phrase which itself is vague).

Note that probing during `G28`,`G29`/`G30`/`M48` is not constrained by `Z_MIN_POS`. I don't see what you are trying to indicate with an unusual `MANUAL_Z_HOME_POS` of -2 and a `Z_MIN_POS` of +2. You would never want to use that combination of settings. `Z_MIN_POS` should never be greater than 0 in normal usage.

edwilliams16, 2019-09-05:
I think @Virtualight 's figure is too ambitious and ends up muddling together several different concepts that could be illuminated by some separate figures.

(1)  The homing process establishes a coordinate system locating the tip of the printer nozzle in x/y/z.  In particular, it establishes a z=0 plane in nominally at the bed surface (which may not be flat).  The end result is determined differently if using a probe or a microswitch and also depends on the values of Z_MIN_POS and MANUAL_Z_HOME.  I can try to make some figures to clarify this.

(2) If we use the probe for anything, since the coordinate system references the nozzle, not the probe, we need to know the offsets of the probe from the nozzle either to manipulate its position or conversely to use its trigger to locate the nozzle in z.

(3) Z_PROBE_LOW_POINT is the minimum z-coordinate to which the probe trigger point is lowered.  In an ideal world, where the nominal bed location is z=0 (so we can print on it), it's relative to the nominal bed height - so it's a measure of the maximum dip in the bed we anticipate.

(4) To move the probe trigger point to Z_PROBE_LOW_POINT, we have to tell the nozzle where to go - to Z_PROBE_LOW_POINT - Z_PROBE_OFFSET_FROM_EXTRUDER.  So as it says in (2), that can't be done unless we already measured and set Z_PROBE_OFFSET_FROM_EXTRUDER.

  A simple diagram shows the relation of the z=0 nominal bed-level, the nozzle location and the probe.  This is shown in @Virtualight 's figure if you delete other stuff.

(5) When we probe during the bed-leveling process, we obtain a table entry - the z-coordinate of the bed surface at the x/y position of the probe (i.e. the x/y position of the nozzle corrected by the x/y probe offset)  - namely the local height of the bed relative to the z=0 nominal bed surface.

(6) During printing, the z-height of the bed is obtained by interpolating into the table at the x/y location of the nozzle.  This is used to offset the z-height of the nozzle to follow the dips and swells of the bed.  

(7) The soft end-stop at Z_MIN_POS (if enabled) is applied to the _uncorrected_ slicer z, allowing printing below z=0 (strictly Z_MIN_POS) in the dips of the bed.

  I suspect you may think all this is obvious (if you think anything is incorrect - please tell me!), but it takes some time to figure out.  If you think I have it right, I'm willing to draft something on the above lines and add some figures.

Virtualight, 2019-09-05:
@thinkyhead - LOL the diagram was meant to clarify not confuse...

The microswitch is just the homing reference and doesn't need to be at bed level although a lot of people have traditionally placed it at this point. The homing microswitch is just there as a failsafe as I and many others have had issues with probes not triggering on homing and crashing into the bed. 

I understand what Z_PROBE_LOW_POINT is.  The point that @adolfie01, @edwilliams16 and I am trying to define Z_PROBE_LOW_POINT you need to first know the value for Z_PROBE_OFFSET_FROM_EXTRUDER and, furthermore, if Z_PROBE_OFFSET_FROM_EXTRUDER changes then you need to change the value for Z_PROBE_LOW_POINT. IMO Making Z_PROBE_LOW_POINT an 'extra' distance is much easier to implement and way less confusing.

Virtualight, 2019-09-05:
@edwilliams16 - thanks for your input. That all seems to make sense - happy for you to draft up an alternative diagram to mine or mark up my diagram.

edwilliams16, 2019-09-05:
@Virtualight  I am actually unclear by what you mean by "extra distance".  Beyond what? Do you mean "additional distance beyond where you probed and made contact last time"?  If so, what happens on the first probe?  If you don't know Z_PROBE_OFFSET_FROM_EXTRUDER you are inevitably probing blind.

Virtualight, 2019-09-05:
@edwilliams16 - Extra is exactly as you suggested

> "additional distance beyond where you probed and made contact last time"?

I agree you are essentially putting in a value not knowing what the actual low or high points are. Leaving Z_PROBE_LOW_POINT at 0 risks you hitting a high point with the nozzle especially if your Z_PROBE_OFFSET_FROM EXTRUDER were a small distance.

edwilliams16, 2019-09-05:
I'm not following...  The current definition makes perfect sense if you make your setup so that z=0 is some point on the bed.  You have an idiosyncratic setup (deriving from not measuring Z_PROBE_OFFSET_FROM EXTRUDER -before- starting bed leveling and homing with the microswitch not the probe) where the bed is on average 1.6 mm from z=0.  This confuses everybody. They expect that you could print at least in some small region on your bed with leveling turned off.  You can't.

>especially if your Z_PROBE_OFFSET_FROM EXTRUDER were a small distance.

 Surely you mount the BLTouch so that Z_PROBE_OFFSET_FROM EXTRUDER is as large (in magnitude) as possible given the constraint that it stows safely higher than the nozzle. Making it small gives you less room to maneuver.

Roxy-3D, 2019-09-05:
> They expect that you could print at least in some small region on your bed with leveling turned off. You can't.

At:  https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_features/unified_bed_leveling.md

In the Synopsis it says:

> You should be able to successfully print a small object at the center of the bed with bed leveling turned off. It's very important to verify that your Configuration.h settings make this possible before trying to bring up UBL. Most problems bringing up the UBL Bed Leveling system occur when this step has been ignored. Please pay particular attention to your Z_PROBE_OFFSET_FROM_EXTRUDER value. Usually it's best to home the Z-Axis in the center of the bed. But wherever you decide to home, the Z value reported on the LCD (or with M114) should be very close to 0.0 mm when the nozzle is just touching the bed. Failure to calibrate Z_PROBE_OFFSET_FROM_EXTRUDER properly will result in dimensional errors in your printed parts.

InsanityAutomation, 2019-09-06:
The theory is you calibrate that with M851 and leveling the corners first, then go back and populate the full mesh. That said if you understand what its doing and know how to shift the mesh that isnt necessary but were also not going to suggest that to the average user :)

Roxy-3D, 2019-09-06:
I'm OK with just:

> You should be able to successfully print a small object at the center of the bed with bed leveling turned off. It's very important to verify that your Configuration.h settings make this possible before trying to bring up UBL.

The bed leveling can correct some really bad things.   But you have to get it into the game.

edwilliams16, 2019-09-06:
Here's the procedure I proposed to @Virtualight  by email. Please pick holes in it if you find any.

Home z
G1 to center of bed
M401  - deploy probe
Lower z until probe triggers.  Note z
M402  - stow probe
Turn off the soft end stops
Lower the nozzle until it just touches the bed by the paper test
Note the z     

The difference between the values of z is Z_PROBE_OFFSET_FROM_EXTRUDER
Use M851 to adjust it to its correct value. Save to EPROM etc.

Now lets get z=0 at the surface of your bed at the probe point. Then we could print in the center of our bed without needing leveling as Roxy3D would like. We want the soft end stop at the bed, ie at z=0 so set Z_MIN_POS 0 and initially MANUAL_Z_HOME 0

Physically adjust your microswitch so that it triggers before the nozzle touches the center of the bed. (I presume it does already!)
G1 to center of bed
Home z   — should show z=0 on the LCD (if it bumps up, G1 down to z=0)
Turn off soft end stops
Lower the nozzle until it just touches the bed by the paper test
Note the z   —  we want this to be 0, to make it so we need to change MANUAL_Z_HOME to minus this value and reflash.

The two procedures are basically the same.  The first measures the distance from the nozzle to the probe trigger point, the second between the bed and the microswitch trigger point.
These two measurements become Z_PROBE_OFFSET_FROM_EXTRUDER and MANUAL_Z_HOME and tell the code the geometry of your machine.

  If you check that this all works like I think it should, I can write a logical document with figures detailing the procedure.  (But write down your present settings first!)

InsanityAutomation, 2019-09-06:
> 
> 
> Here's the procedure I proposed to @Virtualight by email. Please pick holes in it if you find any.
> 
> Home z
> G1 to center of bed
If homing with the probe, G1 to home position, otherwise manually perform level corners before this step.

> Now lets get z=0 at the surface of your bed at the probe point. Then we could print in the center of our bed without needing leveling as @Roxy3D would like. We want the soft end stop at the bed, ie at z=0 so set Z_MIN_POS 0 and initially MANUAL_Z_HOME 0

If both 0, MANUAL_Z_HOME can stay disabled

> 
> Physically adjust your microswitch so that it triggers before the nozzle touches the center of the bed. (I presume it does already!)
Do not adjust this after having set M851 or you will be redoing it

> G1 to center of bed
> Home z — should show z=0 on the LCD (if it bumps up, G1 down to z=0)
> Turn off soft end stops
> Lower the nozzle until it just touches the bed by the paper test
> Note the z — we want [...]

Should reverse these two procedures, get the home point established before adding an offset from it.

edwilliams16, 2019-09-06:
@InsanityAutomation 
 You bring up some good issues.
 However, I don't think it is _necessary_ to reverse the order of the two procedures.  Measuring Z_PROBE_OFFSET_FROM_EXTRUDER takes the difference of two z values, so the location of the origin cancels out.  For a BLTouch, it's a set and forget parameter once it is mounted. However, as @AnHardt has pointed out elsewhere, it can require re-measurement if using a remote sensing probe (capacitive, inductive etc.) if you change the nature of the bed surface.  

Whether using the BLTouch or a microswitch for homing, it does seem to be desirable to mechanically four corner level the (uneven) bed before playing with the homing. This will reduce the overall tilt and reduce the peak-to-valley correction from bed-leveling. It will make the top and bottom surfaces of a test-cube closer to parallel.

I'm having some additional thoughts about the home adjustment. @Virtualight 's objective in using a microswitch for homing rather than the BLTouch is to protect against probe failure during homing and having a bed crash. Suppose he has the microswitch trigger 2mm above the bed and leaves MANUAL_Z_HOME 0.  After probing, even with a flat bed, he will get a leveling correction table full of -2mm entries.
  What happens when he prints? Will that 2mm be faded away up to the fade height, so that printed objects will end up 2mm too tall?   If this is so, it really is more desirable to adjust MANUAL_Z_HOME so that the mean bed-leveling correction is ~0, so that the mean height of printed objects ends up correct. @Roxy-3D I suspect you know the answer to this.

Roxy-3D, 2019-09-06:
In a perfect world...   The mesh should be centered around 0.00 mm.    Most people prefer to print in the center of the bed.    So homing in the center and getting those areas of the mesh to be very close to 0.00 mm usually makes sense.

With that said...   Mesh bed leveling will affect the dimensional accuracy of your part.   If you are printing in an area with a positive mesh value, the part will be too short.   Conversely, if you are printing in an area with a negative mesh value, the part will get stretched downward and the total height will end up too long.

If you turn off the fade height adjustment (and do Z-Height Correction) for the entire height of the part, your dimensional accuracy will be correct at any (X,Y) location.    But the top (and bottom) surfaces will not be smooth (or flat).   Instead, the imperfections of the bed will be propagated to both of those surfaces of the part.

edwilliams16, 2019-09-06:
@Roxy-3D   Thanks for confirming what I expected.  It might be nice to add the mean-height to any mesh printout. That would reduce the labor of figuring out a good offset. For large objects it might make sense to make the mean offset zero.  For small ones, print at the center of the bed with the offset zero there and get good dimensional accuracy.  I'd be inclined to make the latter the default, but it really depends on the application in mind.  I think one could use an M206 in the startup script to switch between the options.  This would require computing the difference between the height of the bed at center and its mean height.
@Virtualight

edwilliams16, 2019-09-12:
@Roxy-3D 
@Virtualight  has been using ABL, not UBL, so I didn't notice you had the ability to translate the mesh with G29 P5/P6 commands.  ABL it seems can only edit point by point. With 100+ points, one would need a script generator.
 With his huge bed, Catmull-Rom interpolation is helpful - I've made some plots of the difference between Catmull-Rom and straight bilinear and it is significant enough to be worth using.
  Any plans to build it into UBL,  and fixing the current bug(s)? https://github.com/MarlinFirmware/Marlin/issues/14963

Roxy-3D, 2019-09-12:
> Virtualight has been using ABL, not UBL, so I didn't notice you had the ability to translate the mesh with G29 P5/P6 commands. ABL it seems can only edit point by point. With 100+ points, one would need a script generator.

You will like the G29 P4 mesh point editor!   And, it is designed to work with a G26 mesh validation pattern still on the bed.