MarlinFirmware/Marlin #11236: Error "DEFAULT_MAX_ACCELERATION" on Raspbian & Marlin 1.1.8

didtoknan, 2018-07-09:
Hi,

The same issue has been reported but it has been unfortunately closed with no solution: https://github.com/MarlinFirmware/Marlin/issues/7850

When I'm compiling on my desktop (Win10) I have no error but my printers are not in the same room.
That's why I use Arduino on Raspbian to compile. 

The error I get:
SanityCheck.h:1307:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 static_assert(COUNT(sanity_arr_3) <= XYZE_N, "DEFAULT_MAX_ACCELERATION has too many elements.");
              
Please help !

melvinisken, 2018-07-10:
Additionally: you don't need to compile on the Pi. Install OctoPrint with the Firmware-Updater-Plugin on the Pi, compile Marlin on your desktop and upload+flash the resulting .hex file using the Updater-Plugin. Works perfectly for me since months and dozens of flashes.

didtoknan, 2018-07-14:
First, thank you for your help.

fiveangle, I don't think it's a Marlin issue but I think there is more chance than Marlin users can help me.

I will check both VSCode and the Firmware-Updater-Plugin.

closed 2018-08-22 by thinkyhead