MarlinFirmware/Marlin #11237: Z-axis raises while homing Z while Z-axis moves in right direction with pronterface. BL-Touch gets in alarm mode

alehanderoo, 2018-07-10:
### Description
I have had a Anet A6 for over a year but wanted to fully rebuild the printer with almost all new hardware(except the motors) to a Anet A6 Evolution Cube. 

This time I decided to go with a RAMPS1.5 and the new TMC2130 drivers and a Geeetech TL-Touch 3D-Touch sensor: https://www.banggood.com/Geeetech-Auto-Bed-Leveling-Sensor-With-High-Compatibility-For-3D-Printer-p-1154826.html?rmmds=myorder&cur_warehouse=CN

With some trial-and-error I could Deploy, retract and test the sensor as suggested in #7024. 
My hardware is all connected properly checked with the "M43 I" Gcode command and looks as it should. 
Checked my config like #5705 and looks okay.

The X and Y axis are homing perfectly but I am now breaking my head for 2 days on the Z homing.. 
When I home all axis with G28 It ends up with probing and going up with the distance of Z_CLEARANCE_DEPLOY_PROBE for 2 times and then stops..

There is something with this sensor as well.. It runs into alarm mode (orange LED flashing) on every power cycle or reset. The sensor is powered true a separate step down converter (5V)  that is powered on when 12V power is supplied. 

### In depth

Pronterface output with M280 P0 testcommands on startup:
(I do need to reset it first to be able to get it in test-mode..)
> M119
< Reporting endstop status
< x_min: open
< y_min: open
< z_min: TRIGGERED
> M280 P0 S160
> M280 P0 S10
> M280 P0 S60
> M119
< Reporting endstop status
< x_min: open
< y_min: open
< z_min: open
> M119
< Reporting endstop status
< x_min: open
< y_min: open
< z_min: TRIGGERED
 
G28 With debugging OFF:
> G28
< echo:busy: processing
< echo:busy: processing
< echo:busy: processing
< echo:busy: processing
< Error:STOP called because of BLTouch error - restart with M999
< Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
< echo:busy: processing
< Error:STOP called because of BLTouch error - restart with M999
< echo:busy: processing
< X:80.00 Y:130.00 Z:2.00 E:0.00 Count X:6400 Y:10400 Z:8000
> M119
< Reporting endstop status
< x_min: open
< y_min: open
< z_min: TRIGGERED

Now the G28 with debugging ON:
> M280 P0 S160
> M111 S32
< echo:DEBUG:LEVELING
> G28
< >>> G28
< Machine Type: Cartesian
< Probe: BLTOUCH
< Probe Offset X:30 Y:-20 Z:0.00 (Right-Front & Same Z as Nozzle)
< Auto Bed Leveling: BILINEAR (disabled)
< set_bltouch_deployed(0)
<   current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
< > endstops.enable(true)
< Raise Z (before homing) to 2.00
< >>> do_blocking_move_to(0.00, 0.00, 2.00)
< <<< do_blocking_move_to
< >>> homeaxis(X)
< Home 1 Fast:
< >>> do_homing_move(X, -330.00, [50.00])
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
< echo:busy: processing
< <<< do_homing_move(X)
< Move Away:
< >>> do_homing_move(X, 5.00, [50.00])
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
< echo:busy: processing
< <<< do_homing_move(X)
< Home 2 Slow:
< >>> do_homing_move(X, -10.00, 25.00)
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
< <<< do_homing_move(X)
< >>> set_axis_is_at_home(X)
< For X axis:
<  home_offset = 0.00
<  position_shift = 0.00
<  soft_endstop_min = 0.00
<  soft_endstop_max = 220.00
< > home_offset[X] = 0.00
<   current_position=(0.00, 0.00, 2.00) :
< <<< set_axis_is_at_home(X)
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
<   current_position=(0.00, 0.00, 2.00) : > AFTER set_axis_is_at_home
< <<< homeaxis(X)
< >>> homeaxis(Y)
< Home 1 Fast:
< >>> do_homing_move(Y, -330.00, [50.00])
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
< echo:busy: processing
< <<< do_homing_move(Y)
< Move Away:
< >>> do_homing_move(Y, 5.00, [50.00])
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
< <<< do_homing_move(Y)
< Home 2 Slow:
< >>> do_homing_move(Y, -10.00, 25.00)
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
< <<< do_homing_move(Y)
< >>> set_axis_is_at_home(Y)
< For Y axis:
<  home_offset = 0.00
<  position_shift = 0.00
<  soft_endstop_min = 0.00
<  soft_endstop_max = 220.00
< > home_offset[Y] = 0.00
<   current_position=(0.00, 0.00, 2.00) :
< <<< set_axis_is_at_home(Y)
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
<   current_position=(0.00, 0.00, 2.00) : > AFTER set_axis_is_at_home
< <<< homeaxis(Y)
< Z_SAFE_HOMING >>>
<   current_position=(0.00, 0.00, 2.00) : sync_plan_position
<   destination=(80.00, 130.00, 2.00) : Z_SAFE_HOMING
< >>> do_blocking_move_to(80.00, 130.00, 2.00)
< echo:busy: processing
< <<< do_blocking_move_to
< >>> homeaxis(Z)
<   current_position=(80.00, 130.00, 2.00) : set_probe_deployed
< deploy: 1
< do_probe_raise(2.00)
< >>> do_blocking_move_to(80.00, 130.00, 2.00)
< <<< do_blocking_move_to
< Home 1 Fast:
< >>> do_homing_move(Z, -375.00, [4.00])
< Error:STOP called because of BLTouch error - restart with M999
< Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
<   current_position=(80.00, 130.00, 0.00) : sync_plan_position
< echo:busy: processing
< set_bltouch_deployed(0)
< <<< do_homing_move(Z)
< Move Away:
< >>> do_homing_move(Z, 2.00, 4.00)
<   current_position=(80.00, 130.00, 0.00) : sync_plan_position
< <<< do_homing_move(Z)
< Home 2 Slow:
< >>> do_homing_move(Z, -4.00, 2.00)
< Error:STOP called because of BLTouch error - restart with M999
<   current_position=(80.00, 130.00, 0.00) : sync_plan_position
< echo:busy: processing
< set_bltouch_deployed(0)
< <<< do_homing_move(Z)
< >>> set_axis_is_at_home(Z)
< For Z axis:
<  home_offset = 0.00
<  position_shift = 0.00
<  soft_endstop_min = 0.00
<  soft_endstop_max = 250.00
< *** Z HOMED WITH PROBE (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) ***
< > zprobe_zoffset = 0.00
< > home_offset[Z] = 0.00
<   current_position=(80.00, 130.00, 0.00) :
< <<< set_axis_is_at_home(Z)
<   current_position=(80.00, 130.00, 0.00) : sync_plan_position
<   current_position=(80.00, 130.00, 0.00) : > AFTER set_axis_is_at_home
<   current_position=(80.00, 130.00, 0.00) : set_probe_deployed
< deploy: 0
< do_probe_raise(2.00)
< >>> do_blocking_move_to(80.00, 130.00, 2.00)
< <<< do_blocking_move_to
< >>> do_blocking_move_to(80.00, 130.00, 2.00)
< <<< do_blocking_move_to
< <<< homeaxis(Z)
< <<< Z_SAFE_HOMING
<   current_position=(80.00, 130.00, 2.00) : sync_plan_position
<   current_position=(80.00, 130.00, 2.00) : clean_up_after_endstop_or_probe_move
< X:80.00 Y:130.00 Z:2.00 E:0.00 Count X:6400 Y:10400 Z:8000
< <<< G28

Probe says it is triggerd.. 
> M119
< Reporting endstop status
< x_min: open
< y_min: open
< z_min: TRIGGERED

#### Additional Information

With the self-test "M280 P0 S120" it goes straight into an error..

Config and pins files:
[github.zip](https://github.com/MarlinFirmware/Marlin/files/2180937/github.zip)

GMagician, 2018-07-10:
> It runs into alarm mode (orange LED flashing) on every power cycle or reset. 

This means it fails to execute selftest. I may suggest to check bltouch top screw if in correct position (on manuals you will find how to adjust it)

alehanderoo, 2018-07-10:
@GMagician the sensor itself is working allright in my findings. The Z axis is moving up when it starts homing the Z axis without deploying the sensor pin. 
The printer homes X and Y then it moves up, deploys the pin, retracts it and then moves up again repeating these steps from right after the Y homing.

I am looking into the code tonight where this all is happening. I'm finding it hard to believe it's a bug. Maybe someone in this great community can have spot something I'm missing in these days.

GMagician, 2018-07-10:
> Antclabs says also to insert the M280 P0 S160

in newer version when G28 is called bl-touch is already reset and stowed so M280 is not required anymore.

> I can only find The Z axis is moving up when it starts homing the Z axis without deploying 

when bl-touch is in error it signal "triggered" this may be the cause of your 'moving up'

closed 2018-11-05 by thinkyhead