MarlinFirmware/Marlin #11239: Rumba + lcd conflict

IvanLev, 2018-07-10:
### Description

Marlin doesn't work on RUMBA when reprapdiscount smart lcd enabled. Also somehow blocks the uart0 so I have to reflash the bootloader to change the firmware. Additionaly the led1 on the board starts t blink like crazy.
tried 1.1.8 and 1.1.4. Everything works when lcd disabled. 

#### Confiuration files
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/2181872/conf.zip)

Grogyan, 2018-07-10:
RUMBA is known to require one of the LCD connectors to be mounted in reverse

IvanLev, 2018-07-10:
Only one of them? I reversed both of them as I read about it before
[image]

thinkyhead, 2018-11-05:
Solved?

IvanLev, 2018-11-05:
> Solved?
No, using the printer without the LCD but would be really happy if you could help get it working.

thinkyhead, 2018-11-07:
It looks like the plugs are backwards or reversed. Sometimes the boards or the displays ship with the key slot on the wrong side of the pins. I suggest you cut out a slot so you can flip the display plugs and try that.

IvanLev, 2018-11-08:
> It looks like the plugs are backwards or reversed. Sometimes the boards or the displays ship with the key slot on the wrong side of the pins. I suggest you cut out a slot so you can flip the display plugs and try that.

I did this by taking the plastic shrouds off and rotating them. The issue is in marlin somewhere as it just refuses to work.
MK4duo too BTW, but it confirms that the display connection is correct as after 5 minutes of booting it does display its bootscreen on the lcd and even after hours nothing else.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead