MarlinFirmware/Marlin #11243: 2.0.x auto bed level not saving and restoring on restart

Black6spdZ, 2018-07-11:
on latest 2.0.x bugfix with MKS sbase hardware, simmilar to issue #5996 , abl data not saving after M500 and available after a reset. Here is a command window showing what I mean

>>> G29
SENDING:G29
Bilinear Leveling Grid:
      0      1      2
 0 +0.003 +0.172 +0.074
 1 -0.104 +0.107 +0.060
 2 -0.091 +0.160 +0.123
>>> M500
SENDING:M500
>>> M503
SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X394.00 Y394.00 Z3175.00 E890.00
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z4.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
**echo:Auto Bed Leveling:
echo:  M420 S1 Z0.00
echo:  G29 W I0 J0 Z0.00302
echo:  G29 W I1 J0 Z0.17216
echo:  G29 W I2 J0 Z0.07364
echo:  G29 W I0 J1 Z-0.10369
echo:  G29 W I1 J1 Z0.10746
echo:  G29 W I2 J1 Z0.05984
echo:  G29 W I0 J2 Z-0.09090
echo:  G29 W I1 J2 Z0.16000
echo:  G29 W I2 J2 Z0.12283**
echo:Material heatup parameters:
echo:  M145 S0 H190 B60 F255
echo:  M145 S1 H220 B90 F255
echo:PID settings:
echo:  M301 P25.10 I1.70 D92.83
echo:  M304 P485.10 I61.82 D951.65
echo:Z-Probe Offset (mm): (mm):
echo:  M851 Z-1.60
-------------------------hard reset board here ----------------------
Disconnected.
Connecting...
Printer is now online.
Using tool 0.
echo:Active Extruder: 0
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> M501
SENDING:M501
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 T1 D1.75
echo:  M200 D0
echo:Steps per unit:
echo:  M92 X394.00 Y394.00 Z3175.00 E890.00
echo:Maximum feedrates (units/s):
echo:  M203 X200.00 Y200.00 Z4.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000 Y3000 Z100 E10000
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P3000.00 R3000.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000 S0.00 T0.00 X10.00 Y10.00 Z0.30 E5.00
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:Material heatup parameters:
echo:  M145 S0 H190 B60 F255
echo:  M145 S1 H220 B90 F255
echo:PID settings:
echo:  M301 P25.10 I1.70 D92.83
echo:  M304 P485.10 I61.82 D951.65
echo:Z-Probe Offset (mm): (mm):
echo:  M851 Z-1.60
>>> M420 S1
SENDING:M420 S1
Error:Failed to enable Bed Leveling
[ERROR] Error:Failed to enable Bed Leveling

echo:Bed Leveling Off
echo:Fade Height Off

gloomyandy, 2018-07-11:
It would probably help if you post your config files. With an sbase the "eeprom" settings are stored to the sdcard, did you have an sdcard in the "drive" on the sbase when you ran the tests? Are any values being saved to "eeprom"? Might be worth changing some other settings (like acceleration or something away from the default and see if that gets saved and restored).

Black6spdZ, 2018-07-11:
I didn't see anything saved on the sdcard but come to think of it I had this exact same problem with smoothieware as well, the bed cal wouldn't save to the sdcard when it was plugged into the PC via USB and the sd mounted.. I'm thinking the same may be true here. I will try later this evening when I get home and report back. Otherwise this firmware is looking great and much improved over the competition

Black6spdZ, 2018-07-12:
ran bed leveling from lcd disconnected from pc.. afterwards no extra file for level data on sd card.. still no abl data on restet

Black6spdZ, 2018-07-12:
ooops, very noob mistake.. think I found the issue and will verify when I get home, this was commented out in config.h    #define EEPROM_SETTINGS // Enable for M500 and M501 commands

closed 2018-07-13 by Black6spdZ