MarlinFirmware/Marlin #11247: [HAL] Anyone porting Marlin to Open-Source RISC-V?

marcio-ao, 2018-07-11:
I recently came across this:

  https://www.sifive.com/products/hifive1

I am curious whether anyone has looked into porting Marlin to this? It would be interesting to see a printer that used an open-source core. The specs of the board look impressive (320MHz, 128 Mbit Flash) and surprisingly affordable ($60). It does lack A2D pins, however, so I guess you would need a SPI based A2D chip for reading thermistors.

-- Marcio

marcio-ao, 2018-07-12:
Bueller, Bueller?

Roxy-3D, 2018-07-12:
I'm all for us adding another board to the list!   But we need somebody to actually do the work.
It would be really cool to get support for this board (and processor).

marcio-ao, 2018-07-13:
Maybe I can persuade people here to take on the project. It's certainly in line with our company's open-source philosophy.

Roxy-3D, 2018-07-13:
The way we got the 32-Bit effort launched was Panucatt helped seed the effort with 8 or 10 Re-ARM boards.   That gave us a 'Critical Mass' of developers that had a 'Reference Platform' to work against.  We now have a beach head that works.   We are not where we need to be.   But we are dug in on the beach of 32-bit land and we have a stable code base that we can start fanning out from.   (BIG THANKYOU TO Panucatt!!!)

My suggestion is seed the community with boards.   Work with the company to get 6 or 8 units (including the Analog support pieces) in a package and we probably can find compiler experts that would be willing to open another front in this 32-bit effort.

marcio-ao, 2018-07-13:
@Roxy-3D : Okay. I'll bring it up and see if there is an interest in doing this!

Nice to know we have Panucatt to thank for 32-bit support, we are thankful because our next printer will be using a 32-bit Archim board!

Roxy-3D, 2018-07-14:
It is actually a good deal for a company to do this.   For a few hundred dollars of hardware, they get ten's of thousands of dollars of engineer time (and expertise) to move Marlin to the board.   And at that point, they get to sell into that market.     Also...   Because the developers have that exact board, users they sell the board to get good support.