MarlinFirmware/Marlin #11250: [FR] Let 'level corners' option always be visible and let this do the homing procedure instead

MathijsG, 2018-07-12:
### Description

Lately I missed the level corners option on my Marlin CR-10. And somehow I found out that it randomly shows up. Later on I discovered that more people where seem to be missing this option, and I've helped them saying that they should need to home the thing first. Which leads me to the proposed Feature Request: Always show the option to level corners in the menu, and just let the procedure home the thing. Now people are often stuck missing this item randomly (perceived as random). I was wondering what the possible downsides would be, and couldn't really think of any.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

closed 2018-11-05 by thinkyhead

MathijsG, 2018-11-05:
Why is this closed? This is still actual.

thinkyhead, 2018-11-05:
The option is always visible in the `bugfix-2.0.x` branch. Marlin 1.1.x is end-of-life.