MarlinFirmware/Marlin #11252: IDEX UBL_G29 scope issue on tool change command

silentninja1, 2018-07-12:
### Description

Compile error with UBL and Duel x carriage in UBL_G29 with current bugfix

### Steps to Reproduce
Uncomment UBL and dual x carriage
Line 297 of UBL_G29.h has the following : 

    // Check for commands that require the printer to be homed
    if (may_move) {
      if (axis_unhomed_error()) home_all_axes();
      #if ENABLED(DUAL_X_CARRIAGE)
        if (active_extruder != 0) tool_change(0);
      #endif
    }

However home_all_axes is overloaded to a blank function in ubl.h then Tool change is out of scope as well. Other abl methods have this step in marlin_main under a g29 in progress conditional, but I didnt see a branch there for UBL. Im not near my Trex 2+ or any other machines today to confirm, but can I enqueue a command here with enqueue_and_echo_commands_P(PSTR("G28")) to home and enqueue_and_echo_commands_P(PSTR("T0")) to change tools, or is the planner blocked at this stage?

silentninja1, 2018-07-13:
Confirmed today this doesnt work, added the header to ubl.h as the home all axes command was under external references and it did not call the tool change, it just rammed the x1 carriage into the x2 carriage till it bumped its home switch then went into the g29 s1 action. Added the enqueue and echo commands and it executed after the G29, which meant t1 was moving, not t0 therefore it came down and rammed the nozzle into the bed then failed probing a bunch of times. I also tried adding the extern flag to the functions in marlin_main to no avail

thinkyhead, 2018-07-19:
The compile issue is now patched, but the actual tool change to carriage 0 has not yet been tested.

InsanityAutomation, 2018-07-19:
Sounds good, I ran out of time to mess with it, but I'll get back to it Saturday most likely and I'll give it a shot. Hopefully I'll have time to look at the excessive movement off the mesh area too, but we'll see. 

Didn't even think about just moving the prototype to .h when the external flag didn't work lol probably need to do the same for the home command right above where this is, since it winds up calling an empty function as is.

Roxy-3D, 2018-07-19:
> Hopefully I'll have time to look at the excessive movement off the mesh area too, but we'll see.

I was under the impression that the 'excessive movement off the mesh' was corrected 6 weeks ago.  Is that not true?   Can you post your configuration.h files and a G29 T W please?

The code hasn't been written yet...  But there was consensus that we should have several options for Z-Height correction when off the mesh.  One option is it should be 0.0000 mm.  Another option is it should be what ever the closest mesh point is.   And one more thought was it should be an extropolation of the bed topology near the 'off the bed' point.

InsanityAutomation, 2018-07-19:
Roxy, Chris from Tinymachines had a user report it on a build synced July 7th, so first goal is to reproduce it here! Configuration is here, both the raptor and cr10 branches have had reports.

https://github.com/InsanityAutomation/Marlin/tree/TM_Raptor

Roxy-3D, 2018-07-19:
> Chris from Tinymachines had a user report it on a build synced July 7th, so first goal is to reproduce it here!

Yeah...   I think Chris is trying to capture a machine that exhibits the problem.   Having a machine that demonstrates the problem makes it much easier to figure out and fix.

silentninja1, 2018-07-21:
It would seem that following a G28 when T1 was active, the tool change will properly park the head for T0 after homing, however the active stepper is still T1. This occurs any time you home with T1 active, not just related to UBL.

silentninja1, 2018-07-21:
Darn this rabbit hole! LOL So from terminal, bootup , T1 then G28. Move X axis by LCD and T0 moves while T1 position changes. From bootup T1 then G28, M91 G1 X5 T1 moves as expected. Seems to be a manual move issue with selected extruder as well. LCD menu item does show select E1 / E2 as expected when tool change is issued, just still moves the wrong extruder.

silentninja1, 2018-07-21:
Ok, lcd side is working see pull request lcd idex xmove #11328

Probably something similar lurking around g28 with how it works called from the ubl code, but for the moment the race condition is avoided with this

idex crash fix #11329

Ill do a bit more testing if it still moves the wrong axis in any cases I can find tomorrow most likely, im out for the day!

Roxy-3D, 2018-07-21:
@silentninja1     I may be able to help you get closer to the root cause of the problem.   It is my belief that this problem started when some of the methods were moved into the planner object.    Specifically, I think there is an issue with regard to enabling and disabling Z-Height correction based on the location of the nozzle.

For UBL...  The way the motion is done, I'm not convinced the correction needs to be un-done for the current tool and then re-done for the new tool.   Please keep that thought in mind as you search for the cause.

silentninja1, 2018-07-21:
@Roxy-3D Sounds good, I didnt get very far on that today (assuming in reference to the z raise / lower off the mesh)  between sitting on a VPN with a PLC system and getting sidetracked on the wrong gantry moving (then again when it did it from the LCD for a totally different reason!). Hoping to get back after that one tomorrow, especially since I have plenty of stuff at home to mess with it on. Only idex stuff is in my office at the moment.

silentninja1, 2018-07-21:
@Roxy-3D What I submitted today handles all the known dual extruder specific stuff to an extent. Shouldnt have needed to reverse command order in the second commit, but it does the trick (and probably cleaner, since then G28 isnt switching from 1 to 0, back to 1, just for UBL to come back to 0...). If something else is still lurking in there, I cant reproduce it from the terminal. I could try those commands back to back in a custom menu item and see if i can get it to show the issue though. Thats for another weekend since at least now no damage will be caused.

Roxy-3D, 2018-07-21:
And you tested tool changes after G28 with UBL too, right?

silentninja1, 2018-07-21:
Yeah, ive used that for quite awhile. Didnt really have any major issue with it until the tool change code was added and it didnt compile, then started down this rabbit hole! I do need to sit down and do more extensive step by step testing sometime. I havent intentionally tipped the bed and ensured both nozzles followed correctly yet. I did get successful dual extrusion and duplicate mode prints with ubl on. The duplicate mode I actually expected to cause more problems than it did, but I was close enough to level on my own it came out fine.

Roxy-3D, 2018-07-23:
I just changed the configuration of my printer to have two Tools.   (Even though it really only has 1 extruder)    From what I can see...  Everything is behaving properly right now.

silentninja1, 2018-07-23:
@Roxy-3D with or without to 2 commits I referenced? The issue is which stepper was pulsed. It will show the position changing for the right gantry, but the left is whats actually moving. In the case of the tool change from ubl_g29.h, the left head actually rammed into the right head trying to reach the right home before I swapped command order.

Roxy-3D, 2018-07-23:
It is with the current Bugfix_1.1.x branch.    I was getting setup to add different off mesh Z-Height correction options.    And I thought before I actually did that....  I would try to get a handle on the Tool change issues.

@thinkyhead   did a couple of changes to the Tool Change in the last few days.   I suspect he was trying to address what you are seeing.     But I have no way to address a multiple gantry machine.   I just configured my machine to have multiple nozzles (with the 2nd one being offset) to see if it did anything reasonable.    And it does.

silentninja1, 2018-07-23:
Yeah, the issues i'm looking at are exclusive to dual x carriage machines right now. They dont change anything outside of that.

The tool change fix he put in was in reference to what was originally discussed here where there was a no-compile scenario as the function was out of scope. Problem came into play that this function being called there then caused a crash, so more fixing is needed beyond the initial patch to at least make it compile.

Chimera type setup, you need to have the leveling just right regardless of UBL or the second nozzle will destroy the part as things adjust. Personally, im not sure of a good safe way to do ubl with a chimera system. Cyclops / mixing, or switching, sure. Idex, awesome.

silentninja1, 2018-07-24:
Ok, thought id get some clarity to what im doing since you dont have any dual x carriage machines. ubl_g29.h 

Before : 
https://www.youtube.com/watch?v=dPRzWsvN40I

After : 
https://www.youtube.com/watch?v=FYGcARM--fY

UltraLCD.cpp change : 
Before : 
https://www.youtube.com/watch?v=1HihDTCni2Y

After : 
https://www.youtube.com/watch?v=MLnv37BTwXs

silentninja1, 2018-07-24:
@Roxy-3D It happens with the rapid tool changes caused when the machine is not homed going into G29 P1. It does not occur if you are homed first. It does not occur if you are on T0 first. Only if you are on T1 and not homed. By changing the order so that you activate T0 before homing, you avoid calling tool change multiple times in a matter of milliseconds. 

The menu issue is separate, but should be pretty self explanatory there as well. You expect X2 to move, but X1 moves instead, while the position for X2 changes with it.

silentninja1, 2018-07-24:
@Roxy-3D https://github.com/MarlinFirmware/Marlin/pull/11329 This is what I did to get it to stop crashing as shown in the video for UBL G29 P1. Conditions are pretty specific to make it occur though.

Roxy-3D, 2018-07-24:
OK!    I'm comfortable merging #11329 even though I can't test it.     I'll make those same changes over in bugfix_2.0.0

silentninja1, 2018-07-24:
@Roxy-3D Sounds good. Did you look at #11328 at all? Thats the other scenario with the LCD. I did find depending on white spacing, if {} blocks worked better than #if #endif so I might just swap it to those quick. I found one scenario with all my echo lines that it landed on the wrong set of lines, probably from a typo putting it in and out lol

Roxy-3D, 2018-07-24:
No...  I haven't looked at that one...    But let me try to get the bugfix_2.0.0 version of the previous one compiling and merged.

silentninja1, 2018-07-24:
@Roxy-3D 

[TM_TREXConfig_20180724.zip](https://github.com/MarlinFirmware/Marlin/files/2225213/TM_TREXConfig_20180724.zip)

Nothing too crazy fancy in this one

silentninja1, 2018-07-24:
@Roxy-3D That was on my plan for my next open weekend. Maker Faire here this weekend, sorry that wins over getting a 2.0 config! If I end up with some down time while im tuning a vision inspection system at a tier 1 supplier tomorrow, ill poke at it and see if I can get that finished. Build environment for 2.0 with platform io and atom is quite a bit different.

Roxy-3D, 2018-07-24:
Yeah... OK!    But I have the G29.cpp change in there.   And it might be compiling clean.
So...   At least the two code bases are synchronized.

Roxy-3D, 2018-07-25:
>  can, but editing is annoying since 1.9.0 beta doesn't show the whole folder tree. I can edit the config and compile without any problem though.

Yeah...   I guess it depends on how you do your work.   I only use Arduino to compile and link.   I use an external editor so that issue doesn't bother me.

silentninja1, 2018-07-25:
I get annoyed switching between programs. One of these days I should see about getting eclipse configured for arduino... Been my editor of choice for a long time on other stuff and has git integration.

silentninja1, 2018-07-25:
BTW, Just noticed these couple questions : 

> Do both of the gantry's on that machine have a probe? 
No, just T0, left head

> But either way... right now, it looks like the probe_pt() function is moving the wrong gantry on your machine.

Probe pt always moved the right one, Only the tool change had the wrong one moving, and if you avoid that by either homing first or ensuring T0 was selected first, it always ran fine.

If you were working on off mesh compensation options, hopefully thatll make it in before I sit down for another round of testing. I dont mind messing with an early version of what youre working with if you want more feedback on it.

Roxy-3D, 2018-07-25:
It will take a while for any improvements to off mesh options.   Please don't wait for me!

silentninja1, 2018-07-25:
Progress waits for no man! Ill stare over your virtual shoulder till its done! Just kidding, I may focus attention on other things in the meantime. Need to circle back to some of the laser stuff I started awhile back, and I have a coreXY machine ive been working on. At a minimum ill verify a fresh clean build pulled from git with my config files works as intended before I move on so we can close this item.

InsanityAutomation, 2018-07-26:
Btw, Chris ran across more idex crash scenarios with just g28 and tool change commands. I lobbed a couple hail Mary fixes at him while I was at a customer site today, I'll be back in my office Friday with the Trex to track it down a bit more. Probably the root cause of the issue we avoided with the change in ubl_g29.h

Roxy-3D, 2018-07-26:
I sent Chris a message via Google Hangouts saying:

> Even a very small dual gantry printer would be very helpful to resolving issues.  And that would enable me to make necessary changes to the G26 code and G29 P4 code.   It is my belief it might be  beneficial to have a separate mesh for each extruder.

Currently, G29 (and for that matter G29 P1) only envision a single nozzle/bed mapping.   But obviously, that isn't accurate with multiple nozzles and especially with dual gantry machines.

If you want to hand off of this issue...  I'll get it fixed ASAP if I have a test platform to code against.

silentninja1, 2018-07-26:
@Roxy-3D They ride on the same rail, just a parallel belt pulling them. If there was a separate probe, I could see another mesh, otherwise the common procedure is to set the 2 nozzle heights together with a single feeler gauge under both. The Trex uses chimera heat breaks on each side so they slide up and down.

I don't necessarily need to hand it off, but I wont complain if you get time first. My time in front of the machine is a bit limited for the next week. Id rather see you focus on the off mesh stuff since that's causing more user issues than this uncommon scenario.

Also, the trex is the only dual gantry machine Chris carries. He does have one in his shop which im sure you could take a look at if you had time to stop over one day.

Roxy-3D, 2018-07-26:
OK.   Good to know.

silentninja1, 2018-07-27:
@Roxy-3D Found it, ill have another pull request in shortly.

silentninja1, 2018-07-27:
@thinkyhead  @Roxy-3D in testing with a clean clone, found the newest bltouch commit ( #11358 ) is causing my machine to ram the nozzle into the bed and never deploy the probe. Im going to look into that as well while I get this pull request together, since I cant test it on a clean copy without fixing that first.

silentninja1, 2018-07-27:
Current also has another homing issue, where the bltouch probe is deployed and an extra raise is triggered, and it thinks its at 1.5mm when its really somewhere around 20mm. Ill dig into this one too.

silentninja1, 2018-07-27:
Fixed - see pull request #11393. Not sure what travis doesnt like is #11392 Let me know if you see an issue with it.

silentninja1, 2018-07-27:
@Roxy-3D I moved it down to the dual x carriage conditional and added the original command in the else below it. Travis is happy now :)

Roxy-3D, 2018-07-27:
@silentninja1   We are trying to keep the bugfix_1.1.x branch and bugfix_2.0.0 branch in lock step...   Can you be troubled to make an identical Pull Request for bugfix_2.0.0 ?     And I'll merge them at the same time.

silentninja1, 2018-07-27:
Yeah, should I put all of the ones I just did into a single pull req for 2.0?

Roxy-3D, 2018-07-27:
Yes...    Pretty much, the logic will be identical.    But because of the hierarchical file layout of bugfix_2.0.0, declarations and such may end up being a little bit different.

silentninja1, 2018-07-27:
@Roxy-3D #11395 - travis is checking it now,

InsanityAutomation, 2018-07-28:
@Roxy-3D gmagician pushed a more thorough change for the bltouch homing issue. I'll get a pair of commits with just the idex issue setup tomorrow.

silentninja1, 2018-07-28:
@Roxy-3D Ended up back on my laptop tonight, so I modified #11392 and #11395 to have just the idex fix.

Roxy-3D, 2018-07-28:
Thank You!   I'll review them in the morning.

InsanityAutomation, 2018-07-29:
@Roxy-3d perhaps if that caused some issue with a different machine, maybe we can check if active extruder position plus tool clearance is beyond the soft limit. In that case we know a move can't occur. Do you know which machines had an issue and I can look at how they x carriages are designed?

InsanityAutomation, 2018-07-29:
@Roxy-3D other option if they actually park in the same position and don't need an offset, aside from offset being 0, would be a config option for extruder can park overlapping to use the old functionality. Either way, need some kind of an idea what problem it causes on what machine because it crashes pretty violently on formbot machines as is.

silentninja1, 2018-07-30:
@Roxy-3D Can you run this suggestion by them : 

We currently have : 
  #define X2_MIN_POS    
  #define X2_MAX_POS

I can add the same for X1 and enforce them in the tool change command. If that wont cause a problem for them, it should be pretty universally configurable.

Roxy-3D, 2018-07-30:
Will do....       

Thank You for thinking about this!

Roxy-3D, 2018-07-30:
@silentninja1    

> I can add the same for X1 and enforce them in the tool change command. If that wont cause a problem for them, it should be pretty universally configurable.

I think this will 'fix' things...   Can we give that a try?

InsanityAutomation, 2018-07-30:
Yep I should have time to sit down and write that if not tonight, tomorrow sometime. The root functionality should be pretty simple but we may also want to take it a step further and look at soft and stops and overwrite those based on active tool as well.

InsanityAutomation, 2018-07-31:
Since the 1.1 branch is now closed, it'll take me another day or two to get a working configuration for 2. 0 on my machine here to actually test what I'm putting in. I have the stuff for the tool change script done, I finished that last night for the 1.1 branch, and was planning to get the soft stop stuff pieced together today, but since that's not going to get into 1. 1 I'll just leave it and move to 2. 0 before I continue.

Roxy-3D, 2018-07-31:
Bug fixes can still be done to 1.1.   Let's focus on 2.0.0.    But if the changes are easy to understand and low risk...   they can still be put into 1.1.

InsanityAutomation, 2018-07-31:
Yeah nothing here is going to be crazy LOL all pretty simple. I've got the code in where it works, just need to get this off stops enforced on tool change then add the new configuration options into the sanity check logic. if you want I can put in a pull request when I get to sit back down in front of my laptop with just the bare bones functioning part of it to have those two manufacturers review if it causes issues for them still then add in the rest of the stuff to make it complete if they agree with the function.

Roxy-3D, 2018-07-31:
Let's not make extra work for ourselves.    Just get it polished and tested...   And we can ask people to test and comment at that point.

silentninja1, 2018-08-01:
@Roxy-3D Was stuck working late last night on the day job, but tested and working this morning. For the sanity checks, I set it to use safe values from the machine limits and bed size as a default if not defined rather than error out. Can change that if youd prefer it to be explicitly set.

silentninja1, 2018-08-01:
@Roxy-3D And travis is all happy, if you can forward this along we can make sure it works for all parties then ill get it moved over to 2.0 before merging.

[4 more replies]

closed 2018-08-07 by thinkyhead