MarlinFirmware/Marlin #11253: [Bugfix 2.0]  [FR]  Option to not activate fan for bed mosfet when using USE_CONTROLLER_FAN

hamster65, 2018-07-12:
Just a thought / a nice to have:

USE_CONTROLLER_FAN activates the controller fan as soon as the heated bed starts to heat up. This is by intention to cool the mosfets.
My bed takes a long time to heat up and I usually sit nearby the printer to prepare the stl file. The fan noise is quite annoying. The mosfet has a heat sink and stays cool without any forced cooling. So it would be great to have more options here, like
USE_CONTROLLER_FAN_FOR_DRIVERS
USE_CONTROLLER_FAN_FOR_MOSFETS

boelle, 2019-02-20:
@hamster65 have you tried to use bigger fans? 80mm maybe?

melslenstra, 2021-12-31:
Thanks @thisiskeithb for spotting that my FR #23401 is a duplicate of this one. Not sure how I managed to not find this before submitting my FR.

tl;dr: In my case, the controller fans are nowhere near the actual bed MOSFETs, so it makes even less sense to have them come on at full blast when I preheat my bed.

ellensp, 2026-11-11:
Done in #27409: disable CONTROLLER_FAN_BED_HEATING in Configuration_adv.h so bed heating no longer turns on the controller fan. The fan's other triggers have their own options too (CONTROLLER_FAN_USE_Z_ONLY, CONTROLLER_FAN_IGNORE_Z, CONTROLLER_FAN_MIN_BOARD_TEMP, CONTROLLER_FAN_MIN_SOC_TEMP).

closed 2026-11-11 by ellensp