MarlinFirmware/Marlin #11254: Continuous Stepper Rotation?

Phife54, 2018-07-12:
I have a unique situation where I need to keep a stepper rotating at a set speed continuously until told to stop.

Can Marlin do this? I run into travel limit issues just trying to use G1 and large travel distance. 

If anyone has any ideas, they would be greatly appreciated.

Thanks

Grogyan, 2018-07-12:
Can you please elaborate?

Phife54, 2018-07-12:
Its for a stepper pump, need to keep the pump running while a print is going. Just want to be able to set a rotation speed and have it run until told to stop. Not sure how else to explain it. sorry.

thinkyhead, 2018-07-13:
If the number of steps-per-second doesn't need to be too many (i.e., 1KHz to 2KHz), the Temperature ISR can be extended with a function to pulse a stepper driver's pins directly, borrowing from the stepper code that does Bresenham stepping to slice up time into semi-regular units.

You could also try building on the babystepping code in `temperature.h` and `temperature.cpp` — adding an extra axis. Babystepping runs at the rate of the temperature ISR, around 1KHz. To vary the feedrate of an axis continuously, you'd basically just keep adding some number of babysteps on every temperature ISR. That value would be fractionated with Bresenham for rates under 1KHz.

Adding an extra axis under the babysteps framework would be challenging, basically adding a whole new-named axis. The HangPrinter PR applies the needed changes for a kinematic 4th axis, and it pervades a lot of stepper/planner and also some high level code.

For an axis that simply needs to move continuously, if it doesn't matter if it starts and stops instantaneously, the easiest solution might be to call directly to the functions defined in `stepper_indirection.h` to set direction and pulse a specific stepper directly, whenever a certain flag or rate value is set to non-zero.

marcio-ao, 2018-08-31:
Wild idea. If you swap a fan PWM pin with a stepper STEP pin in the pins_*.h for your board, you might be able to get continuous stepping by running the fan at a 50% duty cycle. You wouldn't get any speed control, however, other than maybe very roughly by tweaking the microstepping settings.

boelle, 2019-02-20:
@Phife54 is this still relavant?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead