MarlinFirmware/Marlin #11259: coreXY stepping issue

ruggb, 2018-07-13:
CoreXY printer
I am observing an anomaly on a print in progress. I think I have eliminated a mechanical or slicer problem so Marlin is next. When a diagonal prints it appears that when printing perpendicular to a motor and stepping each line toward that motor the step distance is different then when moving away from that motor.
Looking at the gcode, the distance appears to be the same (like xxx.348 or xxx.349mm) in either direction, which I think eliminates a slicer issue. The symptom is that in one direction the surface is rough, in the other it is very smooth - like it is perfectly spaced one way and too close the other. This is a picture of the 1st layer. The other layers exhibit the same anomaly. If it were an extrusion issue it should occur in either direction, right? Since the 2nd layer print perpendicular to the first and the other motor is now involved, I think that zeros out a mechanical issue.

[image]

This is the STL file, gcode, and slic3r config
[DJI.zip](https://github.com/MarlinFirmware/Marlin/files/2193369/DJI.zip)

Marlin is 1.1.x bugfix, commit 4dad862
I have not upgraded because of the ongoing stepper issue.

Win10 1803

Is this a known, fixed, or potential bug? Or maybe it is associated with the timing issue?
Otherwise, everything else is near perfect.

GMagician, 2018-07-13:
Don't know if same for Core machine but for cartesian also mechanical issue may give such result. You have a pulley and a belt that is pulled in one direction and pushed in other

silentninja1, 2018-07-13:
Is this a corexy or an H bot? Many people confuse them because many manufacturers incorrectly label machines... H bot is a single long belt, A true CoreXY is two shorter belts that cross opposite the steppers. H bots are prone to issues that skew parts.

ruggb, 2018-07-13:
coreXY for sure

ruggb, 2018-07-13:
please note the abrupt change when it started printing in the opposite direction. The next layer did the same thing except it was printing at 90° to the 1st and did it with the opposite motor.

crystalhand, 2018-07-23:
Check your z motor voltage.  I often have issues like this when my voltage is not high enough causing some missed steps on the first layer or two.  Usually only occurs on one motor and always in the middle of a layer.  I noticed it happened mainly at the end of a line.  If you put your fingers on the z lead screws you should feel the click and then notice the change in the way the layer looks.

closed 2018-11-05 by thinkyhead