MarlinFirmware/Marlin #11263: Standardising the communication between Marlin and an external LCD touchscreen controller

mofosyne, 2018-07-14:
I noticed that there is cocoon create, Raise3D and wanhou i3 duplicator plus, all with their own implementation of communicating between a touchscreen controller to the Marlin firmware.

But a common factor I noticed is that they communicate to the Marlin firmware via a secondary uart.

So I think what would help is if we can somehow standardise this communication between a Marlin firmware to an external controller. Especially in such a way that is futureproofed (especially considering that LCD firmware do not really get updated all that much).

My proposal would involve sending CBOR messages between Marlin and the controller.

https://en.wikipedia.org/wiki/CBOR

It's a good candidate over say json, as it is much more compact and is an RFC standard. And also being like json, it is easier to add new data items or structures later on.

For a good reference on existing extension to Marlin firmware to support touchscreen controllers, you may want to reference https://github.com/andrivet/ADVi3pp-Marlin . An opensource mod to replace the Wanhou i3 plus 3d printer firmware.

Some capabilities to consider:
* Sending GCODE commands directly. E.g. Allow the touchscreen to send a simple calibration cube to 3d print.
* Allow for sending configuration commands to adjust the eeprom settings of the printer.
* Status reporting of current print behaviour, temperature, etc...

Deneteus, 2018-07-25:
Related:

[FR] Remote Serial Display Functionality for External LCD Controllers C: LCD & Controllers T: Feature Request
#7790 opened on Sep 28, 2017 by Deneteus 

[FR] Adding Nextion HMI Display support C: LCD & Controllers T: Feature Request
#7754 opened on Sep 26, 2017 by DaeMonSx

mofosyne, 2018-09-05:
Oh that's an interesting though. So by having a Document Data Definition kind of approach, are you saying this would allow us to abstract away the Human Machine Interface Controller from the main marlin firmware?

If that's the cases, then we should look into that. I agree there will be more than just touchscreen HMI to deal with. Some may only be hardware buttons and cheap alphanumeric displays as well... so we need to be able to abstract that away indeed.

thinkyhead, 2018-09-06:
Also related: #10610 PanelDue

InsanityAutomation, 2019-05-14:
As a note, this is essentially what ExtUI is aiming to do in Marlin 2.0 and its becoming more robust constantly.

InsanityAutomation, 2019-10-15:
At this point this should be closed. The LCD's mentioned here that are not already in ExtUI have specific issues open and this is arguably complete from an infrastructure standpoint.
@boelle @shitcreek

closed 2019-10-15 by boelle