MarlinFirmware/Marlin #11266: Z_SAFE_HOMING randomly disobeying

knoopx, 2018-07-14:
Hi, using bugfix 2.x @ 8c57cadb with tmc2130 and an inductive probe, my printer, in some random situations, homes x and y but after that does not move to the center and tries to home z at 0,0. I can't figure why this randomly happens. Most of the time it works as expected, homes x and y and then moves to the center (taking into account proper offsets). 

When that happens, resetting the board does not fix it, I have to plug off octoprint and power down the PSU in order to get it properly working again.

Am I missing something?

GMagician, 2018-07-14:
But does it execute z homing when it miss moves to center ?

knoopx, 2018-07-14:
@GMagician yes it does (and works if I put something below) but my probe is outside of the bed so I have to kill the printer before it crashes into the bed.

GMagician, 2018-07-14:
Try to enable DEBUG_LEVELING_FEATURE then before G28 execute M111 S255. Finally post log

teemuatlut, 2018-07-14:
It's a known problem with no known fix yet.
My workaround has been to split the homing command on starting gcode as such:
```
G28 XY
G4 S1
G28 Z
```

knoopx, 2018-07-14:
Alright, thanks for the tips, do you happen to know if this is an all time bug or it was recently introduced? I don't recall having this problem in the past. Going to enable debugging and post log the next time it happens.

teemuatlut, 2018-07-14:
I think it's been there for a while now. I need to hook up a logic analyzer and see what's going on but I just haven't gotten to it yet.
This should largely only concern sensorless homing.

knoopx, 2018-07-14:
I do have physical endstops on x and y, and using a z probe as z endstop. Sensorless homing is not enabled.

GMagician, 2018-07-14:
I already use it with with my old drv8825 and it's working. Yes my compiled version is a little bit ancien (some days before yours) but I think it's not something introduced recently, maybe only something "appeared" recently.

knoopx, 2018-07-15:
Ok got the opportunity to grab a log during my last failure, here is it:

```
Recv: >>> G28
Send: N12 M190 S50*110
Recv: Machine Type: Cartesian
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X:-62 Y:0 Z:-2.20 (Left-Center & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv:   current_position=(0.00, 0.00, 3.00) : bracket_probe_move
Recv: > endstops.enable(true)
Recv: Z_SAFE_HOMING >>>
Recv:   current_position=(0.00, 0.00, 3.00) : sync_plan_position
Recv:   destination=(179.00, 117.00, 3.00) : Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(179.00, 117.00, 3.00)
Recv: <<< do_blocking_move_to
Recv: >>> homeaxis(Z)
Recv:   current_position=(179.00, 117.00, 3.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(5.00)
Recv: >>> do_blocking_move_to(179.00, 117.00, 7.20)
Recv: <<< do_blocking_move_to
Recv: >>> do_blocking_move_to(179.00, 117.00, 7.20)
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Z, -352.50, [4.00])
Recv:   current_position=(179.00, 117.00, 0.00) : sync_plan_position
```

GMagician, 2018-07-15:
That's weird. Position is X179 Y117 and no XY moves... sure you don't have H/W failures? like EN pin not working? When yopu have failure, if you "fake" Z sensor, terminate home and try to move X/Y what happens?

GMagician, 2018-07-16:
you may try to enable debugging commands inside `buffer_segment` function in planner.cpp, now they are commented.

This will give us more info

knoopx, 2018-08-02:
I've been messing with my setup for the past weeks and this issue no longer reproduced. I just disabled every fancy feature I had enabled at source code level and it just works. I'm pretty sure there's a bug in the linear advance or automatic bed leveling compensation code paths and I also suspect is driver-related (pulse, timing or microstepping).

closed 2018-08-04 by thinkyhead

BD992000, 2018-09-14:
Hello,
unfortunately I have the same issue.
I upgraded my Ender-3 with a MKS Gen 1.4 board with TMC2130 drivers for X,Y,Z and E0. All with SPI.
Bed leveling is done with a BLTOUCH (clone).

Everything is working fine. But about 50% after starts of the printer G28 for Z-axis just lowers the z-axis until it crashes into the bed.
I tried this on marlin 1.1.9 and bugfix-2.0. Same issue.
I have to shut down the printer and after powering up again G28 often works just fine. But not always.

Debug output looks very simmilar to the output knoopx posted above.
`Send: G28 Z
Recv: >>> G28
Recv: Machine Type: Cartesian
Recv: Probe: BLTOUCH
Recv: Probe Offset X:-48 Y:-9 Z:1.00 (Left-Front & Above Nozzle)
Recv: Auto Bed Leveling: BILINEAR
Recv:  (disabled)
Recv: 
Recv: set_bltouch_deployed(0)
Recv:   current_position=(0.00, 0.00, 10.00) : bracket_probe_move
Recv: > endstops.enable(true)
Recv: Z_SAFE_HOMING >>>
Recv:   current_position=(0.00, 0.00, 10.00) : sync_plan_position
Recv:   destination=(165.00, 126.00, 10.00) : Z_SAFE_HOMING
Recv: >>> do_blocking_move_to(165.00, 126.00, 10.00)
Recv: <<< do_blocking_move_to
Recv: >>> homeaxis(Z)
Recv:   current_position=(165.00, 126.00, 10.00) : set_probe_deployed
Recv: deploy: 1
Recv: do_probe_raise(15.00)
Recv: >>> do_blocking_move_to(165.00, 126.00, 15.00)
Recv: <<< do_blocking_move_to
[...]
Recv: echo:busy: processing
Recv: >>> do_blocking_move_to(165.00, 126.00, 15.00)
Recv: <<< do_blocking_move_to
Recv: Home 1 Fast:
Recv: set_bltouch_deployed(1)
Recv: >>> do_homing_move(Z, -375.00, [4.00])
Recv:   current_position=(165.00, 126.00, 0.00) : sync_plan_position
[...]`

the output states that the printer is moving to the safe homing position but it just does not execute the movement. And it states the bltouch is deployed but it desn't. 

As I mentioned. The movements, SPI for the stepper drivers, bltouch and even G28, G29 etc. are fully functional. But in about 30-50% of the starts, I have to restart the printer (sometimes several times) until it works.

I have no clue what this is.
Please help!