MarlinFirmware/Marlin #11272: Inconsistent bed distance after homing w/ Marlin 2.0

kookaburra88, 2018-07-14:
Hey all,

with the current Version of Marlin 2.0 I am having trouble with inconsistent bed distance after homing in millimeter range.
What I noticed is, that after the fast homing was done, on the precise homing step, the bed isn't driven against the machanical endstop in at least 2 of 10 cases, resulting in a bed distance of a few millimeters.

- Already exchanged all the hardware components --> nothing changed
- Flashed repetier firmware --> inconsistancy gone
 
Steps to reproduce:
Do multiple z-homings after another and observe the nozzle to bed distance. 

Attached the config files, in case somebody can give me advice, what I need to change.
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2195241/Marlin.zip)

Best regards
Chris

rmoravcik, 2018-07-15:
What is your printer? What is your end-stop switch wiring? Is there any noise filtration capacitor connected to your end-stop switches?

Could you please try enable ENDSTOP_NOISE_FILTER in Configuration.h if it makes any difference?

kookaburra88, 2018-07-15:
Well, the printer is a creality ender-4 with Arduino Due + ramps 1.5 24v mod. Drivers are TMC2208.

From hardware point of few I exchanged nearly every part:
- switched the Arduino Due
- switched to drv8825 drivers 
- switched the z-Endstop to ones with and without capacitor
- switched the z-Endstop cable
- tried dual-z with individual or same driver

From software point of few
- tried ENDSTOP_NOISE_FILTER but without any improvement. One interesting observation with software noise filters was, that during precise homing step, sometimes the z-motors stop for a moment and then proceeds with homing. Didn't observed that before.
- chaged the z-Endstop pins to either z_min or z_max

The endstop wiring is standard creality wiring, which goes from 
- singnal pin ramps --> signal pin endstop
- ground pin ramps --> positive pin endstop

GMagician, 2018-07-15:
Please post configuration.h and configuration_adv.h.

Are you sure bump is enough to release switches?

kookaburra88, 2018-07-15:
You will find the config files in the first post ;)

Honestky, I don't understand the second question

GMagician, 2018-07-15:
@kookaburra88 sorry I missed them.
When you do fast home you touch switches, then go back "a fixed" distance and then search for switch again.
If you don't move away but are still with micro pressed, maybe position search may fail...

GMagician, 2018-07-15:
corexy printer, sorry can't help

GMagician, 2018-07-15:
you have `Z_DUAL_STEPPER_DRIVERS` with `Z_DUAL_ENDSTOPS_ADJUSTMENT=0` are you using latest 2.0? I fixed a bug that should raise a kill message...

kookaburra88, 2018-07-15:
After fast homing step, the z-Endstop is definitely untriggered. In fact most of the time, the z-endstop is untriggered even after the precise homing step. 
Contrary, with repetier firmware, the z-Endstop is triggered ever time after doing homing.

kookaburra88, 2018-07-15:
In the attached link, you can find a video of multiple consecuitive z-homing steps, showing the inconsistancy in bed distance.

[https://www.youtube.com/watch?v=sAF0NTp7kbk](url)

mbuc, 2018-07-17:
I have had this issue as well on Bugfix-2.0, downloaded sometime June 29th.  Every once in a while Z will home incorrectly.  It's not every time, but it's frequent enough that before every print I move the Z axis down (positive direction) 10-20mm if it is close to home, rehome, verify position, start print, watch rehome, verify position, then continue.

Never happened on 1.1.6 or 1.1.7.  Haven't tried latest on either branch yet.  I'll be posting a couple of semi-related 2.0 issues of my own later today.  One of which just happened 6 hours (guessing) into a 12 hour print: the Z axis just stopped working properly.  Sending moves to it would initiate the motor, but the motor would not move at all.  It made a noise that sounded like a much quieter, cleaner set of skipped steps.  No heat in the Z motor at all.  After checking that the motor shaft was indeed still secure, the leadscrew and guide rails were lubricate, and motion worked freely w/ motors off and held position w/ motors on, I power cycled, and now they work just fine.  I have been having several strange movement-related issues like this on every long print ever since I flashed 2.0 a few weeks ago.

kookaburra88, 2018-07-18:
Is there any command to debug the firmware?

GMagician, 2018-07-18:
you may try to enable `DEBUG_LEVELING_FEATURE` and before G28 execute a M111 S255, also remove remarks from debugging commands inside `buffer_segment` function in planner.cpp.

kookaburra88, 2018-07-19:
Ok, here is one example output for a proper homing move

>>> G28
Machine Type: Core
Probe: NONE
  current_position=(0.00, 0.00, 0.00) : bracket_probe_move
> endstops.enable(true)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, -375.00, [4.00])
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:4.00 X:0.00 (0->0) Y:0.00 (0->0) Z:-375.00 (0->-960000) E:0.00 (0->0)
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 2.00, [4.00])
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:4.00 X:0.00 (0->0) Y:0.00 (0->0) Z:2.00 (0->5120) E:0.00 (0->0)
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -4.00, 1.00)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:1.00 X:0.00 (0->0) Y:0.00 (0->0) Z:-4.00 (0->-10240) E:0.00 (0->0)
echo:busy: processing
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 250.00
> home_offset[Z] = 0.00
  current_position=(0.00, 0.00, 0.00) :
<<< set_axis_is_at_home(Z)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Z)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : bracket_probe_move
<<< G28

And here is one for a not so good homing move:
>>> G28
Machine Type: Core
Probe: NONE
  current_position=(0.00, 0.00, 0.00) : bracket_probe_move
> endstops.enable(true)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, -375.00, [4.00])
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:4.00 X:0.00 (0->0) Y:0.00 (0->0) Z:-375.00 (0->-960000) E:0.00 (0->0)
<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 2.00, [4.00])
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:4.00 X:0.00 (0->0) Y:0.00 (0->0) Z:2.00 (0->5120) E:0.00 (0->0)
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -4.00, 1.00)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:1.00 X:0.00 (0->0) Y:0.00 (0->0) Z:-4.00 (0->-10240) E:0.00 (0->0)
echo:busy: processing
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 250.00
> home_offset[Z] = 0.00
  current_position=(0.00, 0.00, 0.00) :
<<< set_axis_is_at_home(Z)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Z)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : bracket_probe_move
<<< G28

GMagician, 2018-07-19:
Since both logs are equal I suspect something is happening somewhere at low level.
You may try to display what's the return value from `_buffer_steps` in `buffer_segment`.

Maybe some quick_stop has benn called

kookaburra88, 2018-07-19:
Did modify the planner.cpp file to stream out the _buffer_steps, but I have trouble understanding the values. Please find attached the modified src.

>>> G28
Machine Type: Core
Probe: NONE
  current_position=(0.00, 0.00, 0.00) : bracket_probe_move
> endstops.enable(true)
>>> homeaxis(Z)
Home 1 Fast:
>>> do_homing_move(Z, -375.00, [4.00])
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:4.00 X:0.00 (0->0) Y:0.00 (0->0) Z:-375.00 (0->-960000) E:0.00 (0->0)
buffer_segment1<<< do_homing_move(Z)
Move Away:
>>> do_homing_move(Z, 2.00, [4.00])
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:4.00 X:0.00 (0->0) Y:0.00 (0->0) Z:2.00 (0->5120) E:0.00 (0->0)
buffer_segment1
<<< do_homing_move(Z)
Home 2 Slow:
>>> do_homing_move(Z, -4.00, 1.00)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  buffer_segment FR:1.00 X:0.00 (0->0) Y:0.00 (0->0) Z:-4.00 (0->-10240) E:0.00 (0->0)
buffer_segment1
echo:busy: processing
<<< do_homing_move(Z)
>>> set_axis_is_at_home(Z)
For Z axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 250.00
> home_offset[Z] = 0.00
  current_position=(0.00, 0.00, 0.00) :
<<< set_axis_is_at_home(Z)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
<<< homeaxis(Z)
  current_position=(0.00, 0.00, 0.00) : sync_plan_position
  current_position=(0.00, 0.00, 0.00) : bracket_probe_move
<<< G28

[planner.zip](https://github.com/MarlinFirmware/Marlin/files/2211305/planner.zip)

GMagician, 2018-07-19:
Since `_buffer_steps` returns true, 1 in your print, then movement should be queued...now we are really going too deep in low level code. I'm not able to help you more

kookaburra88, 2018-07-21:
Any tips how this problem could be tackeled?

GMagician, 2018-07-21:
for print result I prefered something like
```c++
   const bool result = _buffer_steps(target
      #if HAS_POSITION_FLOAT
        , target_float
      #endif
      , fr_mm_s, extruder, millimeters
    );
  SERIAL_ECHOPAIR("buffer_segment", result);
  if (!result) return false;
```

but your code tries to insert movement twice and nothing happens.

You may also try to remove commens in debug code  in function `_populate_block`

and also printout result from same function in `_buffer_steps` call. Using similar code posted above

closed 2018-11-05 by thinkyhead