MarlinFirmware/Marlin #11273: No Fan when Autotune (with latest bugfix_1.1.1)

dteeter, 2018-07-15:
Hi,
It seems I am running into the same bug as described here: https://github.com/MarlinFirmware/Marlin/issues/1069

Details:
1. In Configuration_adv.h settings: 
   #define E0_AUTO_FAN_PIN 8
   #define EXTRUDER_AUTO_FAN_TEMPERATURE 40
  - using RAMPs 1.4
2. With E0 temperature above 40, my heatbreak fan correctly turns on
3. When running PID like M303 E0 S215 C5, fan never turns on, even when heater is above threshold
4. If I reset the board, with E0 temp above 40, fan will turn on
5. Even if E0 temp is above 40 and fan is already started, running PID like M303 E0 S215 C5, will cause the fan to turn off
6. I also tried just turning on the fan via gcode before starting autotune, with the same results
7. I added some code within temperature.cpp to try and understand if it really believes auto fan is enabled like this:
[image]

Output in repetier host when sending the M303 looks like this:
[image]

Here is why I am trying to get it to work this way:
1. The heatbreak fan's only function is to draw heat away from the tube that leads into the heater block (it is *not used* for cooling the filament)
2. So, the heatbreak fan is always on during extrusion and it is important that the normal cooling effect of this fan be taken into effect during autotune
3. If a print is finished and the printer is idling with temp low, there is no reason to have the fan running (that would reduce the service life of the fan and cause unnecessary noise and waste of power)

Config files:
[Config_Files.zip](https://github.com/MarlinFirmware/Marlin/files/2195483/Config_Files.zip)

kAdonis, 2018-07-15:
In your Configuration.h you have 

`#define MOTHERBOARD BOARD_RAMPS_14_EFB`

normally this is using pin 8 for the heated bed
this is in conflict with

`#define E0_AUTO_FAN_PIN 8`

How did you wire your heatbed? Please post your RAMPS.h file

kAdonis, 2018-07-15:
I just found in your Configuration.h:

`#define MOTHERBOARD 45`

it should be

`#define BOARD_RAMPS_14_EFF`

it's much clearer this way
But it is also in conflict

dteeter, 2018-07-15:
OK, @kAdonis  thanks for looking at the configs! I'll try this and let you know if that fixes it.

kAdonis, 2018-07-15:
I'm sorry, I was not clear at this
My suggestions wont fix your issue, but your board definitions were confusing me.
Do you have a heated bed or not? Please post your pins_RAMPS.h file

dteeter, 2018-07-15:
@kAdonis : I do not have a heated bed. My intention is to use pin 8 for the heatbreak fan and pin 9 for the filament-cooling fan. 

Here is pins_RAMPS.h:
[pins_RAMPS.h.zip](https://github.com/MarlinFirmware/Marlin/files/2195864/pins_RAMPS.h.zip)

kAdonis, 2018-07-15:
There is of course a conflict:
pin 8 is defined as parts cooling fan in pins_RAMPS.h **AND** as extruder auto fan in Configuration_adv.h 
It should not compile this way, but the ArduinoIDE does not complain.

I suggest you delete the confusing

`#define MOTHERBOARD 45`

in Configuration.h

and change in pins_RAMPS.h:

```
#ifndef RAMPS_D8_PIN
  #define RAMPS_D8_PIN     -1
#endif
```

dteeter, 2018-07-15:
@kAdonis :   looking at  pins_RAMPS.h, I see this, which seems to be defining Fan1 as pin 8:
[image]

And in Configuration_adv.h, auto fan is defined as pin 8:
[image]

Motherboard is defined as "BOARD_RAMPS_14_EFF"
[image]

If I do a "#define RAMPS_D8_PIN     -1", wont that make fan 1 unusable?

Its not clear to me where the conflict is happening...

kAdonis, 2018-07-15:
> If I do a "#define RAMPS_D8_PIN -1", wont that make fan 1 unusable?

Yes, that's the purpose!

Fan 0 (pin 9) and fan 1 (pin 8) in pins_RAMPS.h  are for cooling the printed part.

The extruder auto fan that you want to use on pin 8 is something else, most people are using pin 4 for it (with additional hardware)

it doesnt matter, if  the motherboard is defined as ramps 14 EFB or EFF
normally the define looks like this:
[image]

kAdonis, 2018-07-15:
It should work during autotune,
I have my extruder auto fan on pin 9, and it works

dteeter, 2018-07-16:
@kAdonis : I did try changing the Auto Fan to pin 9, but that did not fix the problem either. Note: when using pin 9 for Auto Fan, the compiler _will give an error_ in case of a conflict between ramps pins and auto fan setting (and it will not give an error when using pin 8, which might be considered to be a second bug).

I finally completed PID setting by hard wiring the fan during PID check and then returning it to the normal setting (now at pin 9) after that.

At this point, I believe the lack of auto-fan support during pid autotune continues to be a valid bug.

kAdonis, 2018-07-16:
I just tested PID autotune again on my printer, the heatbed is switched off when autotune starts, but my extruder auto fan is switched on just as always above 50°
You found a workaround, thats fine
I cant help you any more, maybe someone else has an idea how to fix this

dteeter, 2018-08-28:
Closing...I am no longer able to duplicate this. It was probably a wiring problem.

closed 2018-08-28 by dteeter