MarlinFirmware/Marlin #11278: adjust of accidently z tilt

derdigge, 2018-07-16:
Dear all.

I went throuh Configuration.h and Configuration_adv.h. The thing i was after is a adjust for ztilt.
My Printer is a CR10 with dual z leadscrews. Sometimes the right Z falls a little bit off. Another case
is during load of Filament or toolheadchanges i accidently move only one side.

As i have a z probe i wonder if this is possible to realign it? Like:
Probe X40 Y165 = 3.25mm from home postion (Z1)
Probe X270 Y165 = 1.1mm from home postion (Z2)

Now calculate the tilt and align the steppers. Is there such functionality in Marlin?
The only thing i found is dual endstops for Z.

If i use the bilenear autolevel the leadscrew is moving all the time during print then? Instead of realigning the z axes.

Thanks

derdigge, 2018-07-24:
No one?

houseofbugs, 2018-07-29:
This has to be done manually since those steppers are wired to the same stepper driver. https://www.youtube.com/watch?v=tOOuJURIQ54

derdigge, 2018-07-30:
For me they arent. I wired em to seperate stepper drivers. I use this feature with klipper and it works great.

comps, 2018-07-30:
Possible duplicate of https://github.com/MarlinFirmware/Marlin/issues/6865 , https://github.com/MarlinFirmware/Marlin/issues/10049 , ..., maybe related to https://github.com/MarlinFirmware/Marlin/issues/10082 and solved by https://github.com/MarlinFirmware/Marlin/pull/11302 .

derdigge, 2018-07-30:
yes. You are awsome! 
Thats it, ecxactly!

closed 2018-07-30 by derdigge