MarlinFirmware/Marlin #11279: Bugfix-1.1.x 256f76d  error compiling - // Check for commands that require the printer to be homed

bruce356, 2018-07-16:
Hi using arduino 1.8.5 to compile bugfix-1.1.x_14-07-2018_commit 256f76d.
I receive the following error (hope someone can help)
sketch\ubl_G29.cpp: In static member function 'static void unified_bed_leveling::G29()':
ubl_G29.cpp:301: error: 'tool_change' was not declared in this scope
         if (active_extruder != 0) tool_change(0);
                                                ^
exit status 1
'tool_change' was not declared in this scope

Error stems from:-
  // Check for commands that require the printer to be homed
    if (may_move) {
      if (axis_unhomed_error()) home_all_axes();
      #if ENABLED(DUAL_X_CARRIAGE)
        if (active_extruder != 0) tool_change(0);
      #endif
    }

Printer is dual X carriage Cartesian with RUMBA board.
Have attached my config files
Thanks.

[14-07-2018_commit 256f76d.zip](https://github.com/MarlinFirmware/Marlin/files/2197582/14-07-2018_commit.256f76d.zip)

InsanityAutomation, 2018-07-16:
This is a duplicate of issue 11252 that I opened the other day. Roxy added that check and another above it that are out of scope and don't function. I can remove them, but in the case of this one it would be great for preventing a potential crash scenario so it'd be nice to keep it.

bruce356, 2018-07-16:
Thanks for the quick reply InsanityAutomation, is Roxy aware of the problem and is there a fix?
If not how far back would I need to go to be able to configure a working of bugfix 1.1.x
thanks

InsanityAutomation, 2018-07-16:
I'm not sure if Roxy has seen it or not yet, but for a short term, commenting out the offending line works. The check was added recently, so it's not a regression. It would be nice to have though as it does protect from an ugly situation.

bruce356, 2018-07-16:
@InsanityAutomation
I do not have programming knowledge, could you guide me which line to comment out in which file.
regards

closed 2018-07-17 by bruce356

thinkyhead, 2018-07-19:
Compile issue is now patched.

commit 1e0edbcc (2018-07-19): Fix tool_change call in UBL G29

Fix #11279
commit 17ed0f81 (2018-07-19): Fix tool_change call in UBL G29

Fix #11279, #11252
commit db057eb4 (2018-07-19): Fix tool_change call in UBL G29

Fix #11279, #11252