MarlinFirmware/Marlin #11280: Q: Need help with 3 point leveling

boelle, 2018-07-16:
For now i have given up on UBL as my bed is not made for it

I have found the precise X,Y cords for 3 of the probe points and i have also verified my probe offset

In the X direction everything is fine, but in Y the probe is about 20mm from reaching the points

`
 #define PROBE_PT_1_X 36
  #define PROBE_PT_1_Y 6.4
  #define PROBE_PT_2_X 36
  #define PROBE_PT_2_Y 202.4
  #define PROBE_PT_3_X 239
  #define PROBE_PT_3_Y 94.4
`

first point if i just change that to 5.4 it fails with out of probe area and nozzle is set to limit at -10 to allow the probe to reach the extreme edge of the bed

configs attached
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2197909/Configuration.zip)

boelle, 2018-07-16:
i got 2 of the 3 points

i changed the Y numbers and the 2 last points are now hit spot on by the probe

  #define PROBE_PT_1_X 35
  #define PROBE_PT_1_Y 6.4
  #define PROBE_PT_2_X 35
  #define PROBE_PT_2_Y 182.4
  #define PROBE_PT_3_X 238
  #define PROBE_PT_3_Y 84.4

the first one i cant get to work, not even if i change my travel limits for Y to -20

#define X_MIN_POS 0
#define Y_MIN_POS -20
#define Z_MIN_POS 0
#define X_MAX_POS 253.4
#define Y_MAX_POS 248.9
#define Z_MAX_POS 209.40

boelle, 2018-07-16:
ahhh i screwed up on probe offset, it was set to in front of the nozzle and not behind

closed 2018-07-16 by boelle

boelle, 2018-07-16:
@Roxy-3D @thinkyhead 

are there any rules for how to place the 3 points?

i placed 2 of them at one side like you would normally expect, but the 3rd i placed on the oposite side but half way so seen from above they form a triangle

AnHardt, 2018-07-16:
The 3 rules are_
* reachable by the probe and probable
* not kollinear, not on the same straight line, building a triangle - what is all meaning the same.
* as large as possible go get less error. (a triangle with sides of 1mm is crap - 1/2 bed is probably good enough)

boelle, 2018-07-16:
:-D

then i should be good, just need to get first layer correct and do a few test prints to see if i even need ubl