MarlinFirmware/Marlin #11281: M600 - Changing filament without using LCD and encoder

bogdanbalanescu, 2018-07-16:
Hello!

I am trying to change the filament using the M600 command without an LCD screen and an encoder. The way I want to do it, is calling other M commands when the printer asks for input.

I have done some modifications to Marlin in an attempt to achieve this (which I will describe further below), but I have failed to achieve my goal. However, in doing so, I have discovered something which I would call a bug, but could possibly be an expected behavior. At some point during the M600, Marlin doesn't accept any command, not even M112 Emergency Stop, which could be dangerous, because I guess anyone could use this command when operating the printer from a very big distance if they notice a danger.

The way M600 works right now is:
1. Issue M600
2. The printer pauses the print job, saves the current location, raises the nozzle, moves to the parking spot, retracts the filament, waits for user to insert filament and to press the encoder (or issue M108)
3. The user inserts the filament, then presses the encoder (or issues M108)
4. The printer extrudes the filament, then asks the user to either continue the print or extrude more filament
5. The user chooses one of the options. (Repeats between 4 and 5 until the user chooses to continue the print) - (User can only communicate now via the encoder, choosing one of the two options - at this point no command is executed by the printer, not even M112). 
6. The printer moves to the printing position, then continues the printing job.

The way I want M600 to work is to let the user make the choice proposed at points 4 and 5, by using either the encoder, or by issuing M601 Resume Print or M602 Extrude More.

In the following I will describes the steps I took into adding M601 and M602 commands.
1. Add these lines to Configuration_adv.h:
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define FILAMENT_CHANGE_HOST_EXCLUSIVE // Enables M601 and M602
#endif

2. In the Marlin_main.cpp, add the following lines of code before the definition of "process_parsed_command()":
#ifdef FILAMENT_CHANGE_HOST_EXCLUSIVE
  /**
  * M601: Advanced Pause Resume Print.
  * M602: Advanced Pause Extrude More.
  * 
  */
  inline void gcode_M601() {
    advanced_pause_menu_response = ADVANCED_PAUSE_RESPONSE_RESUME_PRINT;
  }
  inline void gcode_M602() {
    advanced_pause_menu_response = ADVANCED_PAUSE_RESPONSE_EXTRUDE_MORE;
  }
#endif

3. In the Marlin_main.cpp, in the "process_parsed_command()" function, in the "case: 'M'" of the main switch statement, add the following lines of code at the end:
#if ENABLED(FILAMENT_CHANGE_HOST_EXCLUSIVE)
  case 601:
    gcode_M601();
    break;
  case 602:
    gcode_M602();
    break;
#endif

4. In the Marlin_main.cpp, in the "get_serial_commands()" function, after the "#endif#" of the "#if DISABLED(EMERGENCY_PARSER)", add these lines:
if (strcmp(command, "M601") == 0) { advanced_pause_menu_response = ADVANCED_PAUSE_RESPONSE_RESUME_PRINT; }
if (strcmp(command, "M602") == 0) { advanced_pause_menu_response = ADVANCED_PAUSE_RESPONSE_EXTRUDE_MORE; }

After implementing M601 and M602, the expected behavior is that I could change the filament without using the LCD and the encoder. I can now issue these commands when the printer says "busy: processing", but when the printer says "busy: paused for user" (which it does right when it asks to continue printing or extrude more) I cannot issue any command.

I have tracked down where I think it makes the printer pause for user, and it's in Marlin_main.cpp, in the "static void resume_print(...)" function, this block:
          // Show "Extrude More" / "Resume" menu and wait for reply
          KEEPALIVE_STATE(PAUSED_FOR_USER);
          wait_for_user = false;
          lcd_advanced_pause_show_message(ADVANCED_PAUSE_MESSAGE_OPTION);
          while (advanced_pause_menu_response == ADVANCED_PAUSE_RESPONSE_WAIT_FOR) idle(true);
          KEEPALIVE_STATE(IN_HANDLER);

I have tried to make "wait_for_user" true, but it does not seem to solve the issue.

Is there any way to make Marlin not enter the "busy: paused for user" state, and instead to only stay in the "busy: processing" state while it asks to continue printing or extrude more? (this way I could call those M601 and M602 commands, and probably it would solve the issue).

Also, is it an expected behavior for Marlin not to accept any commands at some points? Not even those from the EMERGENCY_PARSER section?

Any input on this is greatly appreciated. Thanks!

AnHardt, 2018-07-16:
Start with looking up how pingpong (ok) protocol works.
Think about how poisening an regularly sent M105 is.
Think about a M600 in mid of a file.
Look up what the emergency command parser is about.
Realize that needs support by the hosts - it needs to violate the protocol by purpose - i don't know about a host currently doing that. (except a stupid terminal program - always violating the protocol)
Look up what M108 and  `wait_for_user` really does.
Redesign M601/602. Integrate them into the emergency command parser.
Step on the feet of the host developers.

Sorry. Currently you are far away from a working concept.

bogdanbalanescu, 2018-07-17:
@AnHardt 
Thanks. You gave me a lot to think about, and study.
Currently I cannot test anything, as I will not have access to a 3D printer for the week, but I will get back as soon as possible with the solution (or other questions).

bogdanbalanescu, 2018-08-08:
Hello! Sorry for the long wait. It took us a little more to figure out how the host was also working and to test the solution a bit. To change the filament using a host exclusively, one could follow these steps:

1. In Configuration_adv.h enable EMERGENCY_PARSER (make sure to also update the related comment to include M601 and M602).
// Added support for M601 & M602 (changing filament related)
#define EMERGENCY_PARSER
2. In enum.h make enum e_parser_state look like this (this adds support for the two commands):
#if ENABLED(EMERGENCY_PARSER)
  enum e_parser_state {
    state_RESET,
    state_N,
    state_M,
    state_M1,
    state_M10,
    state_M108,
    state_M11,
    state_M112,
    state_M4,
    state_M41,
    state_M410,
    **state_M6,
    state_M60,
    state_M601,
    state_M602,**
    state_IGNORE // to '\n'
  };
#endif 
3. In MarlinSerial.cpp in emergency_parser function:
a. modify the description comment to:
    // Currently looking for: M108, M112, M410, **M601, M602**
b. in the switch(state), modify case state_M like this:
        case state_M:
          switch (c) {
            case ' ': break;
            case '1': state = state_M1;     break;
            case '4': state = state_M4;     break;
            **case '6': state = state_M6;     break;**
            default: state = state_IGNORE;
          }
          break;

c. in the switch(state), add these cases before case state_IGNORE:
        case state_M6:
          state = (c == '0') ? state_M60 : state_IGNORE;
          break;
        case state_M60:
          switch (c) {
            case '1': state = state_M601;    break;
            case '2': state = state_M602;    break;
            default: state = state_IGNORE;
          }
          break;

d. in the switch(state), in the default case, there is another switch(state) case, add these cases before the inner default case:
              case state_M601:
                advanced_pause_menu_response = ADVANCED_PAUSE_RESPONSE_RESUME_PRINT;
                SERIAL_PROTOCOL("M601 reached");
                SERIAL_EOL();
                break;
              case state_M602:
                advanced_pause_menu_response = ADVANCED_PAUSE_RESPONSE_EXTRUDE_MORE;
                SERIAL_PROTOCOL("M602 reached");
                SERIAL_EOL();
                break;

After just these steps, M601 and M602 should work in such a way you don't need an encoder to change the filament any more.
We have successfully tested this with hosts such as Pronterface and OctoPrint.
In Pronterface, changing filament works as follows:
1. M600
2. Wait for the printer to retract the filament and beep.
3. M108
4. Wait for the printer to extrude the filament until it stops extruding.
5. M601 (if you want to continue printing)
6. M602 (if you want to extrude more)

In OctoPrint, the host waits for an "ok" before sending the next command, and because M600 only sends this "ok" when the whole procedure is finished, "Fake Acknowledgement" is required. When having M600 inside a file, we noticed 2 "Fake Acknowledgement" is needed after each of the commands (M108, M601, M602), and only 1 "Fake Acknowledgement" was sufficient when M600 was explicitly sent by us. For instance:
1. M600 (initiated by the file)
2. Wait for the printer to retract the filament and beep.
3. M108
4. 2x Fake Acknowledgement
5. Wait for the printer to extrude the filament until it stops extruding.
6. M601 (if you want to continue printing)
7. 2x Fake Acknowledgement
8. M602 (if you want to extrude more)
8. 2x Fake Acknowledgement

I hope this helps someone! If you have any questions, please do ask.

Mooses344, 2019-12-01:
Hello good morning, I can't make it work, only the m108 works but the M601 and the M602 don't

el-quique, 2020-04-04:
Hello is for Marlin 1.1.9 or 2.0.x. Thanks

G2Barbour, 2020-06-17:
I've done it. 

https://www.youtube.com/watch?v=HJpfO8XifqI

InsanityAutomation, 2020-06-17:
This is rather ancient, host prompt support does this with M876 to choose actions as well.

G2Barbour, 2020-06-17:
Except the emergency parser does not work for M876.

InsanityAutomation, 2020-06-17:
Yes it does, I coded it myself.

G2Barbour, 2020-06-17:
Then it has a bug. It doesn't work when in a busy state waiting for user during an M600 event. I had to overrride all occurrences of #if (DISABLED(EMERGENCY_PARSER) to get it to work in the context of this post. So if you could make a permanent change to the code so we don't have to modify it that would be greatly appreciated! I have an Artillery Sidewinder and out TFT does not have an encoder. I had to jump through hoops to get the runout script working with the TFT *and* Octoprint.

// Enable an emergency-command parser to intercept certain commands as they
// enter the serial receive buffer, so they cannot be blocked.
// Currently handles M108, M112, M410
// Does not work on boards using AT90USB (USBCON) processors!
#define EMERGENCY_PARSER 

Notice how these commands are all *disabled* #if emergency parser is enabled. Is it possible this is supposed to say enabled instead of disabled??? I added the EMERGENCY_OVERRIDE parameter:

/*************************gcode.cpp************************/
      #if (DISABLED(EMERGENCY_PARSER) || ENABLED(EMERGENCY_OVERRIDE))
        
        case 108: M108(); break;                                  // M108: Cancel Waiting
        case 112: M112(); break;                                  // M112: Full Shutdown
        case 410: M410(); break;                                  // M410: Quickstop - Abort all the planned moves.
        #if ENABLED(HOST_PROMPT_SUPPORT)
          case 876: M876(); break;                                // M876: Handle Host prompt responses
        #endif
      #else
        case 108: case 112: case 410:
        #if ENABLED(HOST_PROMPT_SUPPORT)
          case 876:
        #endif
        break;
      #endif

  

/*************************gcode.h************************/
#if (DISABLED(EMERGENCY_PARSER) || ENABLED(EMERGENCY_OVERRIDE))
    static void M108();
    static void M112();
    static void M410();
    #if ENABLED(HOST_PROMPT_SUPPORT)
      static void M876();
    #endif
  #endif

/*************************queue.cpp************************/
        #if (DISABLED(EMERGENCY_PARSER) || ENABLED(EMERGENCY_OVERRIDE))
          // Process critical commands early
          if (strcmp(command, "M108") == 0) {
            wait_for_heatup = false;
            #if HAS_LCD_MENU
              wait_for_user = false;
            #endif
          }
          if (strcmp(command, "M112") == 0) kill(M112_KILL_STR, nullptr, true);
          if (strcmp(command, "M410") == 0) quickstop_stepper();
        #endif

/*************************M108_M112_M410.cpp************************/
#if (DISABLED(EMERGENCY_PARSER) || ENABLED(EMERGENCY_OVERRIDE))

#include "../gcode.h"
#include "../../MarlinCore.h" // for wait_for_heatup, kill, quickstop_stepper
/**
 * M108: Stop the waiting for heaters in M109, M190, M303. Does not affect the target temperature.
 */
void GcodeSuite::M108() {
  #if HAS_RESUME_CONTINUE
    wait_for_user = false;
  #endif
  wait_for_heatup = false;
}

/**
 * M112: Full Shutdown
 */
void GcodeSuite::M112() {
  kill(M112_KILL_STR, nullptr, true);
}

/**
 * M410: Quickstop - Abort all planned moves
 *
 * This will stop the carriages mid-move, so most likely they
 * will be out of sync with the stepper position after this.
 */
void GcodeSuite::M410() {
  quickstop_stepper();
}

#endif // !EMERGENCY_PARSER

InsanityAutomation, 2020-06-17:
What platform are you on? That code makes it run in the normal parser, not the emergency parser. If its getting there at all then the emergency parser is not active at all.

Im just getting things caught back up to date, most of my branches are 2-3mo old right now, after all the pandemic mayhem so its possible it was broken again in the interim.

The TFT on that machine is a serial streamer and can interfere with octoprint. It has no direct interface to Marlin and talks on the same serial channel as USB. It can at times cause random issues when one of its commands injects in the middle of what octoprint is sending.

G2Barbour, 2020-06-17:
I'm running 2.0.5.3 on an MKS Gen L board with an Mkstft_28 clone. I have no encoder and the same serial port is used for both the TFT and the USB cable that plugs into the computer to run octoprint. So having to share the serial port has been a challenge but host actions have saved the day so far. It allows me to remotely remotely parse runout events from the TFT. I had to enable an LCD to be able to activate the family of features associated with advanced pause, so I chose reprap discount smart controller but I'm not using any of the encoder pins. I'm using LCD support as a Trojan horse to activate the M600 features.

InsanityAutomation, 2020-06-17:
Shouldnt need an LCD enabled if emergency parser is enabled. I have a few machines with no LCD enabled but M600 is on just based on host prompt and emergency parser. Ive got the same machine, ill update the code and test a few configurations over the next day or two. In the meantime, post youre config files and ill glance for anything obvious.

G2Barbour, 2020-06-17:
Awesome I appreciate the help. Maybe the lcd is actually causing me the problem in the first place! I will upload the configuration files momentarily.

G2Barbour, 2020-06-17:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4793515/Configuration.h.txt)

G2Barbour, 2020-06-17:
[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4793519/Configuration_adv.h.txt)

G2Barbour, 2020-06-17:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration.h
 *
 * Basic settings such as:
 *
 * - Type of electronics
 * - Type of temperature sensor
 * - Printer geometry
 * - Endstop configuration
 * - LCD controller
 * - Extra features
 *
 * Advanced settings can be found in Configuration_adv.h
 *
 */
#define CONFIGURATION_H_VERSION 020005

//===========================================================================
//============================= Artillery Options ===========================
//===========================================================================

/* #ifdef EMERGENCY_OVERRIDE #define FIL_RUNOUT_PIN 2 */
#define ARTILLERY 

/* Artillery Genius Printer #define CUSTOM_MACHINE_NAME "Artillery Genius"   #define X_BED_SIZE 228  #define Y_BED_SIZE 228  #define Z_MAX_POS 250 */
//#define GENIUS 

/* Automatic Bed Leveling with BL Touch #define BLTOUCH  */
//#define ABL 

#ifdef ABL
  #define NOZZLE_TO_PROBE_OFFSET { 31, -31, 3 }   
  #define WAGGSTER_MOD_WIRING //BLTOUCH to Z+ Z_MAX_PIN (D19) Waggster Mod style wiring, comment out to use Z_MIN_PIN (D18)  
#endif

/* DISABLE S_CURVE_ACCELERATION DUE TO HIGH SPEED STUTTERING. */
#define HIGH_SPEED_PRINTING   

/* BMG EXTRUDER #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.13, 80.18, 400, 413 } #define INVERT_E0_DIR true */
//#define BMG 

/* NOVA HOTEND #define HEATER_0_MAXTEMP 390 #define HOTEND0_BETA 4267 #define PID_FUNCTIONAL_RANGE 25 */
//#define NOVA 

/* HEMERA HOTEND #define TEMP_SENSOR_0 5 #define HEATER_0_MAXTEMP 300 (it's 300 by default) #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.13, 80.18, 400, 409 } #define INVERT_E0_DIR true */
//#define HEMERA 

[... 2303 lines cut ...]

/**
 * R/C SERVO support
 * Sponsored by TrinityLabs, Reworked by codexmas
 */

/**
 * Number of servos
 *
 * For some servo-related options NUM_SERVOS will be set automatically.
 * Set this manually if there are extra servos needing manual control.
 * Leave undefined or set to 0 to entirely disable the servo subsystem.
 */
#define NUM_SERVOS 1 // Servo index starts with 0 for M280 command

// (ms) Delay  before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY { 300 }

// Only power servos during movement, otherwise leave off to prevent jitter
//#define DEACTIVATE_SERVOS_AFTER_MOVE

// Allow servo angle to be edited and saved to EEPROM
//#define EDITABLE_SERVO_ANGLES

G2Barbour, 2020-06-17:
/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#pragma once

/**
 * Configuration_adv.h
 *
 * Advanced settings.
 * Only change these if you know exactly what you're doing.
 * Some of these settings can damage your printer if improperly set!
 *
 * Basic settings can be found in Configuration.h
 *
 */
#define CONFIGURATION_ADV_H_VERSION 020005

// @section temperature

//===========================================================================
//=============================Thermal Settings  ============================
//===========================================================================

//
// Custom Thermistor 1000 parameters
//
#if TEMP_SENSOR_0 == 1000
  #define HOTEND0_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND0_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #ifndef NOVA
    #define HOTEND0_BETA                 3950    // Beta value  
  #else
    #define HOTEND0_BETA                 4267    // Beta value  //Ryan 4267// 
  #endif
#endif

#if TEMP_SENSOR_1 == 1000
  #define HOTEND1_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND1_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND1_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_2 == 1000
  #define HOTEND2_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND2_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND2_BETA                 3950    // Beta value
#endif

#if TEMP_SENSOR_3 == 1000
  #define HOTEND3_PULLUP_RESISTOR_OHMS 4700    // Pullup resistor
  #define HOTEND3_RESISTANCE_25C_OHMS  100000  // Resistance at 25C
  #define HOTEND3_BETA                 3950    // Beta value
[... 3062 lines cut ...]
#if ENABLED(PRINTCOUNTER)
  #define SERVICE_WARNING_BUZZES  3
  // Activate up to 3 service interval watchdogs
  //#define SERVICE_NAME_1      "Service S"
  //#define SERVICE_INTERVAL_1  100 // print hours
  //#define SERVICE_NAME_2      "Service L"
  //#define SERVICE_INTERVAL_2  200 // print hours
  //#define SERVICE_NAME_3      "Service 3"
  //#define SERVICE_INTERVAL_3    1 // print hours
#endif

// @section develop

//
// M100 Free Memory Watcher to debug memory usage
//
//#define M100_FREE_MEMORY_WATCHER

//
// M43 - display pin status, toggle pins, watch pins, watch endstops & toggle LED, test servo probe
//
#define PINS_DEBUGGING

// Enable Marlin dev mode which adds some special commands
//#define MARLIN_DEV_MODE

InsanityAutomation, 2020-06-17:
My branch FWIW - https://github.com/InsanityAutomation/Marlin/tree/ArtilleryX1_2.0_Devel

Can try this and see if it works as-is if youd like before I get a chance to run some tests myself. Rebased it so its current. Day job is calling so I wont be back to this till late tonight.

G2Barbour, 2020-06-17:
I see that you don't have the default configuration set up for the Sidewinder, but I did notice your description for EMERGENCY_PARSER now lists M876, PROMISING!
/**
 * Emergency Command Parser
 *
 * Add a low-level parser to intercept certain commands as they
 * enter the serial receive buffer, so they cannot be blocked.
 * Currently handles M108, M112, M410, M876
 * NOTE: Not yet implemented for all platforms.
 */
#define EMERGENCY_PARSER

OK, I have tried this version of Marlin and disabled LCD support instead of adding EMERGENCY_OVERRIDE and the behavior is definitely different, which could be from disabling LCD support, or it could be from differences in the backend code. I did notice changes to the syntax of some expressions but nothing caught my attention offhand that would be functionally different for this behavior For example, this new way to implement conditionals:  TERN_(O,A). I don't like it personally, and I wonder what it accomplishes.

Action notifications no longer occur at the start of a print "//action:notification Printing..." or the end "//action:notification Artillery Sidewinder X1 Ready". I need these notifications to tell the TFT when to switch modes for an Octoprint session!

And under this configuration, when I respond to the "//action:prompt_begin Nozzle Parked" dialog, it does not stay paused for the Purge More/Continue Dialog, it just continues printing after acknowledging the Nozzle Parked dialog. (And a side note about that dialog: it calls for M108 but M108 was kicking me out of the M600 script in the 2.0.5.3 I was running before. I experimented to find that an M876 S0 did the trick and got me, functionally, to the Purge More dialog, but that doesn't work here.) The strange thing is that it displays the Purge More/Continue Dialog, but it just continues with the print like the dialog doesn't exist and doesn't take the appropriate actions upon acknowledging the M876 S0 or M876 S1.

/*THESE RESPONSES RESULT IN NO ACTION!!! THERE IS NO FURTHER PURGING ON M876 S0 AND IF A PRINT IS IN SESSION IT WILL ALREADY HAVE RESUMED BY THE TIME THESE DIALOGS APPEAR*/
Send: M876 S0
Recv: //action:prompt_end
Recv: //action:prompt_begin Paused
Recv: //action:prompt_button PurgeMore
Recv: //action:prompt_button Continue
Recv: //action:prompt_show
Recv: M876 Responding PROMPT_FILAMENT_RUNOUT
Recv: ok
[...]
Send: M876 S0
Recv: //action:prompt_end
Recv: //action:prompt_begin Paused
Recv: //action:prompt_button PurgeMore
Recv: //action:prompt_button Continue
Recv: //action:prompt_show
Recv: M876 Responding PROMPT_FILAMENT_RUNOUT
Recv: ok
[...]
Send: M876 S1
Recv: M876 Responding PROMPT_FILAMENT_RUNOUT
Recv: ok

It appears that part of Marlin thinks the M600 script is complete as soon as it receives the Nozzle Parked response, and the main gcode execution thread resumes, but another function is still handling the dialogs as if the print is still paused because it is supposed to be.  I am thinking it has to do with this configuration failing to hold a BUSY state to stop the host from sending new gcode. That, of course, doesn't solve the fact that Purge More doesn't work, but it would be half of a solution.

Here are my current configuration files, this should make it relatively easy to test later on. I am using the X+ position for #define  FIL_RUNOUT_PIN 2, otherwise it is configured for a stock setup.

If you want in depth details about how I got it working properly I couldn't say everything I need to say here, it's too much information. But if you have 12min to watch my video I think you'll understand where I am coming from.

https://www.youtube.com/watch?v=HJpfO8XifqI

I would very much like a permanent solution to this so no modification of Marlin is necessary on the backend! Whether that be help configuring something differently or changes to the repository!

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/4795015/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/4795016/Configuration.h.txt)

G2Barbour, 2020-06-17:
Here are the changes that I found necessary to the previous build of Marlin:

//=============================================================================
//================================ Configuration.h ===============================
//=============================================================================

#define FILAMENT_RUNOUT_SENSOR
	/*Self-explanatory. This is the foundation for everything we are doing.*/

    #define FIL_RUNOUT_PIN 2 //MKS GEN L 1.0 X_MAX PIN
/*This assigns the FILAMENT_RUNOUT_SENSOR to the X+ pin (D2) of the MKS GEN L board. We will have to move the runout sensor from the TFT to this location in order to use it from both the TFT and Octoprint*/

/**Print Job Timer
 * Automatically start and stop the print job timer on M104/M109/M190.
 *   M104 (hotend, no wait) - high temp = none,        low temp = stop timer
 *   M109 (hotend, wait)    - high temp = start timer, low temp = stop timer
 *   M190 (bed, wait)       - high temp = start timer, low temp = none
 * The timer can also be controlled with the following commands:
 *   M75 - Start the print job timer
 *   M76 - Pause the print job timer
 *   M77 - Stop the print job timer*/ 
#define PRINTJOB_TIMER_AUTOSTART
/*This is important because Marlin will not monitor the runout sensor unless the print timer is running, so enabling this feature will allow us to activate the print timer while printing from the TFT.*/ 

/////////////////////////NOT USED FOR THIS TEST#define REPRAP_DISCOUNT_SMART_CONTROLLER
/*This allows us to enable the family of features necessary to activate the M600 scripts*/ 

//=============================================================================
//============================== Configuration_adv.h =============================
//=============================================================================

// Enable an emergency-command parser to intercept certain commands as they
// enter the serial receive buffer, so they cannot be blocked.
// Currently handles M108, M112, M410
// Does not work on boards using AT90USB (USBCON) processors!
#define EMERGENCY_PARSER 
	/*Note that this feature does not handle M876 in this description*/

/**
 * Host Action Commands
 *
 * Define host streamer action commands in compliance with the standard.
 *
 * See https://reprap.org/wiki/G-code#Action_commands
 * Common commands ........ poweroff, pause, paused, resume, resumed, cancel
 * G29_RETRY_AND_RECOVER .. probe_rewipe, probe_failed
 *
 * Some features add reason codes to extend these commands.
 *
 * Host Prompt Support enables Marlin to use the host for user prompts so
 * filament runout and other processes can be managed from the host side.
 */
#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PROMPT_SUPPORT
#endif
	/*Both if these would probably work wonderfully if we were using a different serial port for the TFT*/

G2Barbour, 2020-06-17:
UPDATE FROM PRELIMINARY TESTING: 

After enabling LCD support, behavior went back to normal, even without the EMERGENCY_OVERRIDE, so this is definitely an improvement over the previous version! No more backend modification necessary, just some Trojan horse action on the LCD support. It's not as elegant as I'd like, but it does work!

G2Barbour, 2020-06-17:
(I also enabled #define LCD_SET_PROGRESS_MANUALLY for later use. I want to display Octoprint progress on the TFT in the future.)

G2Barbour, 2020-06-17:
This behavior is also new, but the print did resume successfully afterwards:

Recv: //action:paused
Printer signalled that it paused, switching state...
Changing monitoring state from "Printing" to "Pausing"
Recv: //action:prompt_end
Recv: //action:prompt_begin Pause
Recv: //action:prompt_button Dismiss
Recv: //action:prompt_show
Recv:  T:215.31 /215.00 B:59.96 /60.00 @:35 B@:19
Recv: echo:busy: processing
Recv:  T:215.00 /215.00 B:60.05 /60.00 @:40 B@:16
Recv: echo:busy: processing
Recv:  T:214.73 /215.00 B:59.99 /60.00 @:44 B@:18
Send: N2783 M876 S0*87
Recv: M876 Responding PROMPT_GCODE_INFO
Recv: echo:busy: processing
Recv:  T:214.92 /215.00 B:60.02 /60.00 @:40 B@:17
Recv: echo:busy: processing
Recv:  T:215.00 /215.00 B:59.97 /60.00 @:39 B@:18
Recv:  T:214.88 /215.00 B:60.00 /60.00 @:40 B@:17
Recv: echo:busy: processing
Recv:  T:214.92 /215.00 B:59.99 /60.00 @:39 B@:18
Recv:  T:214.92 /215.00 B:59.99 /60.00 @:39 B@:18
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Recv: echo:busy: processing

Neither did I recognize these responses maybe they are new or maybe I just never noticed them being initiated in the terminal:
Changing monitoring state from "Operational" to "Starting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Starting" to "Printing"

InsanityAutomation, 2020-06-21:
FYI, I noticed when rebasing my configs got blown out, so you may have seen a default config there. Its up correctly now. They were sitting in my offline changes not pushed. As the error suggests, have you increased the octoprint communication timeout, and set the checkbox that it is not the exclusive host? I just flashed my machine and ill be testing shortly.

G2Barbour, 2020-06-21:
I wasn't that concerned about the timeout at the moment. It may just be that the M600 is a long command process when the user is not present to handle the dialogs. If that's the case I couldn't avoid that error unless I completely disabled timeout error management.

Let me know how testing goes. I'm particularly interested in whether or not you can get the PurgeMore dialog to work without enabling LCD support.

G2Barbour, 2020-06-21:
I'll look for the exclusive host checkbox now

InsanityAutomation, 2020-06-21:
Ive confirmed the issue is related to the Octoprint queue not Marlin. Ive forwarded some details to @foosel over discord. Once M876 is actually sent by the host, its processed however something is causing it to not jump the queue (maybe something in the capability report?) So we may need some info from that side.

G2Barbour, 2020-06-21:
Awesome! I'm assuming he's an Octoprint guy? If so, could you ask if he could update the parser for Host Prompt support so that Dialogs are automatically cleared when ack_seen("//action:prompt_end")? When I clear a dialog from the TFT I'd like it to clear from Octoprint at the same time. Unless that's what "the checkbox that it is not the exclusive host" does (I haven't gotten around to that yet).

G2Barbour, 2020-06-21:
NOTE: there was an error in my configuration_adv.h where the extruder fan was not defined.

foosel, 2020-06-22:
> Ive forwarded some details to @foosel over discord.

Thanks, but I'll also need an `octoprint.log` and a `serial.log`, otherwise all of this is just blind guessing. I just tested the functionality and it jumps the queue just fine over here if configured accordingly:

[image]

    2020-06-22 09:20:10,458 - octoprint.util.comm - INFO - Force-sending M876 to the printer

Note that I have that still default to `false` in current release versions and it will only force send if there's an `S` parameter in the line (so a selection has been made), NOT when signaling general support with the `P1` parameter during capability exchange. 

Also note that your way of setting it as a emergency command here:

[image]

works just fine on my end as well:

    2020-06-22 09:25:40,314 - octoprint.util.comm - INFO - Force-sending M876 to the printer

Important for this is that the firmware reports `Cap:EMERGENCY_PARSER:1` which I saw in the example that you shot me over discord.

> I'm assuming he's an Octoprint guy?

She.

G2Barbour, 2020-06-25:
To clarify the issue, it is independent of Octoprint and happens any time the M600 script is run.

If I do not define an LCD controller, PurgeMore/Continue selections do nothing at all. Print immediately resumes after the Nozzle Parked dialog. If I define an LCD controller, all operates as expected.

To be honest, knowing that I would not have access to M73 without defining an LCD controller sort of makes me glad this is an issue. I do plan to incorporate the option for the TFT to display a print percentage, even from Octoprint. So at this point I just thought you should be aware that it is not functioning the way you said it is supposed to earlier in the thread. 

And on another note, while printing from Octoprint it would be nice to know what z height the printer is at. I have modified M155 to autoreport XYZE just like an M114 and it seems to work well but maybe you have a better idea. 

Ultimately, it would be  more elegant to autoreport M114 and release M73 for use without LCD support, in addition to negating the need for defining an LCD for our setups, but I do have it running to my satisfaction so no harm no foul if that doesn't happen. I'm just glad the new branched don't require backend mods to run M600! 

Or maybe I can convince foosel to add an option to echo XYZ coordinates and print progress % while printing! That would limit the functionality to Octoprint, but that's all people talk about using anyway.

thinkyhead, 2020-06-27:
> Or maybe I can convince foosel to add an option…

OctoPrint has a complete library of plugins to add extra behaviors.

> If I do not define an LCD controller, PurgeMore/Continue selections do nothing at all

See if it helps to make this change in `Conditionals_adv.h`…

```diff
- #if ANY(EXTENSIBLE_UI, NEWPANEL, EMERGENCY_PARSER, HAS_ADC_BUTTONS)
+ #if ANY(EXTENSIBLE_UI, NEWPANEL, EMERGENCY_PARSER, HAS_ADC_BUTTONS, HOST_PROMPT_SUPPORT)
    #define HAS_RESUME_CONTINUE 1
  #endif
```

G2Barbour, 2020-06-27:
No joy, behavior the same:

Send: M109 S180
echo:busy: processing
ok
wait
 T:179.65 /180.00 B:22.50 /0.00 @:16 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0

Send: M600
//action:paused
//action:prompt_end
//action:prompt_begin Pause
//action:prompt_button Dismiss
//action:prompt_show
 T:177.23 /180.00 B:22.54 /0.00 @:52 B@:0 W:?
 T:177.21 /180.00 B:22.46 /0.00 @:52 B@:0
X:0.00 Y:0.00 Z:0.00 E:-2.00 Count X:0 Y:0 Z:0
echo:busy: processing

echo:Insert filament and send M108 <-----M108 doesn't work. M876 S0 does.
//action:prompt_end
//action:prompt_begin Nozzle Parked
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
 T:180.77 /180.00 B:22.62 /0.00 @:31 B@:0
X:0.00 Y:0.00 Z:0.00 E:-72.00 Count X:0 Y:0 Z:0

Send: M876 S0
M876 Responding PROMPT_FILAMENT_RUNOUT_CONTINUE
 T:180.63 /180.00 B:22.58 /0.00 @:33 B@:0
X:0.00 Y:0.00 Z:0.00 E:-42.00 Count X:0 Y:0 Z:0
echo:busy: processing
//action:prompt_end
//action:prompt_begin Paused
//action:prompt_button PurgeMore
//action:prompt_button Continue
//action:prompt_show
//action:resumed

Send: M876 S0

ok <----expected: "M876 Responding PROMPT_FILAMENT_RUNOUT\nok\n", got: "ok\nok\n"
ok
wait <--------------------STILL NO FILAMENT ON PURGE MORE, NO LONGER "busy:"
 T:178.85 /180.00 B:22.58 /0.00 @:49 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
//action:prompt_end
//action:prompt_begin Paused
//action:prompt_button PurgeMore
//action:prompt_button Continue
//action:prompt_show
M876 Responding PROMPT_FILAMENT_RUNOUT
ok
 T:178.94 /180.00 B:22.58 /0.00 @:48 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
wait

<--------------------STILL NO FILAMENT ON PURGE MORE
Send: M876 S0
//action:prompt_end
//action:prompt_begin Paused
//action:prompt_button PurgeMore
//action:prompt_button Continue
//action:prompt_show
M876 Responding PROMPT_FILAMENT_RUNOUT
ok
 T:179.94 /180.00 B:22.46 /0.00 @:39 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
wait

Send: M876 S1
M876 Responding PROMPT_FILAMENT_RUNOUT
ok
 T:181.15 /180.00 B:22.62 /0.00 @:28 B@:0
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
wait

Roxy-3D, 2020-07-28:
>>  I've forwarded some details to @foosel over discord. 

> Awesome! I'm assuming he's an Octoprint guy?

 Bad assumption.

PrintEngineering, 2020-07-28:
> > > I've forwarded some details to @foosel over discord.
> 
> > Awesome! I'm assuming he's an Octoprint guy?
> 
> Bad assumption.

Thanks I'm sure that was necessary. I've already spoken to her about it.

Papa-LF, 2020-08-01:
Found this thread - strongly believe it will answer the problems I have with my recently installed run-out sensor.

Hopefully the thread will stay in place till I get project completed. 

Lowell

PrintEngineering, 2020-08-01:
> Found this thread - strongly believe it will answer the problems I have with my recently installed run-out sensor.
> Lowell
I am able to get full functionality, even with the quirk that I have to define an LCD controller without actually having one. Luckily my board has enough memory to pull it off for now.

Papa-LF, 2020-08-01:
Sweet!

Roxy-3D, 2020-08-01:
> I am able to get full functionality, even with the quirk that I have to define an LCD controller without actually having one. Luckily my board has enough memory to pull it off for now.

One thing I considered doing was allowing the user to specify an endstop switch to substitute as a click of the LCD Panel's encoder wheel.   That way, people without an LCD can still give the system a 'click' to indicate they are past a particular phase of the filament change sequence.

It really should not be much code or complexity to do that.   But the thing that has kept it from happening is so few printers are missing an LCD Panel now days.

PrintEngineering, 2020-08-01:
Host prompt support is more than capable of handling the situation. It would just be nice if it could be fully operational without defining an LCD controller. the entire artillery 3D printer line is lacking an encoder and they just released version 5 of The Sidewinder exactly the same

Papa-LF, 2020-08-02:
For the sake of discussion - I have an LCD on my MakerGear M2 Dual Rev E.  The challenge has been incorporating the Run-Out switch into the code. 

I've looked but a code modification/installation thread eludes me.  I am currently close - but not close enough.  
Lowell

PrintEngineering, 2020-08-02:
Can you put the sensor on the main board?

The next best option is to have the lcd send an M600 when the filament runs out

Papa-LF, 2020-08-02:
Yes, the sensor is on the mainboard - the RAMBo board and the board supports the function, I have narrowed down the issue to the version of Marlin loaded on the system
[image]
V1.0.2

The last piece is enabling the Advanced Pause Feature - which I cannot find in V1.0.2
[image]

All that said I would love to upgrade to a newer version of Marlin - but am skeptical I can get all the settings correct, and MakerGear has no path forward for that upgrade at this time - to my knowledge.

PrintEngineering, 2020-08-02:
I am unfamiliar with that system or it's limitations. You may want to keep an eye on free memory if it running Marlin 1.0 though

Roxy-3D, 2020-08-02:
> All that said I would love to upgrade to a newer version of Marlin - but am skeptical I can get all the settings correct, and MakerGear has no path forward for that upgrade at this time - to my knowledge.

You might try doing a M503 and see how complete a dump of all the settings you get.    Because if you get a fairly complete dump of all the magic numbers...   Turning on or off what ever other features you want should be fairly tame.

Papa-LF, 2020-08-02:
Thanks for that advice - nice to see folks still want to share their madd
skillsets.

I'll do some more digging and kick the idea around some more. I'm going to
touch base with the technical team that makes the printer and see what
information I can drag out of them.

Take Care - be safe...

(my results)

SENT: M503

READ: echo:Steps per unit:

Steps per unit:

READ: echo: M92 X88.88 Y88.88 Z1007.70 E471.50

M92 X88.88 Y88.88 Z1007.70 E471.50

READ: echo:Maximum feedrates (mm/s):

Maximum feedrates (mm/s):

READ: echo: M203 X200.00 Y200.00 Z10.30 E25.00

M203 X200.00 Y200.00 Z10.30 E25.00

READ: echo:Maximum Acceleration (mm/s2):

Maximum Acceleration (mm/s2):

READ: echo: M201 X1000 Y1000 Z50 E2000

M201 X1000 Y1000 Z50 E2000

READ: echo:Acceleration: S=acceleration, T=retract acceleration

Acceleration: S=acceleration, T=retract acceleration

READ: echo: M204 S2000.00 T3000.00

M204 S2000.00 T3000.00

READ: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate
(mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s), Z=maximum Z
jerk (mm/s), E=maximum E jerk (mm/s)

Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s),
B=minimum segment time (ms), X=maximum XY jerk (mm/s), Z=maximum Z jerk
(mm/s), E=maximum E jerk (mm/s)

READ: echo: M205 S0.00 T0.00 B20000 X4.00 Z0.40 E1.00

M205 S0.00 T0.00 B20000 X4.00 Z0.40 E1.00

READ: echo:Home offset (mm):

Home offset (mm):

READ: echo: M206 X11.11 Y0.00 Z10.97

M206 X11.11 Y0.00 Z10.97

READ: echo:PID settings:

PID settings:

READ: echo: M301 P25.89 I1.94 D86.53

M301 P25.89 I1.94 D86.53

READ: ok

-*-- Lowell ---*

closed 2020-09-07 by github-actions[bot]