MarlinFirmware/Marlin #11282: [1.1.x-bugfix] UBL - Editing mesh with LCD causes erratic Z behavior and crash

mmotley999, 2018-07-16:
There seems to still be bugs with mesh editing via the LCD when using UBL.  Here's the details:

- CR-10S
- Running 1.1.x-bugfix pulled on July 9th (includes the July 8th commit to 'fix' the LCD UBL, and it did... somewhat, more on that below).
- Using UBL, successfully by the way, with a 7x7 mesh.  Inductive sensor that is very stable and has very low standard deviation shown in M48 output. Distance from nozzle is accurate and measured, it to the right and slightly behind the nozzle.

I have a working mesh via G29 P1 then G29 P3.  However, the edge (left-hand side in my case, or lower X values) where P3 calculated the values seems a little off... so I wanted to adjust them.  The rest of the mesh is fine, I'm able to print with consistent 1st layer adhesion (I really like UBL!)

Using the LCD Unified Bed Leveling -> Edit Mesh (I believe, I'm not in front of the printer) shows a graphical screen with the mesh represented, and moves the nozzle to the lower-left corner.  I can rotate the dial and move the nozzle around (this wasn't working before the July 8th commit, this screen would just immediately hang the printer before).

The problem comes if I click the encoder wheel to edit a point... in this case, the first mesh point (lower-left corner).  When I click the encoder to edit, several things happen:

1. The Z axis does something strange.  Once it went up (away from the bed), the second time it went down and crashed into the bed and tried to keep going.....

2.  The LCD (assume the AVR) also crashes and requires a power off reset.

I've verified that UBL is active, stored in S1.

I haven't gotten my head around how to fine-tune a point yet, since I was very lucky and the G29 P1 / P3 combo "just worked" for the most part, which is why I tried to use the LCD.

Here's my G29 T. I'm looking to fine-tune the left-most values, where were calculated from G29 P3.

Recv: Bed Topography Report:
Recv: 
Recv:     ( 10,290)                                      (290,290)
Recv:         0       1       2       3       4       5       6
Recv:  6 | +0.374  +0.319  +0.265  +0.142  +0.112  +0.023  -0.341
Recv:    |
Recv:  5 | +0.528  +0.358  +0.188  +0.164  +0.042  -0.082  -0.463
Recv:    |
Recv:  4 | +0.492  +0.365  +0.237  +0.117  -0.026  -0.135  -0.520
Recv:    |
Recv:  3 | +0.540  +0.340  +0.141  +0.017  -0.135  -0.273  -0.524
Recv:    |
Recv:  2 | +0.510  +0.320  +0.130  -0.014  -0.131  -0.245  -0.478
Recv:    |
Recv:  1 | +0.473  +0.307  +0.141  +0.016  -0.098  -0.197  -0.413
Recv:    |
Recv:  0 |[+0.473] +0.262  +0.115  +0.016  -0.091  -0.138  -0.330
Recv:         0       1       2       3       4       5       6
Recv:     ( 10, 10)                                      (290, 10)

And yes, my bed is not super flat (I'm using a second aluminum build plate with PEI over the main bed).  This mesh actually prints quite well, except for the lower left corner where I'm getting more "squish" (I do a nozzle prime there).

Thanks!

thinkyhead, 2018-11-05:
Is this any better in the latest `bugfix-2.0.x` branch?

mmotley999, 2018-11-05:
I actually haven't tested the 2.0.x-bugfix branch at all, as I've been running 1.1.9 and frankly haven't tried this feature since I posted it; UBL seems to be working just fine without any adjustments.  I may be able to give 2.0 a try later this week, as my printer is currently offline being rebuilt with new extruders and the like.

boelle, 2019-02-20:
@mmotley999 did you give 2.0 a try?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

mmotley999, 2019-03-06:
Yes, I'm closing.  I haven't had a chance to install 2.0 yet, but once I do I'll test and re-open the issue if it occurs then.  Thanks all.

closed 2019-03-06 by mmotley999