MarlinFirmware/Marlin #11283: [Bugfix 2.0] Error in Stm32f4

mjhejazi, 2018-07-16:
hello 
I build Bugfix 2.0 for stm32f4 in platformio
but error in build
*** [.pioenvs\STM32F4\src\src\HAL\HAL_STM32F4\HAL_Servo_STM32F4.cpp.o] Error 1
please check out this part

xC0000005, 2018-07-18:
STM32F4 requires building in Arduino after installing support for the official ST arduino core - if you look back through the commits to that file, the one which changes that HAL to use the official core has some information.

mjhejazi, 2018-07-19:
I compile in arduino 1.8.5 for stm32f407zg In case of that disable smart graphic LCD No mistake.
But if I enable the display in the compile of the error long linker compile.
I use https://github.com/danieleff/STM32GENERIC ,but in platformio does not support this.

closed 2018-11-05 by thinkyhead