MarlinFirmware/Marlin #11284: [2.0.x] Problem in Travis build for PRs

TheLongAndOnly, 2018-07-16:
Travis builds on PRs currently fail due to platform.io dependencies:

.piolibdeps/Arduino-L6470/L6470/L6470.cpp:12:19: fatal error: L6470.h: No such file or directory

Resolution of dependencies works on my local clone.

Roxy-3D, 2018-07-17:
Oooops...   I didn't see this thread.   I kind of created a duplicate thread:   https://github.com/MarlinFirmware/Marlin/issues/11289

Recommendations on the best way to fix it are welcome.    What Pull Request caused it?   Is it something small so it makes sense to just revert it?    Or can we just fix the problem more easily?

I don't know (or understand) what the PlatformIO problem is.

soimy, 2018-07-17:
Just submit PR#11288 for a fix

Roxy-3D, 2018-07-17:
> Just submit PR#11288 for a fix

We will know soon if that fixes it.   I merged it.  And kicked off some Travis checks.

soimy, 2018-07-17:
Oh, at least the fix works for me

Roxy-3D, 2018-07-17:
But THANKS for the help!!!!

soimy, 2018-07-17:
You are welcomed, sorry that didn't help. I'm struggling to get my RAMPS FD V2 up and running 32bit marlin

Roxy-3D, 2018-07-26:
What is that?  Is that a null byte just before the new line?

soimy, 2018-07-27:
No, just indicate the recent commit which changed the header location. That result in PIO complaining missing header file.

closed 2018-11-05 by thinkyhead