MarlinFirmware/Marlin #11286: Strange regular infill shifting

As3uS, 2018-07-16:
Hi after verifying / change mecanical parts i come for aksing you why my infill have blank space.
It's not excatly layershifting because next layer come perfecly over the previous. 
And infill never cross the border

Try Cura, S3D an different stl files it's the same result and same pattern what do you think ?
Change nema, pulley, verify belt tension, steps 16/32, speed and always the same result.

Material : Mks Sbase, mgn12H, Core XY
Firware: bugfix 2.0.x July 2018

[image]

[Marlin-bugfix-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/2199659/Marlin-bugfix-2.0.x.zip)

ManuelMcLure, 2018-07-16:
I've seen behavior like this if the nozzle is too close to the bed as one row pushes/pulls the adjoining row. Does this happen only on the first layer or do you see it also on higher layers?

As3uS, 2018-07-17:
I, yes sorry after many test and long Time if test i do a quick offset for the capture it's not accurate but with correct offset calibration it's the same resuIt. 

I wait 3 layer count and the shifting flip 90° he follow the infill pattern you des on the picture. But my part ils correctif formed with ugly blank Space :)

I trying to print a part with hole and on the second layer the printhead is at the good place and without any shifting. 

It's only during infill not on the contouring.

I Hope you understand because it's a very worse english ;)

As3uS, 2018-07-17:
Sorry i make a mistake closing

As3uS, 2018-07-17:
Solved, linear bearing on my extruder seems to live his life alone.

closed 2018-07-17 by As3uS