MarlinFirmware/Marlin #11289: Bugfix 2.0.0 branch is sick...

Roxy-3D, 2018-07-17:
It looks to me like the Bugfix 2.0.0 branch is sick.   It doesn't matter if you just add a comment to one line of code.   You won't be able to get it to pass the Travis checks.

I guess there are a couple of ways to fix things.   One idea might be to find where things were merged without Travis approving them...  And revert those changes.

Or maybe somebody can figure out an easy way to resolve the root cause problem?   But right now, it looks to me like anybody doing a Pull Request against bugfix_2.0.0 can't pass the Travis tests.

teemuatlut, 2018-07-17:
Can you try using the `master` branch for PIO instead of `develop`? This looks more like a toolchain issue rather than Marlin related one.

ejtagle, 2018-07-17:
This is a toolchain issue 👍  -- Something in the teeny35 target is preventing the u8glib to properly compile, probably a missing include path...

Roxy-3D, 2018-07-17:
> This is a toolchain issue

Yeah...  Because the Pull Request I just did today was fine on AVR's using Arduino.   But some how...  Travis is thinking there is a problem.

ejtagle, 2018-07-17:
Arduino-L6470 ,,, That is the project that is failing to compile

ejtagle, 2018-07-17:
If Travis is pulling from https://github.com/ameyer/Arduino-L6470, i am not surprised that it started to fail. Please look at the last commit to that library, 4 days ago! ... They changed the include path of that exact file!!

ejtagle, 2018-07-17:
Pretty interesting, @thinkyhead is confirming here : https://github.com/ameyer/Arduino-L6470/issues/2 that travis is pulling from this incomplete library...

teemuatlut, 2018-07-17:
It's possible the `L6470` library needs a hierarchy update that follows the spec from Arduino 1.5.x and newer.
Basically rename the `L6470` folder to `src`.
Move `examples`, `keywords` and `properties` to the root folder.
Optionally the `cpp` source should be moved to a subfolder from the header.

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#a-complete-example

Roxy-3D, 2018-07-17:
Is there something simple we can change so our stuff isn't broken while we figure out the 'right' way to fix this?

soimy, 2018-07-17:
@Roxy-3D I don't understand why it's not working on travis with PR #11288 ? I just check https://travis-ci.org/MarlinFirmware/Marlin/builds/404701649?utm_source=github_status&utm_medium=notification 
turned out it pass all the test.

Roxy-3D, 2018-07-17:
I don't understand it either.   but I merged https://github.com/MarlinFirmware/Marlin/pull/11288 and then updated the bugfix_2.0.0 Pull Request I was working on.   And failed the Travis checks.   It might be there is something else wrong with the Pull Request I was working on.    It is here:  https://github.com/MarlinFirmware/Marlin/pull/11285

And in the description of errors, Teensy is front and center.   Without your patch, if I remember right there were 3 different errors.   So maybe your patch is good...  But it is not quite enough to resolve all the issues.    I don't know!

soimy, 2018-07-17:
Make sure travis isn't caching. Talking about Teensy, is there any HATS for 3dprinter available on the market?

p3p, 2018-07-17:
@soimy your PR fixed the build (apart from the cache issue) because PlatformIOs version of the library is older than the one we used directly not newer, your temporary solution (only thing we can do) was mostly correct though targeting `Arduino-L6470@0.6.1` would be better (or the commit hash directly), of course the actual solution is fixing the library. There is no hardware for Teensy35 as far as I'm aware. 

@Roxy-3D reverting the revert and clearing Travis cache ( on the top right under 'more options' menu) should fix this, targeting v0.6.1 for now would probably be better.

Roxy-3D, 2018-07-17:
I think everything is better now!   The bugfix_2.0.0 Pull Request I submitted yesterday now passes Travis checks!    https://github.com/MarlinFirmware/Marlin/pull/11285

:)

thinkyhead, 2018-07-18:
There was a patch to L6470 by a new contributor a few days ago. I was able (as a maintainer) to patch the L6470 library to fix the issue, and everything seems to work now. The L6470 itself has been half-done and unsupported and is only starting to take shape, so it's not in the Arduino or PlatformIO library databases. But we can use git links to specify particular versions if that helps prevent breakage in the future. I updated the repository with a new tag "`v0.6.2`".

closed 2018-07-18 by thinkyhead