MarlinFirmware/Marlin #11291: Makeboard mini requires G21 before full console/gui control

ijustwannafly, 2018-07-17:
I believe I am familiar enough with Marlin at this point.   I have flashed marlin multiple times on my two anet a8 boards and configured multiple items with no issue in finding resolutions.

I have recently acquired a [makeboard mini](https://www.banggood.com/Micromake-Makeboard-Mini-3D-Printer-Control-Board-Support-Hotbed-Compatible-with-RAMPS-1_4-p-1129188.html?cur_warehouse=CN)

I have flashed marlin to it successfully with a reprap full graphics smart controller.  It is configured for corexz and can print and move as expected.

The issue I am having/seeing is that on first startup/power-up I have no control via serial/pc connection except for a few M commands and G28.  Interestingly enough, any G commands for movement (G1 X150 F9000 for example) do not respond.  

I ensured marlin is sending "ok" after each command.  

Sequence testing:  Power-up printer, send M502 (defaults), send M500 (save), send G28 (success).

After G28 attempting to send G1 X150 F9000 results in no movement.  G1 Z0 F9000 will move Z to 0 as expected.

The trip here is if I issue a G21 immediately before or after the first G28 I can use all commands (console or gui) correctly.

I tested this multiple times and have come up with the same results.  G28 without a G21 and I cannot issue movement commands.  Send G21 before or after G28 and I have full control (console or gui).

Using repetier host and repetier server results in the same.  I have tried every baudrate possible, enabled the pingpong mode of repetier and tried other usb cables (from my working anet a8 boards that do not require pingpong mode and are using marlin1.1.8).

Is there a setting in marlin that I am missing?  I have configured my anet a8 boards with marlin many times over and have not had this issue happen before.

GMagician, 2018-07-17:
Are you sure you don't have INCH_MODE_SUPPORT enabled?

GMagician, 2018-07-17:
@thinkyhead  when INCH_MODE_SUPPORT  is defined variables are not initialized and `linear_unit_factor` is default 0. I'm fixing it and posting a PR

ijustwannafly, 2018-07-17:
@GMagician 

That was it!

That single define is not enabled on my anet boards.

I copied the configuration.h file from the micromake example and changed the other settings I needed for my use.

Just re-uploaded with that define disabled and all is well.

Many thanks!

GMagician, 2018-07-17:
Yes, but I'll fix this too

GMagician, 2018-07-21:
I think that you may close this now

PR #11294 (2018-07-18): [1.1.x] Initialize default units
When INCH_MODE_SUPPORT is enabled there is no default unit set, this fix it

Fix #11291
closed 2018-07-21 by ijustwannafly