MarlinFirmware/Marlin #11292: Reprap Discount GLCD & Sbase displays nothing?

forkoz, 2018-07-17:
I got my sbase set up. Flipped both EXP1 and EXP2 at one end. Nothing shows on the LCD. When I rotate the encoder it beeps loudly and reboots. The backlight never comes on. When I connected EXP1 unflipped the entire LCD would light.

On the MKS gen it worked as reprap discount just fine. Has there been some change or something else is missing? I thought people got this working successfully with the sbase. BTW I have disabled SDsupport so that shouldn't be a thing.

forkoz, 2018-07-17:
I took off the LCD and its based on LCD2004 :(  It looks like this: https://www.amazon.com/gp/product/B015NDAKE4/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

p3p, 2018-07-17:
So this is a character based LCD rather than a GLCD as you said in the title? also please always include configs as the issue template requests even if it seems pointless it could help.

forkoz, 2018-07-17:
Yes, it is a character LCD. It worked under reprap discount on 8bit but I took it out, saw LCD2004 on one of the boards and did more digging. Supposedly EXP1 is missing those 3 lines? I ordered the graphics LCD as it might be better than hacking up the cables or if nobody knows.

A little OT but I got pins working for end stop interrupt. I added P0_14 but this pin did not trigger and I ended up using TH4, + 17/16.

BTW, should edit the sample config to have serial set to -1, that took a bit of time to find. With no LCD + no USB it was a bit confusing. Eeprom for sbase sits on the SD card?

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2203095/configs.zip)

p3p, 2018-07-17:
You will have issues as you have found out, as with the Re-ARM, character displays are not compatible out of the box (without a bit of rewiring to connect LCD_D5, LCD_D6 and LCD_D7 ),

Interestingly P0_14 does not exist, that schematic (and the one I have) are wrong J7 Pin 6 is P0_15 ( same as LCD_D4) but I'm glad the other pins worked for you. 

Currently the eeprom data is stored on the on board sd-card yes, but I'm thinking about changing that as so many seem to have issues with sd cards.

I can see the wrong serial connection being selected in the example config being confusing, I'l see about changing that.

gloomyandy, 2018-07-17:
To add a little more input. I got my Sbase card and a Discount Graphic LCD a couple of days ago. I followed the steps for building using Platformio and built the standard sample configuration fine. Then updated the sd card and when I booted the device my PC found the USB devices and the LEDs flashed reassuringly! But I also hod the issue with the serial connection and so could not talk to the board, but remembered seeing mention of this in other issues so soon updated the config and had it working. I then rotated the two LCD connectors and now have the graphic LCD plugged in and working. Next up is to get the external SD card functioning, then I will start to check out motor control and endstops etc. So far it has been a pretty straight forward experience, great job everyone involved! I'm looking forward to getting things working fully and then hopefully contributing to the work on this platform. 

@p3p is there any sort of list of issues, or areas that you are looking for help with?

forkoz, 2018-07-17:
That would explain it. And then even if I select 0_15 it would share the LCD line.

So to make the LCD2004 work you would flip the connectors and then either solder at the back of the sbase to exp1 or break out the cable and then define:

#define LCD_PINS_D5 P1_22
#define LCD_PINS_D6 P1_23
#define LCD_PINS_D7 P2_12

in mks sbase pins h file.

The problem I have with the sd card is that it shows up in the computer, which is good for firmware flashing I think but seems to interfere with serial. Sometimes serial just stops responding and I have to re-plug the usb or reset the printer. SD seems solid the whole time.

Also M112 kill command locks the printer with all 4 lights lit and it needs both USB unplugged /power reset. There is a problem too with the board resuming whatever it was doing when stopped if USB is left connected. I got a bed crash today (loose probe connector) and cut the power but didn't pull the USB, the printer happily kept pushing the nozzle into the bed when I re-applied power. maybe these need separate bug reports

p3p, 2018-07-17:
> The problem I have with the sd card is that it shows up in the computer, which is good for firmware flashing I think but seems to interfere with serial. Sometimes serial just stops responding and I have to re-plug the usb or reset the printer. SD seems solid the whole time.

I've never seen usb stop responding until re-plugged, did the Heartbeat LED stop?, not sure of the top of my head what could cause that. 

> Also M112 kill command locks the printer with all 4 lights lit and it needs both USB unplugged /power reset.

That is by design, the board needs reset after a kill command (or button)

> There is a problem too with the board resuming whatever it was doing when stopped if USB is left connected

Side effect of you not using a kill signal, Marlin wouldn't even notice that the main power was turned off and because the board is powered by USB it just carries on. There isn't anything we can do in software to fix that. My printer has a giant red E-Stop button that I smash cuts main power and sends a signal to the estop pin to kill the board.

forkoz, 2018-07-17:
I'm LCD-less right now so can't hit the button. I'm used to using kill in S3d and the board rebooting. This cut all power thing is new to me coming from AVR.

Heartbeat keeps going but USB refuses to reconnect. I used both the marlin driver and the smoothie driver. I'll watch it more closely when it happens again.

I did M500 to save Z offset and just had this happen:

READ:  T:36.06 /0.00 B:60.19 /60.00 @:0 B@:0
READ: ok T:36.06 /0.00 B:60.19 /60.00 @:0 B@:0
READ:  T:36.21 /0.00 B:60.06 /60.00 @:0 B@:22
READ:  T:36.21 /0.00 B:60.01 /60.00 @:0 B@:35
SENT: M105
READ: ok T:36.21 /0.00 B:60.01 /60.00 @:0 B@:35
SENT: M105
SENT: M105
SENT: M105
SENT: M105
SENT: M105
SENT: M105
  Permissions error connecting to port
    Is there another application already connected to this port?

p3p, 2018-07-17:
@gloomyandy There isn't a list anywhere, mostly the issues we have are with usb, i2c, or spi, or the .. oh and Arduino library compatibility. I'm currently pulling out all the framework into its own project to keep the platform the same complexity as the others. The work in progress Marlin is [here](https://github.com/p3p/Marlin/tree/pr_bf2_pulloutlpcframework) which builds using [this](https://github.com/p3p/pio-framework-arduino-lpc176x) framework, as soon as I have it separated I can start working on getting the framework rewritten to be significantly more Arduino compatible and on a more up to date base, help is always appreciated once I get to a point to be able to accept PRs.

forkoz, 2018-07-21:
Ouch... now I have real GLCD and it displays nothing (screen is lit). The kill button works and reboots the board without power off however.

closed 2018-08-06 by forkoz

patternseek, 2018-10-06:
> The problem I have with the sd card is that it shows up in the computer, which is good for firmware flashing I think but seems to interfere with serial. Sometimes serial just stops responding and I have to re-plug the usb or reset the printer. SD seems solid the whole time.

I think I'm having a similar problem. With the onboard SD card plugged in when I plug the sbase into my pi it detects it, but if i power cycle the sbase the pi won't detect it again until I re-plug the USB. Removing the SD card solved the issue.