MarlinFirmware/Marlin #11296: Please, help me to upgrade Marlin Firmware from 1.1.x to 1.8.1

clarakim1, 2018-07-17:
Dear Marlin professional Engineers,

How are you!
I am Clara from South Korea.

I have been used 3D printer since I bought it in January.
It is really fun and interesting both to make my  own design and print from Thingivers.

But, nowadays, I have something to upgrade my 3D printer.
**That is RGB LED Strip project.**

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

thinkyhead, 2018-07-19:
For best results getting help with configuration and troubleshooting, please use the following resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

Roxy-3D, 2018-07-19:
Install NotePad++ and turn on the differential compare add on.
Go through the differences between you 1.1.x setup to 1.1.8.    You just move over the settings.

You do need to go through each difference and make decisions.   Some stuff has changed names.

jonathanweinberg, 2018-08-15:
This issue should likely be closed since there has been no activity on it.

closed 2018-08-21 by thinkyhead