MarlinFirmware/Marlin #11299: M851 Not setting offset for negative values

mmakam2, 2018-07-18:
I tried everything I am on 1.1.8. Tried in firmware, tried with m851. After running command, and running M500, G28, and M114, my z is still showing 0.

Is there any workaround for this?

forkoz, 2018-07-18:
Its supposed to be 0. First 0 is actually zero. Then when you M851 Z-1 new 0 is -1 but the machine still displays 0.

mmakam2, 2018-07-18:
I am getting an unknown command error

mmakam2, 2018-07-18:
[image]

mmakam2, 2018-07-18:
I also tried setting this value, but nothing happens: Z_PROBE_OFFSET_FROM_EXTRUDER

mmakam2, 2018-07-18:
BTW I am on a Wanhao D6..

mmakam2, 2018-07-18:
I also cannot manually move Z axis to negative values from the menu..

thinkyhead, 2018-07-18:
> I also cannot manually move Z axis to negative values from the menu.. both in 1.1.x and 1.1.8

Try setting your `Z_MIN_POS` to -5, but also set your `MANUAL_Z_HOME_POS` to 0. This will make your homed Z position 0, but also permit some freedom of movement below the homed Z0. After re-flashing the firmware, use `M502`, `M500` to ensure that the updated defaults are used.

thinkyhead, 2018-07-18:
If you're getting an "unknown command `M851 Z...`" then you may need to specify a probe type in `Configuration.h`.

mmakam2, 2018-07-18:
Ah yes, the probe type fixed the unknown command error

mmakam2, 2018-07-18:
Does the M851 command work the same way as setting this value: Z_PROBE_OFFSET_FROM_EXTRUDER?

ijustwannafly, 2018-07-19:
When setting my z offset for my probes I usually do this sequence:

G28 - home all axis
G1 X and Y to the center of the bed (depends on your dimensions)
G1 Z0 - send nozzle to 0 position
M211 S0 - turn off software endstops
Move nozzle down using LCD or PC controls until paper grips between the nozzle and the bed
Note the value needed to bring the nozzle to the bed ie -2.5
M851 Z-2.5 - set z offset
M211 S1 - turn software endstops on
M500 - save changes
G28 - home all axis 
G1 Z0 - to check if the offset stuck

My configuration for my aneta8 on marlin 1.1.8 has my z min pos at 0 (no negative values) with a fix mounted probe.

murdock62, 2018-07-20:
Just to be sure to understand:

 with AUTO_BED_LEVELING_BILINEAR
	
If i keep the "MIN_SOFTWARE_ENDSTOP_Z" activated with "Z_MIN_POS 0" :
	
- I could not manually lower the Z axis below 0.
	
- But during the printing if the topography of the bed requires it the machine can lower the nozzle below 0 to follow the surface of the bed.

- It is therefore not mandatory to disable the "MIN_SOFTWARE_ENDSTOP_Z" function to use the "AUTO_BED_LEVELING_BILINEAR" function ?

- The "MIN_SOFTWARE_ENDSTOP_Z" function is only for manual mouvements ?

Is this correct ?

Many thanks !

AnHardt, 2018-07-20:
Software endstops should be applied before the levelling. You can't go locally below zero but leveling may result in a value below of that.

ijustwannafly, 2018-07-20:
@murdock62 

You are correct in all your points.  If you are using a probe you still must set your Z offset which requires you to temporarily disable the software endstops.  After your Z offset has been set re-enable the software endstops.

@thinkyhead 

Now that I think about it...would the idea of enforcing the software endstops all the time be beneficial versus currently having the ability to enable/disable them in the configuration.h file since M211 does the same thing but does not need a recompile?

@AnHardt 

I agree software endstops, as a safety feature, should be enabled at all times.  I personally only disable them when setting Z offsets for probes **_OR_** measuring axis dimensions but I do so with M211.  By default as long as the software endstops min/max are enabled in the configuration.h any reset/reboot to the printer board will re-enable the software endstops.

AnHardt, 2018-07-20:
Even if configured so, software endstops should not be enabled by reboot/reset, but by homing. Before homing software endstops are at best use- and meaning-less.

ijustwannafly, 2018-07-20:
>Even if configured so, software endstops should not be enabled by reboot/reset, but by homing. Before homing software endstops are at best use- and meaning-less.

I only speak from what I have observed in my systems.  When I issue M211 S0 and do a G28 the software endstops are still disabled.  They stay disabled until I send M211 S1 **_or_** reset/reboot the board.

I did not mean to imply that if you want to reset the software endstops after disabling with M211 that you should reset/reboot the board.  What I meant is that if you forget to re-enable them for whatever reason and decide to power down the machine the endstops will be re-enabled by default on the next power cycle.

murdock62, 2018-07-21:
@ijustwannafly 
@AnHardt 
Thank's a lot for your help.  
so you confirm my observations. Thanks again and good prints!

closed 2018-11-05 by thinkyhead