MarlinFirmware/Marlin #11300: ### Request Menu item options on GLCD Controller

bruce356, 2018-07-18:
### Request
Hi, I have a request regarding Menu item options on GLCD Controller.
Dual-X carriage printer.
Reprap discount full graphic smart controller.
Genuine BLTouch.

Its easy to adjust T0 offset (fine tune) in relation to BLTouch and bed from the GLCD,
but T1 is a pain, have to connect the computer and use M218 to fine tune (Why).
Why is this option not available from GLCD as the next item under "Probe Z Offset".
This is not a once off setting that is rarely used, every time I change nozzles (can be several times a week due to material or nozzle size) T0 is easy to re-tune but T1 is not.

One other minor comment (probably not important), why can PID settings for both hot ends be changed through GLCD but the Bed PID is not available.

Thanks and regards - bruce

gloomyandy, 2018-07-18:
Perhaps a silly question but have you enabled PID use for the bed (by default it may be using bang/bang so there are no PID params to change)? Posting your configuration files and details of what versions of Marlin you are using (always a good idea to try the latest bugfix if you can) will make it easier for folks to help.

bruce356, 2018-07-18:
Hi, gloomyandy thanks for the reply, yes I am using the latest Bugfix v1.1.x and yes I have enabled PID for the Bed & hotends. #define PIDTEMP #define PIDTEMPBED.
I did not include my config files because as a request to extend options on GLCD I thought it would not be necessary.
Regards - bruce
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2207637/Marlin.zip)

InsanityAutomation, 2019-05-25:
@thinkyhead this one can be closed. Its done.

boelle, 2019-06-22:
@thinkyhead Another one that needs to be closed. Sorry, I got bored.

InsanityAutomation, 2019-10-15:
@boelle @shitcreek another done - close

closed 2019-10-15 by boelle