MarlinFirmware/Marlin #11303: Add #define constants to M503 output

eyal0, 2018-07-18:
### Description

Add indication into M503 output if `S_CURVE_ACCELERATION` or `LIN_ADVANCE` or other constants have been set.

### Steps to Reproduce

1. Compile Marlin with S_CURVE_ACCELERATION _disabled_.
2. Run and look at M503 output.
3. Compile Marlin with S_CURVE_ACCELERATION _enabled_.
4. Run and look at M503 output.

**Expected behavior:** [What you expect to happen]

Expect to see some indication in the M503 output that S_CURVE_ACCELERATION is enabled/disabled.

**Actual behavior:** [What actually happens]

No difference in the M503 output

#### Additional Information

I'm working on [software](https://github.com/eyal0/Marlin/tree/estimate-bugfix-2.0.x) that runs a Marlin simulation on gcode and outputs information such as time to print and filament length used.  The simulation uses actual Marlin code that is modified to run only the planner and not drive any steppers.  Instead, it just adds up the time for each block in the planner.

The software takes gcode as input and can read M200-M205 configuration from it to configure the planner's jerk, acceleration, etc.  The M200-M205 information is taken from the M503 output of the printer to simulate.

Options like S_CURVE_ACCELERATION affect the calculation, but there is no way to determine from the M503 output if S_CURVE_ACCELERATION is enabled/disabled.

The solution is just one line, something like:

```
CONFIG_ECHO_START;
SERIAL_ECHOLNPAIR_P(port, " S_CURVE_ACCELERATION: , ENABLED(S_CURVE_ACCELERATION)
```
Added in configuration_store.cpp, around [here](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/configuration_store.cpp#L2150).

It is important that this is **not** in an `ifdef` because we want to be able to distinguish these three cases:
1. S_CURVE on
2. S_CURVE off
3. S_CURVE value not displayed (old code)

I prefer that the `ENABLED` is in that code so the output is always a `0` or `1` and post-processors don't need to know about all the other values congruent to 0/1 like false/true, 0x0/0x1, etc

This and LIN_ADVANCE and other constants could all be displayed the same way, maybe in a block near the end of the M503 output called "Constants".

eyal0, 2018-07-18:
The need is based on this comment: https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/35#issuecomment-406060805

thinkyhead, 2018-07-19:
M115 would be the better choice, but (1) that output is meant to advertise to hosts that there are added G-codes for their benefit, and (2) everything that gets added to that output is added for all builds, so we have to be conservative in its use.

eyal0, 2018-07-19:
Right, this isn't added gcodes, just compile options. So it doesn't exactly fit the definition of M115 but it's close. I read that M115 isn't we'll defined anyway.

What do you mean by point number two? Are you worried about the size of the binary or the length of the output? It seems like it would be really small in the binary, just one function call.  One line about jdev would be far fewer bytes than the actual implementation. How constrained is the binary size?

I also consider the debugging implications. If you ask users to provide the M503 or M115 output in debug reports, you'll get useful info about the build.

AnHardt, 2018-07-19:
> Marlin code that is modified to run only the planner and not drive any steppers. Instead, it just adds up the time for each block in the planner.

How do you determine the actual length of the block queue? If not always full, or long enough, that is likely to make more difference than if S-acceleration is used or not.

AnHardt, 2018-07-19:
I agree. Serial communication is an other source for possible delays.
I was talking about the buffer in between the stepper IRQ and the planner. Last segment there in always ends with ~zero speed. Depending on way/time in the buffer and ability to accelerate/decelerate the target speed is not reached.

eyal0, 2018-07-19:
@AnHardt I don't exactly understand the buffer point.  My code runs the planner and inserts blocks until the buffer is full.  Eventually idle() is called because the buffer is full and I grab the next block in the queue, which has in it the start velocity, end velocity, acceleration, time, etc.  From those, the trapezoidal velocity graph can be made and the area under that graph can be found.

I didn't write the original code, I just adapted it to bugfix-2.0.x and fixed some bugs in it.  This should take into account the inability to reach the target speed.  In fact, on small moves, I see that the final speed isn't necessarily the target junction speed, for example, if the motion is continuing in the same direction.

I don't need to be exact, I'm just trying to get as close as possible.  If you have ideas how to improve, I'm listening!

eyal0, 2018-07-22:
How could I get the most accurate simulation?  Right now I'm reading all the M200-M205 from the printer, running the planner, and then adding up all the acceleration_time + cruise_time + deceleration_time.  For each one, I use the number of steps and divide by the rate to get the total.

Any creative ideas on how to do better?  Are there effective tools for measuring the actual time spent moving versus programmed?  For example, I could attach an oscilloscope to the stepper drivers and see if they are able to achieve the expected rates.  Would that work?

thinkyhead, 2018-07-24:
> What do you mean by point number two?

Often PROGMEM strings take up more space than code, so we try to avoid their use in always-enabled features. This is nicer for boards with 128K or less flash. Since every `Cap:` entry we add to the `M115` command exists for all builds (unless capabilities reporting is itself disabled) we prefer to avoid adding new ones unless it provides information useful to a host.

eyal0, 2018-07-25:
@thinkyhead  Okay, I get. You guys are really stingy with the bits! :laughing: 

In the end, I was able to prove to myself that the time to print with and without s curve acceleration is the same, so no need for it in the end. I made a new branch of bugfix-2.0.x that runs a simulation and at the end I divide the number of clock cycles by 27500000 and get an accurate estimate of the number of seconds that a print will take.

However, it's still not perfect and I can't figure out why. The only possibility that I can think of is a difference between the 32 bit CPU versions of some of the functions and the AVR assembly optimized ones. Maybe? Can you think of something else? Clock drift, does that happens?

thinkyhead, 2018-11-05:
It's very complex code, and I'm sure very hard to get accurate calculations of clock cycles when there are so many potential code paths.

closed 2018-11-05 by thinkyhead

eyal0, 2018-11-05:
@thinkyhead Yup.  Well, it's pretty good.  Maybe one day I'll sit with a logic analyzer and find why it isn't perfect.