MarlinFirmware/Marlin #11304: [2.0] G33 Uneven Leveling and G29 smashing into the bed

wailord284, 2018-07-18:
Hello, I am using the latest Marlin 2.0 Bugfix and when attempting to level my Delta using a BLTouch and Smart Ramps, the end result has one side of the bed significantly lower than on the other side having only about 3x3 inches in the middle perfectly flat. I've also tried using G29 which seems to create the level just fine but then once I begin a print, smashes into the bed regardless of what value I specify for Z_PROBE_OFFSET
### Steps to Reproduce for G33
1. Run M502 M500
2. Run G33 P3 V2 (Or P4-10 doesnt seem to make a difference)
3. Save result with M500
4. Print and see uneven level
### Steps to Reproduce for G29
1. Run M502 M500
2. Run G29
3. Save result with M500
4. Print and see nozzle hit bed
**Expected behavior:**
The print should be flat and have correct Z offset

**Actual behavior:** 
Nozzle either slams into bed or the level is uneven

#### Additional Information
Maybe this is an error with my configs so I attached them. Thanks a lot for all the hard work!
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2207497/Configuration.zip)

AnHardt, 2018-07-19:
Like on cartesian systems `Z_PROBE_OFFSET_FROM_EXTRUDER` has to be constant all over the bed.
For DELTAs the most common way to till the effector by accident is to have uneven length diagonal rods.
If the two rods mounted to the same tower are not the same length, there is currently no way to compensate for it.

wailord284, 2018-07-19:
Well I was able to get a print off and it is the exact dimensions it is supposed to be. I cut the rods myself from carbon fiber and rechecked and they are all 450mm. But even if my rods were the wrong length, I still don't understand why G29 smashes into the bed and continues to move down

wailord284, 2018-07-20:
I was able to fix my G29 issue, turns out I wasnt enableing the mesh G29 created. To do that you need to add M420 S1 to the start of your print. You can save G29 to eeprom and just run M420 S1 every print. Thanks again!

closed 2018-07-20 by wailord284