MarlinFirmware/Marlin #11307: ABL Bilinear - Probe Manually - Probe start point does not restart at 0

ijustwannafly, 2018-07-19:
I configured ABL Bilinear to work with Probe Manually and LCD Bed Leveling options.

It works except for one behavior I have observed.

When starting the bed leveling from the LCD it will home (of course) and then prompt "click to begin".

When starting at point 1 the z axis is set (or starts) at 0.000.  When I adjust the first point down for example -2.000 and then proceed to the next point (point 2) the z axis starts or "sets" itself at the last offset ie -2.000 from the first point.

As the process progresses there are points that will be higher or lower than the previous point causing the nozzle to "crash" into the bed until the offset is adjusted up.

Since it is a manual process shouldn't it be logical to start every probe point at Z 0.000 and then adjust down?

thinkyhead, 2018-07-19:
The procedure has been done a few different ways over time, but settled at some point on the current behavior. I've submitted the PR #11308 to make the height configurable.

ijustwannafly, 2018-07-19:
@thinkyhead 

I should be able to just make the changes to the two files (configuration.h and Marlin_main.cpp) in order to test correct?  Or should I download the bugfix branch?

thinkyhead, 2018-07-24:
> should I download the bugfix branch?

That will be simplest.

closed 2018-07-24 by thinkyhead