MarlinFirmware/Marlin #11312: [2.0] MKS Gen 1.4 + TMC2130 + MKS Mini 12864 Driver error with SDSUPPORT

Lyr3x, 2018-07-19:
I'm using TMC2130 for X, Y and double Z and the MKS Mini Panel. Tried today the first time to activate the SD Support. But than i'm getting an instant driver error and i can see that the Y driver is not initalizing correct
```Y = 0xFF:FF:FF:FF```

Without SD Support everything is working fine for me.

[config.zip](https://github.com/MarlinFirmware/Marlin/files/2211366/config.zip)

MikeLud, 2018-07-19:
@Lyr3x  you can try the attached wiring it is almost the same board, make sure you do pin changes in note 2

[MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf](https://github.com/MarlinFirmware/Marlin/files/2211866/MKS_Gen-L_TMC2130_SPI_Sensor-less_Homing_Wiring_Diagram.pdf)

Lyr3x, 2018-07-20:
Awesome. It's working :)

Lyr3x, 2018-07-20:
Shouldn't we change the pin definition for the MKS_13 board? Check if tmcs or other spi driver are enabled and than override the pin definition ?

MikeLud, 2018-07-20:
@thinkyhead and @teemuatlut  What are your thoughts on change the X & Y CS in the "pins_RAMPS.h" file to 59 and 63 this will address the issues that people are having SD and TMC. Also it will make it a lot cleaner having the CS on aux2 and SPI on aux3. Below is my original recommendation that we were waiting for a response from @makertum 
444725ae9a669c2ab87500346be147802b6a6f20

Lyr3x, 2018-07-20:
But shouldn't we "automate" those basic setups that the user don't need to configure pins at all? Just an idea.

teemuatlut, 2018-07-21:
That's exactly what we're doing and as you can see the definitions are already enclosed in `#ifndef` in the pins files. Only thing that's missing is adding a section to the config file.

zarthcode, 2018-07-26:
I've run into this same issue - MKS GEN 1.4 w/TMC2130 drivers, and a reprapdiscount display w/sd.  The default cs pins 53 & 49 are simply a no-go w/SD support. Pins 63/53 on Aux-2 are a better pick - for MKS boards, at least.

wbarber69, 2018-07-27:
If you follow my reference issue at the tmc drivers github.  I use the servo pins 3,4,5,6 for cs and I use a similar pin configuration to aux3 on aux2 which also carries my fifth cs connection for my mks gen-l. I prefer this method because I only need to make two simple connectors a 4 pin and an 8pin plug.  I initially did it this way with the intent to build a pcb that sat atop the board but I haven’t gotten around to it yet.  Every board and every printer will have/need a different pin layout depending on how and what you have hooked up.  So I prefer the manual way of defining things.

HappyCodingRobot, 2018-07-30:
IMHO there is definitely no need to change  some of the `_pins.h` files. As an quick bug fix you can set your board config to MKS Gen L. These Boards are almost identical and this will work out of the box with TMC2130 over SPI and SDSUPPORT. (see wiring diagram from MikeLud)
However, the `pins_MKS_GEN_13.h` and `pins_MKS_GEN_L.h` should be merged. Also the VIKI settings in here doesn't look right.

closed 2018-09-06 by thinkyhead