MarlinFirmware/Marlin #11314: get_period_inverse divide-by-zero on non-AVR

eyal0, 2018-07-19:
### Description

This line can divide by zero: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/planner.cpp#L697

### Steps to Reproduce

Clone: https://github.com/eyal0/Marlin/tree/estimate-bugfix-2.0.x

It's already fixed in that branch so undo the fix.

```
Run:
make -f Makefile.calc
./calc SOME_GCODE_FILE.gco
```

**Expected behavior:** No crash.

**Actual behavior:** Crash.

#### Additional Information

The get_period_inverse code for AVR returns the same value for input 0 and input 1, so maybe the non-AVR can do the same?

Also, I don't understand why AVR is computing 0x1000000/d and non-AVR is computing 0xFFFFFFFF/d.  The difference is nearly a factor of 16!

p3p, 2018-07-19:
Interesting I had hit a divide by zero issue on my Linux port, hadn't got round to looking into the root cause.

ejtagle, 2018-07-21:
@eyal0 : i did write that function. Basically, if the function is called with a 0 argument, it is a bug. It should never happen.
But, i have nothing against protecting against division by 0.

AVR computes 0z10000000/d because the S-CURVE implementation on AVR has reduced precision (that was the only way to make run in realtime... The "error" of that reduced precision is less than 1 on all calculations, and using a 24bit result saves TONS of 8 bit multiplies. TONS, as AVR is 8 bit and a 32x32 multiplication requires way more 8bit multiplications than 24x24 multiplications

On ARM, 32x32bit multiplications are native, so just 1 cycle to compute them, and there was no point reducing precision,

The reason for ARM to use 0xFFFFFFFF/d instead of 0x100000000/d is that the first one can be computed with a native 32x32bit division instruction, but the 2nd one required a software implemented 64/32 bit division, that is 10x slower

eyal0, 2018-07-21:
In the same file planner.cpp, there is a call to get_period_inverse that is called with the time to accelerate from initial to cruising speed. If those speeds are the same, the time is zero. Crash.

But the inverse period is used to compute bezier coefficients and the bezier is only used when the step count is in the scurve range, which will be zero.

So the inverse of zero will be wrong but the number will have no effect anyway.

The non AVR code has no time constraints, it doesn't need optimizing. I suggest that inverse of 0 be set to same as inverse of 1, to be consistent with AVR.

ejtagle, 2018-07-21:
i have no.problems with that :+1:

closed 2018-07-25 by thinkyhead