MarlinFirmware/Marlin #11318: Extruders not working

caster1, 2018-07-20:
Everything is working except for extruder, I tried to change vref changed the driver changed the ramps board changed the cable and the motor is working fine when attached to any other axis,the motors shaft is so tight to rotate when the power is on and when I turn off power it's making noises.buzzing noises but not rotating at all,plz help me out

ManuelMcLure, 2018-07-20:
Have you preheated the hot end to the extrusion temperature? Marlin by default will not extrude if the hot end is cold. See `PREVENT_COLD_EXTRUSION` in `Configuration.h`.

caster1, 2018-07-20:
Yep the problem seems to be that my new driver is also defective and yea I allowed cold extrusion thank you it's solved.

closed 2018-07-20 by caster1