MarlinFirmware/Marlin #11320: bugfix v1.1.x 3bbb121 #error "You are using an old Configuration_adv.h file

bruce356, 2018-07-20:
Tried to compile as downloaded with no changes made to config.h or adv.h
In file included from sketch\MarlinConfig.h:41:0,

                 from sketch\G26_Mesh_Validation_Tool.cpp:27:

SanityCheck.h:46: error: missing binary operator before token "("

 #if !defined(CONFIGURATION_H_VERSION) || HEXIFY(CONFIGURATION_H_VERSION) < HEXIFY(REQUIRED_CONFIGURATION_H_VERSION)

                                                ^

SanityCheck.h:50: error: missing binary operator before token "("

 #if !defined(CONFIGURATION_ADV_H_VERSION) || HEXIFY(CONFIGURATION_ADV_H_VERSION) < HEXIFY(REQUIRED_CONFIGURATION_ADV_H_VERSION)

                                                    ^

exit status 1
missing binary operator before token "("

Roxy-3D, 2018-07-20:
I'm not sure why G26 needs to know the version of the Configuration.h file???   That isn't something I put in there.

bruce356, 2018-07-20:
How would I go about removing the "version" sanity check (so it compiles), from my perspective its a pain
regards

GMagician, 2018-07-20:
> Tried to compile as downloaded with no changes made to config.h or adv.h

Are you coming from a previous marlin version?

bruce356, 2018-07-20:
No this is a new install, I am trying to compile the latest bugfix1.1.x (testing before I make any changes to config files) using Arduino v1.8.6

Roxy-3D, 2018-07-20:
What bed leveling system did you turn on?    I'm wondering if the default
configuration (with the bed leveling system you are using) and G26 turned
on will cause the problem.
If we can duplicate the problem, it should be easy to fix.

bruce356, 2018-07-21:
Thanks for the replies, I have just downloaded the latest "Marlin-bugfix-1.1.x_21-07-18_commit 3bbb121 again. This time it compiled normally with no errors. Not sure but it seems there was a problem with my first download (sorry).

I have a problem with UBL on my dual X carriage Cartesian printer.
For every tool change to T1 the Z axis lowers closer to bed (by about 0.10 or 0.15 but the GLCD continues to show the correct height for the first layer of 0.25mm ),  Z remains at that lowered height when T0 is activated for printing the second time, on tool change back to T1 the Z axis lowers again closer to bed. Over several cycles, this results in the hotends crashing into the bed.
I remember some time back Thinkyhead became aware of this, but I guess it still has not been rectified.
Note! When bed levelling is configured using Bilinear the above Does Not occur.
I should open another issue for this.
Thanks and regards - bruce