MarlinFirmware/Marlin #11326: Mix extruder do not use DISTINCT_E_FACTORS for second extruder motor

mtmello2003, 2018-07-21:
1.1.x-bugfix

My config.h:
#define EXTRUDERS 1

//#define SINGLENOZZLE

#define MIXING_EXTRUDER
#if ENABLED(MIXING_EXTRUDER)
  #define MIXING_STEPPERS 2        // Number of steppers in your mixing extruder
  #define MIXING_VIRTUAL_TOOLS 7 //16  // Use the Virtual Tool method with M163 and M164
  //#define DIRECT_MIXING_IN_G1    // Allow ABCDHI mix factors in G1 movement commands
#endif

#define DISTINCT_E_FACTORS
#define DEFAULT_AXIS_STEPS_PER_UNIT   {159.0568463, 159.0568463, 802.6794205, 173.06, 1658 }

Using mix extruder option (cyclops hotend), the second E step/mm is completely ignored by marlin and it uses the first E step/mm value. 

I have two different extruders: "mk8" for E0 and "gregs wade" for E1. Both need different step/mm config and DISTINCT_E_FACTORS tag is enabled.

The gcode command also gets error:
>>> M92 T1 E1658.00
SENDING:M92 T1 E1658.00
echo:M92 Invalid extruder 1

GMagician, 2018-07-21:
I have not checked code and maybe this is a real bug, but have you tryed with M502  M500?

thinkyhead, 2018-09-12:
`DISTINCT_E_FACTORS` is not supported with `MIXING_EXTRUDER` at this time.

boelle, 2019-02-20:
@thinkyhead so we can close this one? or make it a feature request?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead