MarlinFirmware/Marlin #11327: Print pauses at different spots in the print

bfrakes256, 2018-07-21:
Using Cura to set up a small print to determine if the holes I need are certain size for some items I want to place in them.  Have set retraction and z hop when traveling over printed areas.  

As far as I can tell, it is retracting when moving but is not doing the z hop.  In Cura there are some settings that I can tweak, so I'm not too concerned with that unless it is causing the pausing when the retraction is happening.  

In the configuration.adv_h there is a setting FILAMENT_UNLOAD_DELAY which is set to 5000 ms.  I suspect this might be causing it.  This pausing is making the print take longer than it should and also puts a divit in the print from the nozzle melting the material.  Later as it moves over these divits it can cause the y or x axis to lose steps and the print skews off usually in the y direction.

Need some ideas on how to determine what is going on here...

TIA

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

forkoz, 2018-07-23:
Use a different slice, turn off retraction, that's what I'd try first. The unload delay... are you changing filament mid print?

bfrakes256, 2018-07-23:
No, single extruder.  I put another computer on the printer and it did the same thing.  I don't think it is the cable or USB port.  It said in Repetier to set the buffer back from 127 to 63.  Seems backwards to me.  Normally buffers go larger rather than smaller.

forkoz, 2018-07-24:
Then its probably nothing to do with filament unload delay. Would be good to know what board and what version of marlin and maybe post configs.

bfrakes256, 2018-07-25:
Megq 2560 with Ramps 1.4  Marlin bugfix 2.0

attached are the .h files in a zip.

gloomyandy, 2018-07-25:
You can't upload files to git via email responses, you need to use github directly to upload your configuration files.

moclub4, 2018-07-25:
Same issue here, print pauses, display shows "ready". After a few seconds the print restart automatically, but at the point of pause, there is a thick filament spot on the object. 
Board Melzi Creality v 1.1.2 on Ender 3, Marlin Bugfix 2.0

bfrakes256, 2018-07-25:
I'm using Rebetier (latest build).  Wondering if it is a bug in their code?  Never noticed it on the previous version.  Still have the previous build.  Will reload and try again.

bfrakes256, 2018-07-25:
Also, tell me if you see this...

kK?ok in the scrolling window.  Every other time it stops, I see this while it is halted.  On my printer the halt lasts more than a few seconds.  Usually between 10 and 20 seconds.

Running Repetier on an Azulle diskless box with W10 and 4G memory.  Thought it might be IO bound.  Changed to an Asus Laptop with 16G memory.  Same story.  Not the computer or cable.

bfrakes256, 2018-07-25:
Just completed one of the prints on a piece that stopped many times in Repetier (latest) using Pronterface.  No halts at all and the same part came out much better than in Repetier.

Tried an earlier version of Repetier and the print also halted several times before I stopped it.

It does appear that Repetier has some issues.  Going to report my tests to Repetier and see what happens.
On 7/25/2018 10:47:09 AM, Bob Frakes <bfrakes256@gmail.com> wrote:
Also, tell me if you see this...

kK?ok in the scrolling window.  Every other time it stops, I see this while it is halted.  On my printer the halt lasts more than a few seconds.  Usually between 10 and 20 seconds.

Running Repetier on an Azulle diskless box with W10 and 4G memory.  Thought it might be IO bound.  Changed to an Asus Laptop with 16G memory.  Same story.  Not the computer or cable.

bfrakes256, 2018-07-25:
Answer from Repetier...

This is a simple communication error making communication stall until timeout triggers. A good solution is to tell marlin to include linenumbers and send wait when idle. That allows the host early detect this and solve such problems on the fly.

Any ideas on how to do this?
On 7/25/2018 3:12:26 PM, Bob Frakes <bfrakes256@gmail.com> wrote:
Just completed one of the prints on a piece that stopped many times in Repetier (latest) using Pronterface.  No halts at all and the same part came out much better than in Repetier.

Tried an earlier version of Repetier and the print also halted several times before I stopped it.

It does appear that Repetier has some issues.  Going to report my tests to Repetier and see what happens.
On 7/25/2018 10:47:09 AM, Bob Frakes <bfrakes256@gmail.com> wrote:
Also, tell me if you see this...

kK?ok in the scrolling window.  Every other time it stops, I see this while it is halted.  On my printer the halt lasts more than a few seconds.  Usually between 10 and 20 seconds.

Running Repetier on an Azulle diskless box with W10 and 4G memory.  Thought it might be IO bound.  Changed to an Asus Laptop with 16G memory.  Same story.  Not the computer or cable.

thinkyhead, 2018-07-27:
This option:
```cpp
// Bad Serial-connections can miss a received command by sending an 'ok'
// Therefore some clients abort after 30 seconds in a timeout.
// Some other clients start sending commands while receiving a 'wait'.
// This "wait" is only sent when the buffer is empty. 1 second is a good value here.
//#define NO_TIMEOUTS 1000 // Milliseconds
```

bfrakes256, 2018-07-27:
Thank you.

I'll set it tomorrow...

This option:

lfps29, 2018-08-21:
I was facing the same issue, with: "kk?ok"

the option mentioned, totally works for me. Thanks.

closed 2019-02-19 by thinkyhead