MarlinFirmware/Marlin #11332: [bugfix-2.0.x] Serious bug in Delta

JXPA, 2018-07-22:
Hello.
A fatal bug occurred in Delta.
Reproducibility is 100%.

I took a video when a bug occurred.
Printers are running with dry run to save heating time and safety.
[Video(imgur)](https://i.imgur.com/b9CXaOk.mp4)

This runaway occurs when the nozzle being printed approaches to a place close to the tower.
In video, it is occurring in Y tower, but a similar runaway occurred when approaching another tower by another Gcode.

The 51st line of Gcode is displayed in the log just before the emergency stop of the printer.
However, it is near the 40th line that the nozzle runaway.

It does not occur when approaching the tower with G29, G33.
It occurs only during printing.

config files
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2216768/Configuration.zip)

Gcode
[CFDMP_MB_HOLDER.txt](https://github.com/MarlinFirmware/Marlin/files/2216769/CFDMP_MB_HOLDER.txt)

wailord284, 2018-07-22:
Out of curiousity are you also having leveling issues? I just built myself a Delta from scratch and the machine is physically sound but has a very hard time leveling with G29 or G33. I dont have the same issue as you with runaway but I'm almost positive the leveling issue is a bug as well in 2.0

JXPA, 2018-07-22:
No, there is no problem with leveling.

mylife4aiurr, 2018-07-23:
As of 3bbb121 bugfix 1.1x - G33 did the same for me on my delta with bltouch....doesn't obey bed size. G29 does obey bed size.

I mentioned the issue here as well #11229, but the focus of that thread was bltouch 2.0.

thinkyhead, 2018-11-05:
`G33` seems to require a probe which uses the nozzle, and does not support probes with an X or Y offset. Maybe it would work with a very small XY offset.

closed 2018-11-05 by thinkyhead