MarlinFirmware/Marlin #11335: TMC2130 Can't change direction

ticolaugs, 2018-07-22:
### Description
I have recently bought 4 Fystec TMC2130 V1.1 (the SPI version). After setting them up on my Geeetech gt2560 board i tested them. While testing i ran into the problem that all axes move forward when receiving a signal to move forward as well as when receiving a signal to move backwards.

After reading several issues on the Marlin GitHub with a similar problem i have tried all the following things:
- Using a multi meter tested to see if the direction pin is connected to the TMC2130 package (it was). [Reference](https://goo.gl/2HsUa7)
- Using a multi meter tested to see if the TMC2130 receives a direction signal (it does).
- Changed direction normal High in Marlin (didn't solve it).
- Both with a visual inspection and using a multimeter tested to see if the SPI bridge and CFG4 and 5 where soldered up correctly, using  [this](https://goo.gl/ciW9nh) as reference (everything was soldered correctly).
- Tried to see if the problem also occurs with a different (a4988) driver (it doesn't).

- Checked to see if it is an end stop issue using M119 (everything works fine).
All end stops unpressed
`X_min: Open`
`Y_min: Open`
`Z_min: Open`
Pressing X end stop:
`X_min: Triggered`
`Y_min: Open`
`Z_min: Open`

### Steps to Reproduce
1. Wire up TMC2130 V1.1 on the gt2560.
2. Install Marlin on gt2560
3. Make correct changes to the Configuration.h and Configuration_adv.h.
4. Using Repetier Host (or other platform) move and axis forward.
5. Using Repetier Host (or other platform) move and axis backward.

**Expected behavior:**
Axis moves forward. Axis moves backward.

**Actual behavior:**
Axis moves forward. Axis moves forward again.

#### Additional Information
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2216915/Configuration.zip)

mbuc, 2018-07-24:
Is wiring those different than wiring an official trinamic driver?

wbarber69, 2018-07-27:
Make sure you didn’t enable the bit about sensorless homing.  Even if you aren’t using it.

ticolaugs, 2018-07-27:
Thank you for the replies.

**mbuc**: The wiring of the Fystec drivers is the same as official (Watterott) tmc drivers.

**wbarber69**: I have the sensorless homing disabled.

I found a post from a person that had the same problem with a different driver. He was able to solve it by changing the wiring from the driver to the motor (switch the spools). I will try this soon and comment the result.

thinkyhead, 2018-09-06:
Assuming this has been solved…?

ticolaugs, 2018-09-09:
@thinkyhead sadly it has not. I have tried everything i can imagine yet i can't get it to work. I have given up hope on solving it but decided to keep this issue open in case anyone came up with a (possible) solution.

thinkyhead, 2018-09-12:
Have you been able to test with a new or different set of drivers? Do those drivers have a trace on the board that must be either cut or shorted in order to get DIR control? Please post close up photos of your drivers for inspection. Top and bottom!

esrefulus34, 2018-12-24:
I have 4 tmc2130 spi drivers. 3 of them works properly, one of them has the same problem. I suspect somebody placed driver reversed on a board. The dir pin has been connected to "vm" pin of the board, and died. 
But interstingly when I tested the "bad" stepper with arduino and (https://github.com/teemuatlut/TMC2130Stepper) library, the motor turns forward and backwards properly. 
 I am not an expert, but when digged into the library code, (I think) TMC2130stepper library sends direction commands over spi ,by shaft_dir() function. But the library which is used by marlin firmware(TmcStepper) changes dir pin directly (5v or 0v) to determine the direction.

wbarber69, 2018-12-24:
Only configuration and diagnostic info is passed over spi.  Motor movements and speed etc are sent exactly the same way any other driver is.

boelle, 2019-02-20:
@ticolaugs has this been solved in latest bugfix 2.0?

ticolaugs, 2019-02-26:
@boelle I have since bought a new printer and don't have the old drivers anymore so i can't test

closed 2019-02-26 by ticolaugs

puddly, 2019-03-31:
I encounter this problem on and off with the latest `bugfix-2.0.x` branch and genuine TMC2130 drivers with the Re-ARM. It affects my x axis more often than the y but is usually solved by just rebooting a few times.

I've replaced the x axis driver with a new TMC2130 and it still happens but now moving the axis forward with `G1 X20` after moving it backward with `G1 X-20` seems to move it backward by 5 and forward by 15, like there's a delay before the DIR pin is toggled.

Bob-the-Kuhn, 2019-03-31:
Try increasing the stepper current.  You can get strange behavior like this if the current is too low.

puddly, 2019-03-31:
@Bob-the-Kuhn I was using 1100mA but changing to 800mA or 1300mA doesn't help, neither does disabling stealthchop.

Bob-the-Kuhn, 2019-04-01:
Try increasing it to the maximum current.

Do you have any other types of driver?  The A4988 seems to be able to drive steppers that others can't.

If things work properly with another type of driver then either the TMC2130 just doesn't have the muscle or you need to go to lower microsteps and try again.  The higher the microsteps the more current is needed to drive the stepper.

ButchMonkey, 2019-05-10:
Also having this issue with 1.1.7. After switching from A4988 to TMC2130 the printer homes in the positive direction despite having home direction set to min 
```
// Direction of endstops when homing; 1=MAX, -1=MIN
// :[-1,1]
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1
```

puddly, 2019-05-10:
@ButchMonkey does it still move properly? I think my problem may have been caused by the driver seating in either the stepper protector boards or the thin stackable headers of the stepper protector boards in the RAMPS driver sockets, so you may want to check those too.

ButchMonkey, 2019-05-10:
@puddly See my edit. I just swapped the direction of the motors and its okay now. I couldn't move the steppers correctly as I needed to home first. Now everything works fine. Not entirely sure why the direction was flipped with the TMC2130 as I didn't remove the steppers at all