MarlinFirmware/Marlin #11337: Problem with Delta Calibration - Probe z-offset / G33 P-1

grbd, 2018-07-22:
### Description

I've been experimenting with the bugfix-1.1.x on a Anycubic Delta printer
but noticed a problem with the Probe z-offset / G33 P-1

Printer: Anycubic Delta Kossel Linear Plus
Firmware: bugfix-1.1.x / latest 3bbb121e7d7aa47af80a4e491fcaef271ceee79c

Under Prepare -> Delta Calibration

  * Auto Calibration - works fine
  * Set Delta Height - works fine
  * **Probe z-offset - touches probe to bed, then scrapes probe alog side the bed** G Code G33 P-1

As far as I can tell it was introduced with https://github.com/MarlinFirmware/Marlin/pull/8821
It managed to nock a bit of the end of my V1 probe, (it still works but has ended up being a bit shorter)
so I've now ordered a V2 probe off of ebay since it looks like they're a bit more resilient

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/2217325/Config.zip)

grbd, 2018-07-28:
Okay I've now discovered how I think this is supposed to work (there's currently no docs on this yet)
The idea is to

  * Measure the Delta Height with a Z Min Probe first
  * Remove the Z Min Probe from the printer
  * Select Probe z-offset from Delta Calibration
  * the printer then moves down to 5mm above the bed
  * I think the idea is to then put a piece of paper underneath and adjust the head down until it touches.
  * By selecting a new value this then auto adjusts the Probe z-offset

There seems to be a couple of problems I've observed though (assuming this is how it's supposed to work)

  * When I've tried this, and adjusted the value by about 0.05mm from 0, the Z Probe Offset jumps up from -16.00 to -25.00mm approx which is way out
  * It might be an idea to add in a safety feature / option to abort the procedure if the Z Min Probe is detected as switching when the head is moving downwards (I managed to break a bit off the end of mine)

brandstaetter, 2018-09-06:
Oh great, you had the same experience as me. My probe is now shorter too.

Suggestion (don't know my way around the code yet, sorry - I'd love some instructions though)

`#if ENABLED(FIX_MOUNTED_PROBE)`
and `Z_PROBE_OFFSET_FROM_EXTRUDER < 0`
PAUSE after the Z-probe (at `Z+Z_CLEARANCE_DEPLOY_PROBE`), print message on screen "Remove Probe if necessary" and only after the user klicks or otherwise continues, do the paper test. Might save some pool microswitches.

thinkyhead, 2018-09-09:
This issue is _50 1337_!

Jvbx, 2018-09-15:
Had the same issue. 
Looks like setting next options to values bigger then Z Probe offset fixes such behaviour:

#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
**#define Z_CLEARANCE_BETWEEN_PROBES  5** // Z Clearance between probe points
**#define Z_CLEARANCE_MULTI_PROBE     5** // Z Clearance between multiple probes   (i'm change this values to 20)
#define Z_AFTER_PROBING            30 // Z position after probing is done

As I assume - after first probing, nozzle trying to go to Z coordinate of 5 mm, while typical microswitch probe's height are more then 15mm. Result is predictable, so this values shold be increased to Z Probe Offset + some free space to detach the probe. 
Of course it's not an ultimate solution, but could be used as temporal workaround to keep our probes from digging the bed.

------
And sry for possibly poor english, it's not my native.

brandstaetter, 2018-09-15:
Oooh, that sounds plausible. I'll try adding the probe offset as you suggested.

brandstaetter, 2018-09-15:
Okay, It resolves the problem for now, but I think that the comment above hints at the fact that there might be a bug in the implementation. I already tried to understand G33.cpp but I could not find the issue yet (my C++ is a bit rusty, more of a Java guy myself...)

Pull request incoming.

tcm0116, 2018-09-20:
>It might be an idea to add in a safety feature / option to abort the procedure if the Z Min Probe is detected as switching when the head is moving downwards

I think this is a good idea, but could only only work for normally-open probes since Marlin would think that a normally-closed probe is triggered when it's removed. 

> PAUSE after the Z-probe (at Z+Z_CLEARANCE_DEPLOY_PROBE), print message on screen "Remove Probe if necessary" and only after the user klicks or otherwise continues, do the paper test. Might save some poor microswitches.

I think this is a great idea, and can be coupled with the above (if possible).

> Okay, It resolves the problem for now, but I think that the comment above hints at the fact that there might be a bug in the implementation.

I agree that there might be a bug. It sounds like something is not taking the psychical offset of the probe into account as the documentation implies. Would it be possible to take a video of the issue occurring (with the probe removed, so as to not cause any more damage)?

grbd, 2018-09-26:
I'm afraid I can't test myself as I've switched over to the Duet board
but if it's resolved then please feel free to close the issue

boelle, 2019-02-20:
@thinkyhead i think we can close this one

hreality, 2019-02-28:
I’m attaching my probe and have the same crashing issue without any kind of pause for removal (Kossel).  Need more digging on my end..

brandstaetter, 2019-02-28:
The pause feature is available on the Bugfix 2.0.x branch

closed 2019-05-16 by thinkyhead