MarlinFirmware/Marlin #11338: Is it possible to ad a feature to turn off linear advanced easy in the LCD.

TB1405, 2018-07-22:
Hi 
iam using the linear advanced feature on my printer and it works so nice with PLA and PETG. 
But if i print with flex. filament the result is not very nice. 
So my question is. 
Is it possible to turn off the linear advanced feature for one or 2 prints without loading the firmware new ? 
Is there feature?
Thanks 
TB1405

ManuelMcLure, 2018-07-22:
Setting the Linear Advance K-factor to 0 should disable LA, so a simple `M900 K0` at the start of your gcode will do the trick. In Slic3r you can set different start gcode for different filaments which makes this easy.

J-charles-C, 2018-07-24:
I,

you have the solution here :

https://youtu.be/ldTYYAidTT4?t=3m26s

best

forkoz, 2018-07-24:
set linear advance to 0 in the firmware and only set it with gcode script. Your advance is probably different for PETG and PLA anyways, not just TPU.

TB1405, 2018-07-25:
i have to try it with set the k-factor in the slicer (Cura) to zero. But the  k-factor is a value between 0-2 and so i think Linear Adnvanced isnt disabled with zero.
Yes u know that all materials have a diffrent k-factor. 
Does the k value also change with layer height.

thinkyhead, 2018-07-27:
> so i think Linear Advance isn't disabled with zero

It is effectively disabled if set to zero.

```cpp
      block->use_advance_lead =  esteps
                              && extruder_advance_K
                              && de > 0;
```

TB1405, 2018-07-27:
Alirght than i will close this thank you for your replys.

closed 2018-07-27 by TB1405